arendst/Tasmota #3308: Rule trigger that differentiates power via button/cmnd vs power via timer?

roguestreak, 2018-07-23:
- [ X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X ] Hardware used :  Sonoff S22 socket
- [ X ] Provide the output of command ``status 0`` :
```
20:20:11 MQT: stat/sonoff_S22/STATUS = {"Status":{"Module":8,"FriendlyName":["Sonoff_S22"],"Topic":"sonoff_S22","ButtonTopic":"","Power":0,"PowerOnState":0,"LedState":8,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:20:11 MQT: stat/sonoff_S22/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:06:32","StartupUTC":"2018-07-23T10:13:39","Sleep":0,"BootCount":111,"SaveCount":517,"SaveAddress":"FA000"}}
20:20:11 MQT: stat/sonoff_S22/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-07-22T20:47:42","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:20:11 MQT: stat/sonoff_S22/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["OPTUSFEFF922","indebuurt2"],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
20:20:11 MQT: stat/sonoff_S22/STATUS4 = {"StatusMEM":{"ProgramSize":433,"Free":568,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F80A794","00000000","00000496","00000000"]}}
20:20:11 MQT: stat/sonoff_S22/STATUS5 = {"StatusNET":{"Hostname":"sonoff_S22-3886","IPAddress":"192.168.1.22","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:3A:E8:0F:8F:2E","Webserver":2,"WifiConfig":1}}
20:20:11 MQT: stat/sonoff_S22/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"sonoff_S22","MqttClient":"sonoff_S22","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:20:11 MQT: stat/sonoff_S22/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 23 10:20:11 2018","Local":"Mon Jul 23 20:20:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10,"Sunrise":"14:12","Sunset":"05:41"}}
20:20:11 MQT: stat/sonoff_S22/STATUS10 = {"StatusSNS":{"Time":"2018-07-23T20:20:11"}}
20:20:11 MQT: stat/sonoff_S22/STATUS11 = {"StatusSTS":{"Time":"2018-07-23T20:20:11","Uptime":"0T00:06:32","Vcc":3.208,"POWER":"OFF","Wifi":{"AP":1,"SSId":"OPTUSFEFF922","RSSI":94,"APMac":"BC:AE:C5:EA:C9:A7"}}}
```

Thanks for implementing a solution for Issue #3270 (enable/disable timing system), but now I have an issue with finding a rule to implement it selectively enough;  I want to disable the timing system when I manually turn on a device, and re-enable it when I turn it off (via button or mqtt cmnd).  The following rule achieves this, BUT it also gets triggered by timer actions, which then disables the timing system which then doesn't let the timer to turn it off which then...

rule1 on Power1#state=1 do Timers 0 endon on Power1#state=0 do Timers 1 endon

I've tested with Button1 and Switch1, and neither of these seem to be triggered by the button press at all.  Both ButtonTopic and SwitchTopic are set to 0.  Any suggestions for a trigger that will achieve my goals, possibly after changing a configuration setting somewhere, or is there a need for a new rule trigger?

ascillato, 2018-07-23:
Hi,

Sorry, I'm not understanding exactly what you want to achieve.

You want to do the command `timers 0` when pressing a button, and command `timers 1` when pressing again?

As you want to turn all timers off when you left your home (as I think I have understood before), why not setting that by mqtt like: `cmnd/sonoff_S22/timers 0` ?

roguestreak, 2018-07-23:
Hi ascillato

In a nutshell, yes.

While witek1308's initial suggestion for the enabling/disabling was to easily enable someone going on holiday via mqtt comand, I realised it should also enable a "manual bypass" system whereby a rule could be setup in the device itself so that if you manually press the "on/off" button on the device you also inversely toggle the timer system;

Let's say, as in my case, you have a pool pump, with three pairs of on/off timers scattered throughout the day.  Nominally I need to run my pump for about a minimum of 2hrs total to achieve maintenance-level filtering and chlorination, so I might have it setup for 45min, 45min and 30min spread across the main period of solar PV (eg. trigger at 10am, 12pm, and 2pm respectively)

However if/when you want to go for a swim, you want the pump running at all times during that period, without it turning off when it comes across one of your pre-set timer "off" times.  So I figured with a combo of the new timer disable/enable system and an inverse rule I could have it work as the following, a "manual bypass";

- physically turn the device relay ON, rule DISABLES timing system
- physically turn the device relay OFF, rule ENABLES timing system

The rule I posted above achieves this on button press, but is also triggered when a timer powers on/off the relay; when the first instance of the time powering on the relay happens it DISABLES the timing system, which means the "off" timer can no longer trigger.

I can't figure out a rule that is triggered when you press the button manually, experiments with the "button" and "switch" rule triggers have yielded nothing but I don't know if I'm just seting them up wrong.

ascillato, 2018-07-23:
> physically turn the device relay ON, rule DISABLES timing system
> physically turn the device relay OFF, rule ENABLES timing system

```
rule on switch#state=1 do backlog power1 on; timers 0 endon on switch#state=0 do backlog power1 off; timers 1 endon
rule 1
```

roguestreak, 2018-07-24:
Hi ascillato, that rule doesn't trigger the timer enable/disable in my case, here's the results of physically toggling the button/power several times
```
18:34:50 CMD: rule 1
18:34:50 MQT: stat/sonoff_S22/RESULT = {"Rule1":"ON","Once":"OFF","Free":399,"Rules":"on switch1#state=1 do backlog power1 on; timers 0 endon on switch1#state=0 do backlog power1 off; timers 1 endon"}
18:34:56 MQT: stat/sonoff_S22/RESULT = {"POWER":"ON"}
18:34:56 MQT: stat/sonoff_S22/POWER = ON
18:35:01 MQT: stat/sonoff_S22/RESULT = {"POWER":"OFF"}
18:35:01 MQT: stat/sonoff_S22/POWER = OFF
18:35:10 MQT: stat/sonoff_S22/RESULT = {"POWER":"ON"}
18:35:10 MQT: stat/sonoff_S22/POWER = ON
18:35:12 MQT: stat/sonoff_S22/RESULT = {"POWER":"OFF"}
18:35:12 MQT: stat/sonoff_S22/POWER = OFF
```

ascillato, 2018-07-24:
A switch to be used in rules needs to be configured that gpio as switch in your module configuration. If you have that configured aa button it will not work.

roguestreak, 2018-07-24:
I'm using a stock-standard S22, and will be re-deploying to a stock Basic, so this is just using the button on GPIO0 which has no web interface option to change; what commands do I need to do to get this setup correctly for your rule to operate?  I've just spent the past hour scouring the wiki, issues, sonoffusers etc and am still none the wiser.  I've been here for 7 months now and I still don't understand the distinction between "button" and "switch", and not for lack of trying.

ascillato, 2018-07-25:
> what commands do I need to do to get this setup correctly for your rule to operate?

In the **web menu**, in the **configuration module**, define your module as **_Generic_** and set:

```
GPIO00 as Switch1
GPIO02 as AM2301 (your optional Temperature sensor)
GPIO12 as Relay1
GPIO13 as LED1i 
```

> I still don't understand the distinction between "button" and "switch"

**Button**: is a push button (it clicks and come up again). It is a pulsator like the one in your doorbell.

**Switch**: is a key. The one of your home lights. You press it and it stays in that position until you press it again.

ascillato, 2018-07-25:
Try that to see if fits your requirements, but I think that should be better to use the configuration module as Sonoff TH and leave the control of your timers to your Home Automation Software. Your HA can control your timers with the MQTT `cmnd/sonoff_S22/timers 0` and `cmnd/sonoff_S22/timers 1`

roguestreak, 2018-07-25:
Initially I was just getting "cmnd/s22_manual/POWER1 = TOGGLE" with no relay action, after much messing around with switchmode, switchtopic, etc. I got the relay working as intended but the rule isn't triggering at all; the below console output shows no difference if the rule is enabled or disabled.  Tried adding uppercase to match the Rule wiki but no effect.

```
06:49:40 MQT: stat/sonoff_S22/RESULT = {"POWER":"ON"}
06:49:40 MQT: stat/sonoff_S22/POWER = ON
06:49:45 MQT: stat/sonoff_S22/RESULT = {"POWER":"OFF"}
06:49:45 MQT: stat/sonoff_S22/POWER = OFF
06:50:37 CMD: rule 0
06:50:37 MQT: stat/sonoff_S22/RESULT = {"Rule1":"OFF","Once":"OFF","Free":438,"Rules":"on Switch1#state=1 do Timers 0 endon on Switch1#state=0 do Timers 1 endon"}
06:50:43 MQT: stat/sonoff_S22/RESULT = {"POWER":"ON"}
06:50:43 MQT: stat/sonoff_S22/POWER = ON
06:50:46 MQT: stat/sonoff_S22/RESULT = {"POWER":"OFF"}
06:50:46 MQT: stat/sonoff_S22/POWER = OFF
06:50:51 CMD: rule 1
06:50:51 MQT: stat/sonoff_S22/RESULT = {"Rule1":"ON","Once":"OFF","Free":438,"Rules":"on Switch1#state=1 do Timers 0 endon on Switch1#state=0 do Timers 1 endon"}
06:50:57 MQT: stat/sonoff_S22/RESULT = {"POWER":"ON"}
06:50:57 MQT: stat/sonoff_S22/POWER = ON
06:51:00 MQT: stat/sonoff_S22/RESULT = {"POWER":"OFF"}
06:51:00 MQT: stat/sonoff_S22/POWER = OFF
```

PS yes that would be my normal definition of "button" and "switch", but it works differently in Tasmota land; the fact that we're talking about using a button as a switch for one, for another that certain things only work for switches and not buttons and vice-versa...

OK, coming at this from a different direction; does the timer system issue an mqtt cmnd to trigger the internal relay, or does it do it via another method?  If not using internal mqtt, could an additional Rule be created (or does it already exist undocumented as an Event?) to be triggered by a specified mqtt?  eg. "cmnd/sonoff_S22/POWER"?

If I had to bring in the HA system I'd revert to what I was attempting to setup previously, publishing the mqtt cmnd on manual switching (via button or interface), which I found did NOT publish when the timer kicks in.  The HA can monitor that topic and publish a timers 1/0.  However this is not ideal, and we seem to now have a syste in place that theoretically could do all of that onboard.

I'd only use the HA as a timer backup, using it to monitor the on/off msgs from the timing system and only kick in if things don't go as planned.  Yes you could go the other way with everything done by the HA and the timers only being an internal backup, but I think I prefer to operate the other way around.

ascillato, 2018-07-26:
I'm sorry. I do not know what else to say to you besides that your HA can control your timers with the MQTT `cmnd/sonoff_S22/timers 0` and `cmnd/sonoff_S22/timers 1`.

ascillato, 2018-07-31:
Hi,

> Rule trigger that differentiates power via button/cmnd vs power via timer?

It is not supported.

Use another approach:

* Rules with switch
* MQTT message from you Home Automation Software.

Hope this helps.

Thanks

roguestreak, 2018-08-01:
As above, I tried implementing the switch rules as per your instructions but it's not triggering.

ascillato, 2018-08-24:
Hi,

Your rule:

`rule on Switch1#state=1 do Timers 0 endon on Switch1#state=0 do Timers 1 endon`

is not triggering because you are assuming that you have 1 and 0 for your switch. If you did not change your `switchmode` option, it will be giving the state=2 (toggle)

Extracted from the wiki:

Command           | Payload        | Description
------------------|----------------|-------------------------------------------------------------------
SwitchMode\<x\>  |              | Show current external switch mode
SwitchMode\<x\>  | 0            | (default) Set switch mode to TOGGLE
SwitchMode\<x\>  | 1            | Set switch mode to FOLLOW (0 = Off, 1 = On)
SwitchMode\<x\>  | 2            | Set switch mode to inverted FOLLOW (0 = On, 1 = Off)
SwitchMode\<x\>  | 3            | Set switch mode to PUSHBUTTON (Normally 1, 0 = toggle)
SwitchMode\<x\>  | 4            | Set switch mode to inverted PUSHBUTTON (Normally 0, 1 = toggle)
SwitchMode\<x\>  | 5            | Set switch mode to PUSHBUTTON (Normally 1, 0 = toggle, Hold = hold)
SwitchMode\<x\>  | 6            | Set switch mode to inverted PUSHBUTTON (Normally 0, 1 = toggle, Hold = hold)
SwitchMode\<x\>  | 7            | Set switch mode to PUSHBUTTON (0 = toggle, 1 = toggle)

So,

assuming that your switch is on `GPIO00` and configured as `Switch1`, please, try in your console:

```
switchmode1 1
rule 1
rule on Switch1#state=1 do Timers 0 endon on Switch1#state=0 do Timers 1 endon
```

I test that and behaves as you requested. Also, I have added this example to the wiki, so to clarify this.

Hope this solves your issue and Thank you for the patience and for sharing your results. :+1:

ascillato2, 2018-08-27:
Please, share your results when you have time.

Thanks.

roguestreak, 2018-08-31:
Thanks Adrian, I pasted in those three lines of code from the pull request and recompiled, and with the pushbutton rule above can now toggle the timers using the built-in button.

However once the rule is brought in it stops the relay responding, I believe this is as per normal you just need to add your own rule to handle the relay, but so far I haven't been able to come up with a rule that works despite copying various rules from the examples and trying all sorts of combinations; perhaps there's something missing in my setOptions?
Examples of what I've tried; tried explicitly comparing states, tried words (toggle) vs numbers (2), etc., pretty sure I'm missing something "obvious" but my searches and experiments haven't turned up a result

on button1#state Power1 Toggle endon
on button1#state=2 Power1 Toggle endon
on button1#state=1 Power1 2 endon (eg. testing if on when pressing down)

I don't need to publish a cmnd when dealing with the onboard relay correct?

ascillato, 2018-08-31:
Ah, ok, so, besides the timers you want to toggle the relays?

Use the `backlog` command to it.

So, rewriting the rule for pushbuttons:

[ Pushbutton Configured as button ]

```
rule 1
rule on button1#state do backlog power1 2; Timers 2 endon
```

[ Pushbutton Configured as switch ]

```
switchmode1 5
rule 1
rule on switch1#state do backlog power1 2;Timers 2 endon
```

The command `timers 2` is now included in last development version

roguestreak, 2018-08-31:
Thanks Adrian I'll give that a go tonight, didn't consider backlog.

Yes the intention of this, and the original timer toggle request, is that often if you are manually operating the physical button on the device you want to be in a "manual bypass" mode, which shouldn't be interrupted when a pre-set timer kicks in.  The issue was how to differentiate between a manual activation (button1) and an activation from elsewhere (time, mqtt, etc).  Hopefully with the new "Timers 2" combined with the above rule this will be achieved.

I wasn't sure if the PR was in the latest development version as the changelog doesn't mention it as such

ascillato, 2018-08-31:
Mmmm, ok. So, in that case should be better to apply [this rule example](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold)

So, simple press to toggle the relay, Hold 4 seconds to toggle global enable timers.

```
buttontopic 0
setoption1 1
setoption11 1
setoption32 40
rule on button1#state=3 do timers 2 endon on button1#state=2 do delay endon
rule 1
```

(doble press do nothing - delay command)

Hope this solves your issue.

closed 2018-09-04 by ascillato2