arendst/Tasmota #3311: trigger "counter increment" by X

MatzeJoerling, 2018-07-23:
**Have you look for this feature in other issues and in the wiki?**

Yes

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

I would like to count a 30ms impulse in "realtime". The impulse is generated by a S0 power counter (S0 interface is availabe for many counters like gas, water, power ...).

30ms is to short for a switch and teleperiod is to long for realtime. 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

I would like to have a trigger for a rule that is triggered every N counter increments. 
For me N would be one because i have 1 impulse per second maximum, but there are counters which generate multiple impulses per second.
A impuse is defined as: 30 ms ≤ t_EIN ≤ 120 ms 30 ms ≤ t_AUS
https://de.wikipedia.org/wiki/S0-Schnittstelle
So there is a maximum of impulses per second.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

I have patched tasmotas main loop to run ever 1/40 second instead of 1/20.
That is working for me but hard to maintain. with 1/40 second (25ms) i catch every impulse...

**Additional context**
_Add any other context or screenshots about the feature request here._

this is a new way to come around my old problem ;-) #1235

thanks for considering ... M.

ascillato, 2018-07-23:
Hi,

I'm sorry. Tasmota is not a Real Time focused software. For making that work we should make tasmota to stop all routines to take care of that. I think that would be better to have a specific device for that like an arduino.

You can see this fundaments in the https://github.com/arendst/Sonoff-Tasmota/pull/2979 that shows how many cycles of the main Tasmota thread are free. There you can see that (because is not real time focused) the amount of free cycles are fluctuating depending on the tasks that Tasmota has to do.

That is why would be much better and realiable to pass that task to a specific device and leave your Sonoff to do just home automation.

Again, sorry if that was not the answer you were expecting.

Frogmore42, 2018-07-24:
@MatzeJoerling I have been following this request for a while.  It seems you are certain your way is what you want.  Several people have told you how to do it in a way that is consistent with the way Tasmota is designed.  There is always more than one way to do things.

It seems you need to count impulses.  Tasmota has support for that with Counters.  Counters were designed to interface with meters that use pulses for indicating measured quantities.  I use them for interfacing with a gas meter and tipping bucket rain gauges (I use the later for measuring the output water of my dehumidifier).  I find they work very well.  Before finding Tasmota, I wrote some custom code to this and it took a little while to get debouncing to work well.  In Tasmota, using a Button or Switch is not the best way to interface with these types of things, unless they move very slowly and even then it becomes more complex than using a counter.

I believe the other thing that you want is to get an MQTT message every time the counter changes.  Tasmota does not have that capability currently.  But, if you REALLY, REALLY, REALLY NEED to have every counter pulse send a MQTT message, you have access to the source and can make it do that.

Take a look at xsns_01_counter.ino 
CounterUpdate() gets called when the interrupt fires.  You could put a call to send the MQTT message then, but that would be a generally not best software practice (since you need to do the minimal work in the interrupt loop).

So, you need some way to get information to some code that is running regularly not in the interrupt context.  You would also need to make sure the value is protected from being changed by both the interrupt and non-interrupt contexts in conflict with each other (or you might miss an update, but it might not matter depending on what you are sending).  If you are going to use Tasmota to send an MQTT message for every counter impulse, this could work. 

The more normal way to do it with Tasmota is to use a counter and set teleperiod to the value you need.  People have successfully set teleperiod to 1s, which should be fast enough for what you said you needed.  I believe your concern in the other issue was that you didn't want the "overload" of sending many MQTT messages by setting teleperiod to a small value.  If so, you could modify the code that sends MQTT messages every teleperiod to not send anything if it is the same as the last message that was sent.

If none of that works for you, you should probably just create your own project.  If I were doing that, I would have an Interrupt Service Routine hooked up to a GPIO (you will need to figure out how you want to debounce that, but there are lots of examples for that).  That updates something in memory.  You have something else that runs in the normal context that is waiting for something to do.  When it sees that there has been a event generated by an impulse, it sends the MQTT message you want.

As you have discovered, getting Tastmota to continue doing all it does (and the things that it will likely get asked to do in the future) and sample fast enough to do it the way you want is very hard.

I don't believe you will find any support for changing the very basic assumption that Tasmota has made that the loop runs every 50ms as it would be a small change with very big consequences and likely lead to regressions.  It is also not necessary for any of the scenarios for which Tasmota has been designed.

MatzeJoerling, 2018-07-24:
Hi,

thanks both for for your really comprehensive answer, but let me explain some things.

@ascillato 
I dont want to have a real time tasmota os!
Only a input that writes every impulse to mqtt, not a 50kHz impulse, only up to 5Hz.
Dont want to speed up the loop, ony a trigger for counter increment.
Speed up the loop was my dirty hack - i have had no better idea, coding skills.
Counting power, water, heat and reacting to events is also home automation, not only turning lights on and off.

@Frogmore42 
I have tried patching my own tasmota, i have learned basic git, but i am struggling keeping the pace.
I have arount 15 "normal" and 2 special tasmotas (and about 80 KNX devices). 
That gives me problems on every tasmota update.
This "teleperiod to 1 second thing" is no option for me (to much overhead, not written for real time updates.) If teleperiod is written every sec the chip has the power to write a mqtt on every interrupt up to 5 Hz.

If this request doesnt made it (as it looks like) i will try to look into CounterUpdate() and try to fire a mqtt event. Pehaps easier to patch than 1/40sec main loop.

tnx M.

ascillato, 2018-07-24:
Hi,

The Tasmota loop quantity per second is fluctuating attending WIFI, MQTT, etc. so a fast counting will miss some pulses. I'm using the Tasmota Counter Feature with a water flow and works really good without any modification needed.

So, again, If you need something faster and reliable at fast speed, I recommend you to use an arduino or something specific for that, and, if you want, send those values once per second by serial to Tasmota so as to be resent by MQTT.

MatzeJoerling, 2018-07-25:
okay, let my real time approache aside.

what about a trigger that fires after n counters. 
Example: 
i want to fill up my pool, that are 1000 liter, with my water counter that are 10000 implses, tasmota should stop the pump after 9900 impulses.
i want to give my flowers about 50-80 liters water, depends on how dry the soil is. I want 50 liters, not 10min. so a trigger stop after 500 to 800 impulse would be nice.
i want to fill the akku of my bike, i know that are 54W/h about 300 impulse, then switch off after 350.
i am allowed to send 2kw/h power from my solar plant to the energy provider, after that i have to switch off or pay fee. tasmota switches of the solar plant after 4000 impulses.

A trigger for counter would be useful in other cases also, not only my "wrong" real time approach.

Any change a generic trigger after N impulses would made it into tasmota?

arendst, 2018-07-25:
Did you try to use a rule like:
```
rule2 on counter#c1>400 do publish cmnd/theo2/counter %value% endon
rule2 5
rule2 1
```
which monitors counter C1 for overflowing 400 and then executes a publish command. This could easily be changed to:
```
rule2 on counter#c1>9900 do power 0 endon
rule2 5
rule2 1
```
which will monitor the counter and turn power off after 9900 impulses.

MatzeJoerling, 2018-07-25:
Hi,

no - i did not try "on counter" because i did not find it in the wiki... and i did not read the source ;-)

```
rule2 on counter#c1>9900 do power 0 endon
rule2 5
rule2 1
```
I think i know what line 1 does, but what about line 2 aund 3?

MatzeJoerling, 2018-07-25:
@ascillato thanks for the copy & paste!
@arendst when will this trigger executed, when the counter reaches 401 or when the counter appears in any telemessage. if so, the counter could probably 600 when trigger executes?
Sorry for asking agian, for me the wiki is not clear about that. Example 5 states that a tele- prefix is only executet at teleperiod time, your example is triggered as soon as counter > 400?

thanks for your patience ...

ascillato, 2018-07-25:
Rule trigger: on counter#c1>400 do
Is executed: when the counter reaches 401. If you do not have the one-shot enabled, also is going to be executed every second after reaches 401until you reset the counter

Rule trigger: on tele-counter#c1>400 do
Is executed: every teleperiod only if that condition is met.

arendst, 2018-07-25:
@MatzeJoerling it's time you start experimenting....

arendst, 2018-07-25:
Note: rule check sensors round-robin within 2 seconds depending on the amount of sensors active so do not expect exact counter values but test on greater or smaller values

MatzeJoerling, 2018-07-25:
check! will do. 
But, consider a trigger that is interrupt driven (as counters are) to get exact values. 

tnx M.

closed 2018-07-25 by MatzeJoerling