arendst/Tasmota #3312: Livili support of RF-Bridge?

olili, 2018-07-23:
Does anybody see a chance to get Livolo protocol support included in RF-Bridge?
Details of the protocoll see here:
https://forum.arduino.cc/index.php?topic=153525.180

Thx in advance
O

Jason2866, 2018-07-23:
Take a look here
https://github.com/Portisch/RF-Bridge-EFM8BB1
could be possible via
Bucket Transmitting using command 0xB0
it is up to you -;)

olili, 2018-07-23:
Thx. 
Do you have any hint how to translate Broadlink notation of Livolo protocol (https://dimagoltsman.github.io/Random-Broadlink-RM-Code-Generator/# ) to the "bucket notation" ?

Jason2866, 2018-07-23:
No, sorry.

ascillato, 2018-07-29:
Hi

@olili

Any update to this issue? Did you find the answer to your question?

olili, 2018-07-29:
@ascillato,

not yet.

ascillato2, 2018-09-04:
Hi,

As there is no activity in this issue, I'm closing it. Please, if you have more information/questions ask to reopen it.

closed 2018-09-04 by ascillato2

helenclarko, 2019-10-10:
I'm about to attempt the Portisch firmware with this switch, I'll keep everyone updated.

murdemon, 2019-10-22:
Livolo 433 wall switch is working with https://github.com/robertcsakany  this soft and Portisch firmware

shajek, 2019-10-26:
you posting wrong link

rsneh, 2020-10-17:
There is any news regarding this? I'm not sure what is so different in Livolo switches that RF-Bridge cannot received... someone make it work?