arendst/Tasmota #3314: MQTT lost connection problem happen after Wifi router restart.

wongnam, 2018-07-24:
My sonoff basic, sonoff touch, RF bridge are facing issue with MQTT not reconnect after i restart my wifi router.

All my Sonoff devices are using Tasmota version 6.1.1b, core 2.30, MQTT_ESPMQTTARDUINO, self compile by VSC.

symptom:

Sonoff devices can reconnect to wifi router.
Sonoff devices webserver can use normally
Sonoff devices are no response to MQTT server on my Pi 3 B+. (no mqtt reconnection happen)
restart Sonoff devices-> all will back to normal function as expect.
Important note:  above issue will happen if i am using option WifiConfig 5

10:23:04 CMD: status 0
10:23:04 MQT: stat/mypc/STATUS = {"Status":{"Module":1,"FriendlyName":["My PC"],"Topic":"mypc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:23:04 MQT: stat/mypc/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"5T15:17:53","StartupUTC":"2018-07-18T12:05:11","Sleep":0,"BootCount":4,"SaveCount":9,"SaveAddress":"FB000"}}
10:23:04 MQT: stat/mypc/STATUS2 = {"StatusFWR":{"Version":"6.1.1b","BuildDateTime":"2018-07-18T17:42:33","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:23:04 MQT: stat/mypc/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Multimedia Wi-Fi Network","Wong"],"TelePeriod":300,"SetOption":["00008009","55A18000"]}}
10:23:04 MQT: stat/mypc/STATUS4 = {"StatusMEM":{"ProgramSize":417,"Free":584,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F083380","0C000000","00000096","00000000"]}}
10:23:04 MQT: stat/mypc/STATUS5 = {"StatusNET":{"Hostname":"mypc-6980","IPAddress":"192.168.12.148","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.12.1","Mac":"68:C6:3A:95:9B:44","Webserver":2,"WifiConfig":5}}
10:23:04 MQT: stat/mypc/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_959B44","MqttUser":"admin","MqttType":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:23:04 MQT: stat/mypc/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 24 03:23:04 2018","Local":"Tue Jul 24 10:23:04 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":7,"Sunrise":"05:40","Sunset":"18:18"}}
10:23:04 MQT: stat/mypc/STATUS10 = {"StatusSNS":{"Time":"2018-07-24T10:23:04"}}
10:23:04 MQT: stat/mypc/STATUS11 = {"StatusSTS":{"Time":"2018-07-24T10:23:04","Uptime":"5T15:17:53","Vcc":3.174,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":62,"APMac":"B0:C7:45:36:85:E0"}}}

wongnam, 2018-07-27:
TASMOTA develop many function in FW but it's still unstable. some function very simple for example "Wifi connection", "MQTT connection" it's still having problem so far.

nosaj66au, 2018-07-27:
Hi Wongnam,
   I agree I'm having a similar issue. I have about 24 devcies (16) H801 and (8) Wemos D1R2. the H801 are running 5.12.0i and the D1;s 6.0.0a. Never had a problem with the H801's but the D1's just stop producing MQTT but the web server has no problem. I had suspension it was something wifi related as the 2 with the lowest wifi signal would drop off most frequently. I have been monitoring the wifi strength and the issues would occur after a fall in signal to below 50%, however this is inconsistent.
As the Web server is still running fine i use the console to investigate and everything appears normal.
   If i running MQTTfx all other devices on the network are good but the LWT for the effected D1 is offline and no MQTT data. To fix the device, I reboot from the HTTP interface but within 24 hours its dead again.
I really need a fix for this it is driving me crazy not to mention the wife
Jason

Frogmore42, 2018-07-27:
If you search the issues list in the esp8266 arduino repo you will find MANY related to WiFi.  Tasmota uses that code, which in turn is based on the code/SDK from espressif.  All of the released libraries have some issues.  Some of the issues are so severe they make it unusable, but only under certain conditions.  There is a new release of the core scheduled for August 1.  It will likely take awhile before there is the ability to use it with Tasmota.  Hopefully, 2.4.2 will be better than 2.4.0, which is worse than 2.3.0.

wongnam, 2018-07-27:
@Frogmore42 I think you've read the article not carefully. We said that the web server in TASMOTA was reachable(even Alexa can control the sonoff normaly) so it's not related to wifi communication issue as you mention about. please do not assume Wifi issue here.
if I am wrong please let me know the relative of wifi issue and MQTT issue if you think it's due to bug in the espressif core lib.

@nosaj66au you are right, you descripted the exactly the same issue that my sonoff devices are facing.
> quote:
 "As the Web server is still running fine i use the console to investigate and everything appears normal.
If i running MQTTfx all other devices on the network are good but the LWT for the effected D1 is offline and no MQTT data. To fix the device, I reboot from the HTTP interface but within 24 hours its dead again."

Frogmore42, 2018-07-27:
My comment was in reference to the comment about WiFi instability.

But, the issues with WiFi, which is networking can indeed impact MQTT.  You are correct that since the web console is working it seems WiFi is fine and it might actually be fine.  You might also be right that there is a problem with MQTT not reconnecting properly.  If so, that is probably caused by the underlying MQTT code, but could be due to the way Tasmota is using it.

It could be that once WiFi is reliable, the MQTT issue will become nearly impossible to hit.

I have never had an issue with MQTT not connecting.  That doesn't mean there isn't an issue.  It could be the changes in 6.x make it more likely to occur.

What I am saying is that since there are know strange issues with WiFi that could be causing the problem and there is a release of the core scheduled soon, it is like best to wait till that is integrated and see if it addresses the issue.  If it doesn't, then someone will need to take the time to understand what exactly is failing and why.  I doubt that would be a simple task.

Since you are using a non default mode for WiFi (no restart) it is possible that the mqtt code only tries once.  If it fails, no more mqtt.  Other people wouldn't see this, since when WiFi goes out their devices restart.

It is easy to complain about the quality of something.  It is much harder to do something constructive to improve it.

nosaj66au, 2018-07-28:
OK Guys Checked my notes from when compiling 6.0.0.a and found I used the Tasmota alternative MQTT lib, however when compiling the 5.12.0i I didn't. So now have gone back to 5.14 release with the original lib. I'll run this for a few day see what happens. I did have a case today with a MQTT dropout at 80% wifi signal. The Unifi software did not report loss of device. So I'm rethinking some of my earlier Wifi theories.

Jason

nosaj66au, 2018-07-30:
I have been running well over 24hours now without a failure. I not doing to call it yet but so far so good.

uberflyx, 2018-07-30:
I want to add that I have the same issue. As of the previous version where we could select an alternative MQTT lib, after a few days I find my devices cant be switched via MQTT until a reboot.

I have tried TASMOTA_MQTT and well as the other ESP mqtt one. Both have the same issues. Of course, pubsubclient works well, but sucks because of the switch delay wich is untenable as I have sonoffs wired as all my house lights.

I was unable to determine what causes this issue except that it is inevitable after a few days. I think that it's possible my wifi router reboots itself / drops the signal. Perhaps the interrupt for wifi reconnect isn't reliable and so mqtt doesn't reconnect properly? Here's my status 0 of one that is currently not responding:

04:23:03 CMD: status 0
04:23:03 MQT: stat/bedrooms/lights/STATUS = {"Status":{"Module":18,"FriendlyName":["Man Cave","Gills Room"],"Topic":"bedrooms/lights","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
04:23:03 MQT: stat/bedrooms/lights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"10T09:25:40","StartupUTC":"2018-07-19T16:57:23","Sleep":1,"BootCount":107,"SaveCount":705,"SaveAddress":"F4000"}}
04:23:03 MQT: stat/bedrooms/lights/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-20T11:30:35","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
04:23:03 MQT: stat/bedrooms/lights/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"192.168.0.6","LogPort":514,"SSId":["Ubernet","indebuurt2"],"TelePeriod":300,"SetOption":["00000008","55818000"]}}
04:23:03 MQT: stat/bedrooms/lights/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
04:23:03 MQT: stat/bedrooms/lights/STATUS5 = {"StatusNET":{"Hostname":"bedrooms/lights-5653","IPAddress":"192.168.0.30","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.7","Mac":"5C:CF:7F:18:D6:15","Webserver":2,"WifiConfig":3}}
04:23:03 MQT: stat/bedrooms/lights/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.6","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_18D615","MqttUser":"home","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
04:23:03 MQT: stat/bedrooms/lights/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 30 02:23:03 2018","Local":"Mon Jul 30 04:23:03 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"06:47","Sunset":"17:40"}}
04:23:03 MQT: stat/bedrooms/lights/STATUS10 = {"StatusSNS":{"Time":"2018-07-30T04:23:03","Switch1":"OFF","Switch2":"OFF"}}
04:23:03 MQT: stat/bedrooms/lights/STATUS11 = {"StatusSTS":{"Time":"2018-07-30T04:23:03","Uptime":"10T09:25:40","Vcc":3.160,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Ubernet","RSSI":84,"APMac":"10:62:EB:2C:99:xx"}}}

I'm pretty sure I've also tested with both Arduino version 2.3 and 2.4.1 but I can't remember.. Perhaps my wifi router is overloaded haha

wongnam, 2018-07-30:
@uberflyx the issue is due to FW does not reconnect to MQTT broker sometimes until it is restart.
It will not connect to the mqtt server if it has had a disconnection with the wifi signal. However it can be connected to wifi but mqtt is not reconnected.
@arendst Recommended. I would appreciate if you could consider the solution for it to be able to reconnect to mqtt.

nosaj66au, 2018-07-31:
Me again,
    Well 72 hours with the pubsubclient MQTT Lib and all is well.  But the switch delay is now the problem, The family have got conditioned to the speed of the TASMOTA MQTT lib. Guess I can't win.
Jason

wongnam, 2018-07-31:
Hi Jason and All,

These are the experiences I've had before with the TASMOTA fw configuration as below:
- Pubsubclient MQTT -> If you lose connection with MQTT it will have about 10s lag phenomenon. (@nosaj66au Jason, your device will experience this problem.) so i have to use the alternative lib to prevent lag phenomenon happen, eg TASMOTA MQTT or MQTT_ESPMQTTARDUINO. 
- WifiConfig 3 -> when the Wifi connection is lost it will automatically restart the device, it's very helpful to reconnect to wifi and MQTT server but it is a disadvantage as I am saying below
- When the device is in "ON" state - in this state, if the device is restarted as described above, the relay will flicker rapidly causing the load to interrupt suddenly. Therefore if i want to avoid device self restart i have to use WifiConfig 5.
- Wificonfig 5(no restart) -> As @Frogmore42 said that "WiFi (no restart) it is possible that the mqtt code only tries once. If it fails, no more mqtt. "

Do you see this is the endless loop that I have experienced through.
I would appreciate if TASMOTA FW can able to reconnect to mqtt server with no need to restart the device is prefect.

Thanks.

wongnam, 2018-08-01:
@ascillato I use MQTT Broker and Node-red in the same Pi 3B+.

ascillato, 2018-08-01:
Ok

Let's try this to know what is happening in your case. 

Using wificonfig 5,

* Disconnect your wifi
* connect your router
* Wait few seconds
* open your Tasmota console
* toggle your output

And share your console log. 

If Tasmota is not connected to the MQTT broker, when toggling it should say RSL:, if connected it should say MQT:

wongnam, 2018-08-01:
- Disconnect your wifi -> wifi of which device? Sonoff or Access Point.
- connect your router -> Are you saying Wifi Access Point?

ascillato, 2018-08-01:
Disconnect your wifi -> turn off your router
connect your router - > turn on your router

wongnam, 2018-08-01:
Here you are: (i am out of my house, i use VPN to perform as you request. i can not touch the button yet.)
10:42:49 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:42:49","Uptime":"0T00:13:23","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
10:42:49 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:42:49","AM2301":{"Temperature":31.4,"Humidity":72.3},"TempUnit":"C"}
10:43:49 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:43:49","Uptime":"0T00:14:23","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
10:43:49 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:43:49","AM2301":{"Temperature":31.4,"Humidity":72.4},"TempUnit":"C"}
10:44:49 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:44:49","Uptime":"0T00:15:23","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
10:44:49 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:44:49","AM2301":{"Temperature":31.3,"Humidity":72.4},"TempUnit":"C"}
10:45:00 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
10:45:19 WIF: Connect failed with AP timeout
10:45:38 WIF: Connect failed with AP timeout
10:45:39 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
10:45:49 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:45:49","Uptime":"0T00:16:23","Vcc":3.161,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":26,"APMac":"38:17:C3:F1:90:00"}}
10:45:49 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:45:49","AM2301":{"Temperature":31.4,"Humidity":72.3},"TempUnit":"C"}
10:45:57 WIF: Connected
10:45:57 DNS: Initialized
10:45:57 UPP: Multicast (re)joined
10:46:49 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:46:49","Uptime":"0T00:17:23","Vcc":3.162,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":36,"APMac":"38:17:C3:F1:90:00"}}
10:46:49 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:46:49","AM2301":{"Temperature":31.4,"Humidity":72.4},"TempUnit":"C"}
10:47:57 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:47:57","Uptime":"0T00:18:31","Vcc":3.161,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":38,"APMac":"38:17:C3:F1:90:00"}}
10:47:57 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:47:57","AM2301":{"Temperature":31.3,"Humidity":72.3},"TempUnit":"C"}
10:49:03 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:49:03","Uptime":"0T00:19:37","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":32,"APMac":"38:17:C3:F1:90:00"}}
10:49:03 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:49:03","AM2301":{"Temperature":31.4,"Humidity":72.3},"TempUnit":"C"}

wongnam, 2018-08-01:
Lost MQTT, No LWT send to inform Online yet.

ascillato, 2018-08-01:
The log shows that your device is connected to your mqtt broker again.

Your issue is that your node-red is not updating the LWT information.

Exactly the same was happening to me but in home assistant. So, I add an automation in my Home Assistant to ask again for status to the grouptopic of all my Tasmotas (cmnd/sonoffs/POWER)

wongnam, 2018-08-01:
@ascillato I did the same as your idea so far but no MQTT response back from devices. 
eg: i send PowerOnState to cmnd/smartplug/POWER but to response from device.
even i send ON of OFF to cmnd/smartplug/POWER -> device no action at all.

wongnam, 2018-08-01:
@ascillato I would like to perform the test one more time(as the last time the device auto roam to my upstair 2nd AP), so i disable the Radio of my up stair 2nd AP now, i use only 1 AP from my router. Result: MQTT still lost after turn off/on router AP.

Here is the log:
`00:00:00 Project sonoff Smart Plug (Topic smartplug, Fallback DVES_0F86C9, GroupTopic sonoffs) Version 6.1.1c-2_3_0
00:00:00 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on smartplug-1737.local with IP address 192.168.12.123
00:00:03 UPP: Multicast (re)joined
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/smartplug/LWT = Online (retained)
00:00:05 MQT: cmnd/smartplug/POWER = 
00:00:05 MQT: tele/smartplug/INFO1 = {"Module":"Sonoff TH","Version":"6.1.1c","FallbackTopic":"DVES_0F86C9","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/smartplug/INFO2 = {"WebServerMode":"Admin","Hostname":"smartplug-1737","IPAddress":"192.168.12.123"}
00:00:05 MQT: tele/smartplug/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: stat/smartplug/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/smartplug/POWER = ON
00:00:05 UPP: Multicast (re)joined
00:00:06 MQT: stat/smartplug/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/smartplug/POWER = ON
11:43:01 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T11:43:01","Uptime":"0T00:00:14","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
11:43:01 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T11:43:01","AM2301":{"Temperature":31.7,"Humidity":70.0},"TempUnit":"C"}
11:43:22 MQT: stat/smartplug/STATUS = {"Status":{"Module":4,"FriendlyName":["Smart Plug"],"Topic":"smartplug","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:43:22 MQT: stat/smartplug/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:35","StartupUTC":"2018-08-01T04:42:47","Sleep":0,"BootCount":15,"SaveCount":53,"SaveAddress":"F8000"}}
11:43:22 MQT: stat/smartplug/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-07-31T10:57:42","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:43:22 MQT: stat/smartplug/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Multimedia Wi-Fi Network","Wong"],"TelePeriod":60,"SetOption":["00008009","55A18000","00000001"]}}
11:43:22 MQT: stat/smartplug/STATUS4 = {"StatusMEM":{"ProgramSize":426,"Free":576,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F083390","0C000000","00000096","00000000"]}}
11:43:22 MQT: stat/smartplug/STATUS5 = {"StatusNET":{"Hostname":"smartplug-1737","IPAddress":"192.168.12.123","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.12.1","Mac":"2C:3A:E8:0F:86:C9","Webserver":2,"WifiConfig":5}}
11:43:22 MQT: stat/smartplug/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F86C9","MqttUser":"admin","MqttType":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:43:22 MQT: stat/smartplug/STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 01 04:43:22 2018","Local":"Wed Aug 01 11:43:22 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":7,"Sunrise":"05:41","Sunset":"18:17"}}
11:43:22 MQT: stat/smartplug/STATUS10 = {"StatusSNS":{"Time":"2018-08-01T11:43:22","AM2301":{"Temperature":31.7,"Humidity":70.0},"TempUnit":"C"}}
11:43:22 MQT: stat/smartplug/STATUS11 = {"StatusSTS":{"Time":"2018-08-01T11:43:22","Uptime":"0T00:00:35","Vcc":3.161,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}}
11:44:15 WIF: Connect failed as AP cannot be reached
11:44:16 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:44:23 WIF: Connect failed as AP cannot be reached
11:44:23 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:44:30 WIF: Connect failed as AP cannot be reached
11:44:30 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:44:37 WIF: Connect failed as AP cannot be reached
11:44:37 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:44:44 WIF: Connect failed as AP cannot be reached
11:44:44 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:44:51 WIF: Connect failed as AP cannot be reached
11:44:51 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:44:58 WIF: Connect failed as AP cannot be reached
11:44:59 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:45:06 WIF: Connect failed as AP cannot be reached
11:45:06 WIF: Connecting to AP1 Multimedia Wi-Fi Network in mode 11N as smartplug-1737...
11:45:06 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T11:45:06","Uptime":"0T00:02:19","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
11:45:06 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T11:45:06","AM2301":{"Temperature":31.7,"Humidity":69.9},"TempUnit":"C"}
11:45:25 WIF: Connect failed with AP timeout
[... cut ...]
11:46:06 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T11:46:06","Uptime":"0T00:03:19","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
11:46:06 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T11:46:06","AM2301":{"Temperature":31.7,"Humidity":69.9},"TempUnit":"C"}
11:47:09 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T11:47:09","Uptime":"0T00:04:22","Vcc":3.161,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
11:47:09 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T11:47:09","AM2301":{"Temperature":31.7,"Humidity":69.9},"TempUnit":"C"}`

nosaj66au, 2018-08-02:
Hi again,
     Similar thing was happening to me running the alternative TASMOTA lib LWT was indicating offline, however I know this was not the case. I have been using the original MQTT lib 5 days not a single drop out.  I have tried a number of tests include power cycling the router , not a problem. Dame that delay!!!
Jason

wongnam, 2018-08-02:
Yep, pubsubclient MQTT Lib has limited issue

webxl, 2018-08-02:
@wongnam How did you create that page?

wongnam, 2018-08-03:
@webxl your comment is not related to my topic.
BTW, I created it before i know that there is TasmoAdmin present in the world and it can do the better job then my self create page by Node-red Dashboard.
FYI: https://github.com/reloxx13/TasmoAdmin

ascillato2, 2018-08-22:
@wongnam 

Have you find any solution? have you tried other MQTT library?

wongnam, 2018-08-22:
@ascillato2
I have tried 3 libraries
1- Tasmota MQTT -> Mqtt reconnecting issue
2- Espmqttarduino MQTT -> Mqtt reconnecting issue 
3- PubSubClient MQTT -> MQTT reconnecting successful ( cons: it will cause the device lag few seconds in case of mqtt connection temporary lost)

ascillato, 2018-08-23:
@wongnam

Your findings are very useful to search the bug. Thanks for share.

wongnam, 2018-08-23:
I’ve just recompiled as your advice. I test it now, will revert you back for the result.

Thanks

wongnam, 2018-08-23:
@ascillato I can confirm that the issue is gone and all my Tasmota devices are stick to mqtt broker now.

Thank you.

closed 2018-08-23 by wongnam