arendst/Tasmota #3317: After Upgrade from 5.9.1 to 6.1.1 via Web no reaction.

SeriousOri, 2018-07-24:
Hello
I have updated the SonOff Basic via the web interface.
"Upgrade by web server"
Now he does not work anymore.
It was previously Program Version 5.9.1.
I wanted to flash the v6.1.1.
Can I flash that again, so it goes again, or is it dead?

meingraham, 2018-07-24:
@SeriousOri 

If you are compiling your own binaries, please take a look at the recent tracks regarding 6.1.x and compatibility with particular ESP core versions. The consensus is that 2.3.0 is the most (only) reliable version to use.

Having said that, when I was having issues and my device would not connect to Wi-Fi (i.e., could no longer OTA flash), then reverting to wired (serial) flashing worked without issue - i.e., the device was not bricked.

Mike

SeriousOri, 2018-07-24:
hello @ meingraham
No did not compile it yourself. 
I have no idea and do not know myself so well.
I thought it is about the web interface.
I just pressed Start Ugrade. Now nothing flashes.
I have to buy such a Seriall Flasher now. Thank you

meingraham, 2018-07-24:
@SeriousOri 

You must have had Sonoff devices with factory firmware versions earlier/older than v1.6. Newer versions of the Sonoff firmware no longer allow OTA flashing out of the box. I was not so fortunate; my Sonoff devices have all had newer factory firmware. This means that I had no choice but to begin with "wired" flashing. Now that I have TASMOTA on the devices, I am able to update their firmware OTA (wirelessly).

I have not used web server based OTA. I upgrade using a local firmware file via the second option on the web admin interface. At one point I did have problems successfully completing a firmware upload. This turned out to a firewall issue of some sort. I switched to a different computer and have not had any more issues. In my situation, when an upload failed, the old firmware was still present and the device would reconnect. You don't appear to be so fortunate.

I think, perhaps, in your case, your attempted TASMOTA OTA blew away the factory firmware environment or configuration. Try pairing the device with eWeLink again.
- If you do get it to pair and connect:
1. Be sure to NOT let it upgrade the firmware. Otherwise it will install a version that will not allow OTA flashing.
2. Try to OTA with TASMOTA sonoff_minimal.bin.
3. If you successfully flash the minimal image, connect it and then flash the "final" TASMOTA image you want.

My thinking is that the MINIMAL image may be able to load successfully in the available free program space remaining with the factory image. Then you'll have a stable "platform" from which you can flash TASMOTA from this point forward.

If not, then you'll have to take the plunge and learn how to flash TASMOTA via a wired serial interface. Fortunately, this should be a one time action (per device). Once you have TASMOTA flashed, then you'll get OTA capability (as long as the image size remains small enough... which so far it is).

Mike

SeriousOri, 2018-07-24:
Now that I have TASMOTA on the devices, I'm ready to update their firmware OTA (wirelessly).
Hi @ meingraham
My English is not good.

I had Tasmota firmware V5.9.1 on it. I have OTA on v6.1.1. tried and it did not work.

I had previously tried Upgrade by file upload. That did not work either.
[image]

meingraham, 2018-07-24:
@SeriousOri 

> Now that I have TASMOTA on the devices

How did you get TASMOTA on there originally?

Can you get the device to connect at all now and access the web admin? If so, try the "Upgrade by file upload" from a different computer on your home network. As I mentioned, I had some kind of firewall issue on one computer (even though it looked like it was set up to pass all traffic on my LAN). Switching to a different computer solved my "upload by file" problem.

Or, if you have not tried this, upload the minimal image first and then come right back and upload your "full featured" image.

Mike

dojf, 2018-07-24:
Just a short note of my experience: 
I have about 10 Sonoff's working (basic, pow, s20) and had them flashed a couple of months ago with Tasmota 4.x using PlatformIO, see [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite). 

Last week I tried to OTA update them to 6.1.1 in the multistep method as [described](https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade#migration-path). But only two succeeded that way, the others didn't complete the OTA update and weren't reachable anymore. Dunno what's the cause, but it just happened and with different effects.

So I saw no other way then to have them flashed again non-OTA, just as in the original instruction ([wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite)). 
So now they are back again :-).

I'm afraid that if they aren't reachable then you have to flash them the original way.

ascillato2, 2018-07-25:
Hi,

Have you managed to solve your issue by flasing by wire?

SeriousOri, 2018-07-25:
Hi @ meingraham

> How did you get TASMOTA on there  originally?

I flashed it with sonota.

Today I tried to flash it with Wire. No reaction.
I've tried Atom, Arduino and NodeMCU-PyFlasher-3.0-x64.
No reaction. Is probably dead.

SeriousOri, 2018-07-25:
Hi @ meingraham

> How did you get TASMOTA on there  originally?

I flashed it with sonota.

Today I tried to flash it with Wire. No reaction.
I've tried Atom, Arduino and NodeMCU-PyFlasher-3.0-x64.
No reaction. Is probably dead.
Hi@ascillato2

Unfortunately, no. If you know a good method, you can show it to me.

closed 2018-07-25 by SeriousOri

ascillato, 2018-07-25:
Sorry, I just flashed as explained in the wiki. If your device did not respond could be a power problem, or if you mix VCC with GND it is probably burnt. Sorry.

dojf, 2018-07-25:
@ascillato2  Yes, flashed over the wire using a serial programmer as the wiki explained. All Sonoff's are now flashed to 6.1.1.

SeriousOri, 2018-07-25:
Hi@ dojf
Hi what program exactly do you mean

SeriousOri, 2018-07-25:
@dojf 
Thanks. 
I will try again