arendst/Tasmota #3318: WiFi Smart Config - not working on T1

chietanjain, 2018-07-24:
After I set the SSID to a new SSID using the COMMAND ssid1 NewWiFiSSID, the sonoff T1 (3) is not connecting to the wifi (the password for the newly set ssid and the old ssid are the same), neither is 3 rapid button presses enabling WiFi Smart Config. When using 4 button presses the T1 exposes an AP but I am unable to connect to it. Before I can connect to it, it disappears. 5 button presses and 40 second hold are also not working.

Is there any other way I can reset the settings to default. Is there something I am doing wrong.

Is there any use of the reset button on the T1 (its marked S1).

The last option is to reflash it, but I don't have the serial UART with me and it will take a while before I can get one.

Thanks in advance!

chietanjain, 2018-07-24:
Note that I haven't enabled single button press for quick response.

Frogmore42, 2018-07-24:
It sounds like your only option is to try serial flash.  You have already tried all the things you can except that.  You might have downloaded or built a bad binary.

chietanjain, 2018-07-25:
Thanks! Did that! and replaced the CFG_Holder with a new one by reflashing twice once with new CFG Holder and then back to default CFG Holder.

Its fine now, but I have 4 sonoff basics which aren't connecting to wifi after an OTA upgrade to 6.1.1. Will raise this as a sperate issue.

Thanks.

ascillato2, 2018-07-25:
If OTA has failed, I'm sorry to say that you need to flash by wire. Sorry.

As you have stated that you solved this issue, I'm closing this in order to track issues. Thanks.

closed 2018-07-25 by ascillato2