arendst/Tasmota #3319: "Reset" additionally to "Toggle"

HugoHiasl, 2018-07-24:
I think it would be helpful in some cases to have the software not only being able to "toggle" the power status or to switch it on or off.

Imagine the switch is in a remote location to be able to restart some logging devices for solar plants or other devices.  To be able to use logging you provide at the external location a router with internet connectivity and WLAN to which the sonoff is connected.

Sometimes you want the router to be restarted. As soon as you toggle it "OFF" the sonoff will not be able to get the command for the "ON" via internet since it switched internet off.

A reset function would help. Switch OFF, wait 10 seconds and switch ON again would solve that.

Maybe that is something that can go into the software. It would help me a lot.

Thanks.

ascillato, 2018-07-24:
Hi,

All that is already implemented.

Please read the wiki at commands. Thanks

ascillato2, 2018-07-25:
You can use a rule like:

_(type in the console)_

```
rule 1
rule on event#makereset do backlog power off; delay 100; power on endon
```

and call that event:

```
by mqtt: `cmnd/sonoff/event makereset`
by http: `http://sonoff_IP/cm?cmnd=event%20makereset`
```

If this solves your question, please close this issue. Thanks.

ascillato2, 2018-07-28:
Closing this issue as there is no feedback. Please, ask to reopen if you need anything else. Thanks

closed 2018-07-28 by ascillato2