arendst/Tasmota #3320: Sonoff 4Ch R2 - Unable to Flash - Espcomm_sync failed

msgowdara, 2018-07-24:
I'm trying to Flash Sonoff 4Ch R2 V1.0.
My connections as below.

GND-GND
TX-RX
RX-TX
3.3V-3.3V

Using Arduino IDE with board version 2.4.1.
Entering properly to flash mode (no blink d6).
Tried Generic esp8266 and esp8285 board.

Flash size- 1M (no SPIFFS)
Flash mode-DOUT
IwIP variant - v1.4 Higher Bandwidth
Reset method - Nodemcu
Upload speed - 115200

But at the end shows error shows
espcomm_sync failed espcomm_open failed
espcomm_upload_mem failed

........

Kindly help.

Frogmore42, 2018-07-24:
Please read the wiki and search the issues.  There are a lot of people who run into this issue.  There are a lot of troubleshooting hints in the other issues and the wiki.

The error you are seeing is caused by one or more of:

1. Not enough power (the FTDI is not rated to supply enough power to flash, but it sometimes works)
2. The connection is not correct.  Try reversing Rx an tx.
3. The esp8266 is not in bootloader mode
4. You have the wrong com port selected

Since you are getting the failure right away #2,3, or 4 are the likely ones.  Or the board is not powered at all.

meingraham, 2018-07-24:
Sometimes a lower baud rate helps if it's noise that's causing the comms failure. I had these errors and lowering the baud rate resolved them.

Frogmore42, 2018-07-24:
115,200 is the default and pretty slow.  What baudrate do you use?

I typically use 500ish K with wemos devices, which is why I like them.  I find 115.2K to be really slow.  I can't imagine how slow something slower would seem.

meingraham, 2018-07-24:
57.6K - yeah crawling. Actually, now that you mention it, I've gone both down and up on the baud rate to get some upload love.

P.S. 115,200 slow... I once worked with 110 baud. You've come a long way baby ;-)

msgowdara, 2018-07-24:
Thanks for immediate reply.

I can operate the board with pre-installed Sonoff software while connected to USB. But, during flash it says
error:cannot access /dev/ttyUSB0

espcomm_open failed
espcomm_upload_mem failed
and so on.

Is it power issue as @Frogmore42 suggested?

Frogmore42, 2018-07-24:
That error likely means you have the wrong port, since it can't open it.  This is the port the USB serial adapter is enabling on your computer.  Not sure how you find out the correct port on Linux.

meingraham, 2018-07-24:
It could be any of the things suggested, including insufficient power.

RX/TX - double check your FTDI pin-out. The FTDI TX should go to the Sonoff RX contact and vice versa for the FTDI RX.

COM port - What operating system on your computer? If Windows, check the Device Manager and see if your FTDI device shows up and on which port. See if it shows as operating properly (as far as Windows thinks). If you're running on Mac or Linux... someone else will have to give you additional guidance as I am not well versed at all.

Insufficient power - the 4CH has a separate DC power jack. I use that when flashing mine. I ground GPIO 0 and then plug in the power supply. I then connect my serial patching cables and initiate the upload.

Bootloader mode - My experience is only with a 4CH Pro (rev 1). But, if the power LED (blue on my devices) comes on or blinks or flashes at all even for a moment, then the device is not in programming mode. I have had a couple of my 4CH Pro switches light up their relay LEDs while in programming mode, but the power LED has not lighted. The takeaway - make sure you power up your 4CH while GPIO 0 is grounded and that your power LED stays off.

Baud rate - play around with different selections.

You'll have to eliminate the possibilities in a methodical troubleshooting approach.

Mike

msgowdara, 2018-07-24:
Hello,
How to give external power to sonoff 4ch?

Frogmore42, 2018-07-24:
http://ewelink.coolkit.cc/?p=458
Is the picture of the R2.  You can provide power via the barrel jack.

I doubt that is the problem, but it could be.

You first need to see that you have things connected correctly and you need to be very precise with your replies.  We can't see what you can and we don't know how you have things hooked up and if you have ever successfully flashed something.  The first several times are hard if you have never done it before and even if you have can still be frustrating.

I know the first time I tried to flash an esp8266, I said why does it have to be so hard and I had flashed a lot of different things before.

While lots of people have problems, they break down into the classes of problems I listed above.  Get all of those right and it should and usually does just work.  But, get any one or more of them even a little bit wrong and it doesn't work at all.

You did not say what USB serial adapter you are using.  If you are using an FTDI based adapter and even if you are not, you should power the Pro with a DC adapter via the barrel jack.  That will ensure it has proper power.

Then, you need to make sure the serial communication is working correctly.  I use a terminal emulator to test this.  This makes sure I know which port to use and that I have the RX and TX lines correctly wired.  I use TeraTerm, but I believe putty is available for Linux and should work just as well.  Look for a tutorial on it if you don't know how to use it.  Once you have it hooked up and putty open, cycle power on the Pro.  You should see something, might look like garbage, since the baud rate will be wrong.

Now that you know the serial lines are connected and you know the com port to use, you can try flashing.

You will have to figure out how to specify the correct com port with the flashing tool.  I use PlatformIO in VSCode on Windows and it is smart enough to automatically find the port or I can specify it in one of the files.

msgowdara, 2018-07-24:
@meingraham 
"Insufficient power - the 4CH has a separate DC power jack. I use that when flashing mine. "

How to do that? Where is that separate DC Jack situated?

meingraham, 2018-07-24:
I misread your original post... you have a 4CH (not Pro). The 4CH Pro has the external power jack.

msgowdara, 2018-07-24:
To check all the things, I flashed my Sonoff basic. Everything went well.

The problem still persist with Sonoff 4ch R2 (not pro).

Frogmore42, 2018-07-24:
@msgowdara 
That doesn't check all things.  It does verify that you have the correct com port which is good.  That was item #4 in the list. Items #1, 2, or 3 could still be wrong.

What USB serial adapter are you using?
Take pictures of how you have things hooked up.

How are you putting it into bootloader mode?

Have you verified you can get serial output from the 4ch (not Pro) R2, using a terminal emulator?

It is possible that you 4ch is defective, but I would try the above things before giving up on it.

ascillato2, 2018-08-18:
Hi,

Closing this issue as there is no feedback. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2

jhnnbe, 2019-01-20:
Hi do you finaly flach your 4ch R2 ?
I've the same Sonoff 4ch R2 V1.0    NOT THE 4 CH PRO

I have everything checked:
Cross the rx & tx and not cross
I check the voltage of my 3.3 volt and it is good.
I check the com ports, try several speeds.
My usb uses the prolific driver.

Nothing is done, no way to flash it.

I had no problem to flash my other sonoff.

tr1plus, 2019-01-20:
@jhnnbe I managed to flash it. Made a youtube video. - https://www.youtube.com/watch?v=VZVJTD6tq2A
(Hope links are allowed, sorry if not)

jhnnbe, 2019-01-20:
I show you video yesterday, i'va download all, firmware , tools, but the sonoff don't power in flash mode.
I will retry it this evenig.

jhnnbe, 2019-01-20:
@tr1plus 
just for info, you use this settings in nodemcu flacher : 

Baudrate      115200
Flashsize      1 Mb
Speed SPIO  80Mhz
Mode           DOUT

Do you cross tx & rx ?

nvisible29, 2019-07-16:
@jhnnbe I struggled with with the same issue. In fact, my sonoff 4ch v2 (not pro) was not in flash mode. To get it done, you have to keep GPIO0 button down (bottom one) when powering the device. Just to make sure it works, I kept it for another 7s, but I'm sure it will work right away. Power button didn't light up or blink, but NodeMCU 8266 flasher successfully connected and flashed my device.

Settings are the same as yours. RX &TX crossed. 

Cheers!