arendst/Tasmota #3322: timestd & timedst return invalid JSON

jcheger, 2018-07-24:
TimeSTD and TimeDST return an invalid JSON string: « "Offset":xx" ». The last quote is invalid.

Bug confirmed to appear on fw 5.12 and 6.6.1

    20:42:53 MQT: tasmota/anet/stat/RESULT = {"TimeStd":{"Hemisphere":0,"Week":0,"Month":10,"Day":1,"Hour":3,"Offset":60"}}
    20:42:53 MQT: tasmota/anet/stat/RESULT = {"TimeDst":{"Hemisphere":0,"Week":0,"Month":3,"Day":1,"Hour":2,"Offset":120"}}

ascillato, 2018-07-24:
Hi,

I could not reproduce that issue.

What do you mean with invalid? If I change the values as explained in commands in the wiki, those values change accordingly.

ascillato, 2018-07-24:
Ok.

I could reproduce it. @arendst 

is `"Offset":60"` and should be `"Offset":60`

Thanks for pointing this out :+1:

ascillato2, 2018-07-25:
Tested ok!

Thanks :+1:

closed 2018-07-25 by ascillato2