arendst/Tasmota #3324: Problem booting 6.1.1

robygi, 2018-07-24:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X ] Hardware used : Sonoff basic
- [X ] Provide the output of command ``status 0`` :
```
22:03:20 CMD: status 0
22:03:20 MQT: switch01/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"switch01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:03:20 MQT: switch01/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://Roberto-NBDell:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:44","StartupUTC":"2018-07-24T21:02:36","Sleep":0,"BootCount":3,"SaveCount":8,"SaveAddress":"F4000"}}
22:03:20 MQT: switch01/stat/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-07-24T22:58:57","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
22:03:20 MQT: switch01/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"192.168.1.121","LogPort":514,"SSId":["MyHome_260760","MyHome_260760"],"TelePeriod":120,"SetOption":["00008009","55818000"]}}
22:03:20 MQT: switch01/stat/STATUS4 = {"StatusMEM":{"ProgramSize":424,"Free":576,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","05002790","0C000001","00000096","00000000"]}}
22:03:20 MQT: switch01/stat/STATUS5 = {"StatusNET":{"Hostname":"switch01-5307","IPAddress":"192.168.1.154","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"5C:CF:7F:3C:14:BB","Webserver":2,"WifiConfig":2}}
22:03:20 MQT: switch01/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.121","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C14BB","MqttUser":"mqttrgusr","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:03:20 MQT: switch01/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 24 21:03:20 2018","Local":"Tue Jul 24 22:03:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
22:03:20 MQT: switch01/stat/STATUS10 = {"StatusSNS":{"Time":"2018-07-24T22:03:20","DS18B20":{"Temperature":40.0},"TempUnit":"C"}}
22:03:20 MQT: switch01/stat/STATUS11 = {"StatusSTS":{"Time":"2018-07-24T22:03:20","Uptime":"0T00:00:44","Vcc":3.541,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyHome_260760","RSSI":58,"APMac":"9C:5C:8E:C5:CE:A0"}}}

20:34:05 CMD: status 0
20:34:05 MQT: switch02/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"switch02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:34:05 MQT: switch02/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://Roberto-NBDell:80/api/arduino/sonoff.ino.bin","RestartReason":"Power on","Uptime":"1T01:51:25","StartupUTC":"2018-07-23T17:42:40","Sleep":0,"BootCount":8,"SaveCount":28,"SaveAddress":"F9000"}}
20:34:05 MQT: switch02/stat/STATUS2 = {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-21T09:42:10","Boot":4,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
20:34:05 MQT: switch02/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"192.168.1.121","LogPort":514,"SSId":["MyHome_260760","MyHome_260760_U"],"TelePeriod":120,"SetOption":["00008009","55818000"]}}
20:34:05 MQT: switch02/stat/STATUS4 = {"StatusMEM":{"ProgramSize":479,"Free":524,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
20:34:06 MQT: switch02/stat/STATUS5 = {"StatusNET":{"Hostname":"switch02-4436","IPAddress":"192.168.1.151","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"5C:CF:7F:3C:11:54","Webserver":2,"WifiConfig":2}}
20:34:06 MQT: switch02/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.121","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3C1154","MqttUser":"mqttrgusr","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:34:06 MQT: switch02/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 24 19:34:06 2018","Local":"Tue Jul 24 20:34:06 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
20:34:06 MQT: switch02/stat/STATUS10 = {"StatusSNS":{"Time":"2018-07-24T20:34:06","DS18B20":{"Temperature":37.7},"TempUnit":"C"}}
20:34:06 MQT: switch02/stat/STATUS11 = {"StatusSTS":{"Time":"2018-07-24T20:34:06","Uptime":"1T01:51:26","Vcc":3.513,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyHome_260760","RSSI":54,"APMac":"9C:5C:8E:C5:CE:A0"}}}

```
Hello,
I'm using a couple of Sonoff with Tasmota firmware.
Till version 5.12 everything was fine, non problem for flashing or for the updating via web and their behaviour was correct.
Then I upgraded one of the two Sonoff to version 6.1.1b. At first everything seemed working fine, but when I reboot the device (powering off and then powering on again) it doesn't boot, and I have to let the device unplugged a long time before powering it again: at that point the device boots and everything is fine.
The other Sonoff is still on 5.12.0m (as you can see from its status 0 output) and has no problem in booting quickly as soon as the power i connected: I can repeat several times the unpowering and powering cycle without failing a single boot.
I don't understand where the problem could be. I just flashed the release 6.1.1c, but the problem is still there.
Could you please help me?
TIA
Roberto

Frogmore42, 2018-07-24:
Both of your firmware builds are using core 2.4.1.  Look in the wiki on the section about cores.  2.3.0 is the one that is likely to provide the best results, at least until 2.5.0 gets released and then maybe it will be better.

Make sure you clear the flash, use the correct mode, and do it by serial.  That will probably help.

robygi, 2018-07-25:
Frogmore42, the Sonoff with 5.12.0m is on 2.4.1 since a lot of release ago, and it is working perfectly.
I also tried the 2.3.0 with the Sonoff on 6.1.1 but with the result wad the same.
I also changed the cfg_holder to write the config to flash.
I’ll try to reprogram the Sonoff via serial, having done it via web till now.
I’ll report the result as soon as possible.

Frogmore42, 2018-07-25:
I don't know why people find it hard to believe that something that is unstable will work fine sometimes but not at all other times.  That is the definition of unstable 😀.

Many people have been surprised to find that their problem went away when using 2.3.0.  It is not the cure for all problems, but since it is known to be bad sometimes, it is best to eliminate the possibility that it is causing the problem.

Frogmore42, 2018-07-25:
There are a few ways that one can get things into a confused state by flashing, using either an incorrect binary or flashing it incorrectly.  Normally, flashing will fail with an error when it is incorrect.  However, sometimes it will say success, but the firmware is not usable.  This can be very confusing.

You have not provided the log from boot up of the failure, so it is hard to say what might be the real issue.

robygi, 2018-07-26:
As I said in the previous post I tried 2.3.0, but the behaviour is the same.
You wrote "...you have not provided the log from boot up of the failure...", but it is not clear to me what are you meaning. The device doesn't boot, so (up to my knowledge) there is no log. Nothing is recorded in the syslog server. Is there something I did not understand?
I need to be more precise regarding "device doesn't boot".
Actually, if I unplug the 6.1.1 device and replug it after 4-5 seconds, the device after 3 minutes had not boot yet.
If I replug it after 2 minutes (or more) it boots immediately.
Between 4-5 seconds and 2 minutes the booting time decrease moving toward 2 minutes.
Put in this way it could seem a hardware problem, but the device was perfectly functioning on 5.12.0 before flashing it to the 6.1.1
Tomorrow I will have more time to try flashing it via serial as suggested and to make more tests.
Thank you for the support up to now.

Frogmore42, 2018-07-26:
You need to attach to the serial port and use a terminal emulator to capture the serial log.  That will indicate what is not working.

robygi, 2018-07-28:
**UPDATE**
After connecting via serial I was not able to reproduce the lack of booting (when connected via serial the device booted everytime, no matter of firmware or core versions).
When, after a lot of tests and flashings, I disconnected the serial and I tried to power the device by mains, the device didn't boot anymore.
So I found that something in the power supply circuitry had failed: when the device is powered by mains there is non 3.3V!
So, as it seemd from the beginnig, **it was a hardware problem**: something was failing, but sometimes it still worked!
I have no time at the moment to check and repair the failed device, so I flashed another one with 6.1.1c (and core 2.3.0, because I found the last device unstable with 2.4.1).
Everything is OK now!
Thank you for the support

closed 2018-07-28 by robygi

Riddlr, 2018-08-03:
I have a similar issue but what happens for me is that it will only boot properly after flashing.  If I select restart from the web console it the serial monitor shows:

05:33:38 SRC: WebGui from 0.0.0.0
05:33:39 APP: Restarting

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

And it never comes back - I have to flash it again.  Not sure what could be the issue?

Riddlr, 2018-08-03:
My issue turned out that I compiled with 512K(no SPIFFS) and not 1M...