arendst/Tasmota #3326: IRreceive parsing raw data

jnt2007, 2018-07-25:
Hi.

I have Sonoff basic with connected TSOP IR receiver. After activating Samsung AC remote control more debug log write next:
```
00:00:14 IRR: RawLen 74, Bits 37, Value 3FFF4B54, Decode -292978525
00:00:15 IRR: RawLen 74, Bits 37, Value 3FFF4B54, Decode 961051959
```
I'm trying to understand how to send this IR signal via another Sonoff Basic with [this mod](https://github.com/altelch/SonoffIR).

How to decode into IRsend command?

Jason2866, 2018-07-25:
What do you see on the webconsole (no debug modus) if ir received?

jnt2007, 2018-07-25:
@jason2866 Nothing. No any info in web console if debug mode disabled.

Jason2866, 2018-07-25:
Then it is not supported.

jnt2007, 2018-07-26:
Maybe reading and sending raw data might be implemented in the future?

ascillato2, 2018-08-25:
Hi,

Duplicated issue. See #2116

Thanks.

closed 2018-08-25 by ascillato2