arendst/Tasmota #3327: Alexa an Belkin Emulation for Sonoff dual R2

rewegit, 2018-07-25:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : Sonoff dual R2
- [ ] Provide the output of command ``status 0`` :
```
12:29:59 MQT: sonoffd/stat/STATUS = {"Status":{"Module":39,"FriendlyName":["Sonoffd1","Sonoffd2"],"Topic":"sonoffd","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:29:59 MQT: sonoffd/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:55:08","StartupUTC":"2018-07-25T10:34:51","Sleep":0,"BootCount":18,"SaveCount":182,"SaveAddress":"F5000"}}
12:29:59 MQT: sonoffd/stat/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018.07.23 11:20:42","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:29:59 MQT: sonoffd/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["RWBuero8","RWBuero9"],"TelePeriod":300,"SetOption":["00000009","55A18000"]}}
12:29:59 MQT: sonoffd/stat/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","0C000000","23B6179E","00000000"]}}
12:29:59 MQT: sonoffd/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoffd-0657","IPAddress":"192.168.99.109","Gateway":"192.168.99.253","Subnetmask":"255.255.255.0","DNSServer":"192.168.99.253","Mac":"60:01:94:98:C2:91","Webserver":2,"WifiConfig":4}}
12:29:59 MQT: sonoffd/stat/STATUS6 = {"StatusMQT":{"MqttHost":"raspidb","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_98C291","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:29:59 MQT: sonoffd/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Jul 25 11:29:59 2018","Local":"Wed Jul 25 12:29:59 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:14","Sunset":"20:38"}}
12:29:59 MQT: sonoffd/stat/STATUS10 = {"StatusSNS":{"Time":"2018-07-25T12:29:59","Switch1":"ON","Switch2":"ON"}}
```

Sonoff works perfectly with Alexa and BELKIN emulation. 
With my SonOff dual R2 I can only see one device in the Alexa app with Belkin emulation.
With the Hue Bridge emulation I can see both channels, but cannot switch to Socket.
Is there a solution to use both channels?

arendst, 2018-07-25:
Tasmota is designed with WeMo emulation supporting one relay and Hue emulation supporting more relays.

closed 2018-07-25 by ascillato2

fragolinux, 2018-11-11:
> Tasmota is designed with WeMo emulation supporting one relay and Hue emulation supporting more relays.

hi
which is the correct command to turn on or off "hue emulated" devices? If i say "alexa accendi alza" or "alexa accendi abbassa" (italian for: alexa, turn on ALZA, or alexa, turn on ABBASSA, which are the names of the 2 relays in a sonoff dual r2) she says these commands are not available in the device...

ClauRevolver, 2018-11-11:
Which Tasmota version are you using? If you're using 6.3.0 and Core 2.3.0 apparently there's a bug with the new Echo firmware. I'm also unable to operate my Sonoff T1 2CH.
Alexa can discover the devices (but they do not show under routines) and trying to toggle them on/off I receive an error like "device is not responding" or "il server non risponde" in italian.
I guess we will have just to wait for the next tasmota releases.

ascillato, 2018-11-11:
Hi, you don't need to wait. Just use the precompiled bins from development branch that are working great:

http://thehackbox.org/tasmota/

recommended flavour: http://thehackbox.org/tasmota/sonoff.bin

fragolinux, 2018-11-11:
```
21:49:12 CMD: status 0
21:49:12 RSL: stat/sonoff/STATUS = {"Status":{"Module":39,"FriendlyName":["ALZA","ABBASSA"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:49:12 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-IT.bin","RestartReason":"Power on","Uptime":"0T01:03:12","StartupUTC":"2018-11-11T19:46:00","Sleep":0,"BootCount":16,"SaveCount":170,"SaveAddress":"FA000"}}
21:49:12 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0","BuildDateTime":"2018-10-30T17:43:43","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:49:12 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shark-HOME2","SharkAzzo2"],"TelePeriod":300,"SetOption":["0000C009","55C18000","00000000"]}}
21:49:12 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":490,"Free":512,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000410","0FDAE794","000183A0","23B617CE","00003BC0"]}}
21:49:12 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4023","IPAddress":"192.168.1.231","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:9D:6F:B7","Webserver":2,"WifiConfig":5}}
21:49:12 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D6FB7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:49:12 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 11 20:49:12 2018","Local":"Sun Nov 11 21:49:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:53","Sunset":"17:15"}}
21:49:12 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-11T21:49:12","AM2301":{"Temperature":26.8,"Humidity":41.8},"TempUnit":"C"}}
21:49:12 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-11T21:49:12","Uptime":"0T01:03:12","Vcc":3.182,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"SharkAzzo2","BSSId":"A0:63:91:E0:A1:84","Channel":2,"RSSI":22}}}
```
standard latest SONOFF-IT.bin from releases page, is it different from the ones on thehackbox?

fragolinux, 2018-11-11:
flashed 6.3.0.7 from thehackbox, same results, alexa says that turning on or off the 2 named relays is a not available function...
```
21:54:13 CMD: status 0
21:54:13 RSL: stat/sonoff/STATUS = {"Status":{"Module":39,"FriendlyName":["ALZA","ABBASSA"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:54:13 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-IT.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:17","StartupUTC":"2018-11-11T20:52:56","Sleep":0,"BootCount":17,"SaveCount":174,"SaveAddress":"FA000"}}
21:54:13 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0.7(c0cfdde-sonoff)","BuildDateTime":"2018-11-11T19:08:35","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:54:13 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shark-HOME2","SharkAzzo2"],"TelePeriod":300,"SetOption":["0000C009","55C18000","00000000"]}}
21:54:13 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000410","0FDAE794","000183A0","23B617CE","00003BC0"]}}
21:54:13 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4023","IPAddress":"192.168.1.231","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:9D:6F:B7","Webserver":2,"WifiConfig":5}}
21:54:13 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D6FB7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:54:13 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 11 20:54:13 2018","Local":"Sun Nov 11 21:54:13 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":"+01:00","Sunrise":"07:53","Sunset":"17:15"}}
21:54:13 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-11T21:54:13","AM2301":{"Temperature":26.9,"Humidity":41.7},"TempUnit":"C"}}
21:54:13 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-11T21:54:13","Uptime":"0T00:01:17","Vcc":3.182,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"SharkAzzo2","BSSId":"A0:63:91:E0:A1:84","Channel":2,"RSSI":20}}}
```

ascillato, 2018-11-11:
yes, the release page is the version 6.3.0
the ones from http://thehackbox.org/tasmota/ are the precompiled bins of the development branch (newer ones) now it is 6.3.0.7

fragolinux, 2018-11-11:
> yes, the release page is the version 6.3.0
> the ones from http://thehackbox.org/tasmota/ are the precompiled bins of the development branch (newer ones) now it is 6.3.0.7

look my other comment, same on 6.3.0.7

ascillato, 2018-11-11:
yes we write at the same time.

ascillato, 2018-11-11:
have you turned on hue emulation ?

have you set the friendly names on the configuration page of Tasmota?

fragolinux, 2018-11-11:
yes to both questions, friendly names are in my comments, don't know if the emulation is hidden in all that code, too, but yes, it's turned on and alexa can detect both relays ALZA and ABBASSA, but can't turn them on

ClauRevolver, 2018-11-11:
Thank you @ascillato, I have the same issue, the channels are recognized as dimmable lamps but when trying to use them from the app there’s an error saying the device does not support the parameter.

ascillato, 2018-11-11:
@have you tried lastest version from http://thehackbox.org/tasmota/ ?

ascillato, 2018-11-11:
@fragolinux to discard a 6.3.x issue. Can you try 6.0.0?

ClauRevolver, 2018-11-11:
@ascillato, yes sorry! I should have explained better. I flashed with 6.3.0.7 as you suggested. 
I have to say from time to time the switches work even if the app return that error. Most of the times the routines do not work.

fragolinux, 2018-11-11:
everything else you need, i'm here, device is not yet mounted on my blinds shutters, so it's here for testing :)

Jason2866, 2018-11-13:
@ascillato @fragolinux 
Tested HueEmulation with Alexa with a WemosMini with two connected Relais.
Works fine with latest dev build 6.3.0.7 and core 2.3.0

```
09:10:55 CMD: Status 0
09:10:55 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Hund","Katze"],"Topic":"sonoff","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:10:55 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:21","StartupUTC":"2018-11-13T08:04:34","Sleep":1,"BootCount":16,"SaveCount":92,"SaveAddress":"F8000"}}
09:10:55 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.3.0.7(sonoff)","BuildDateTime":"2018-11-13T08:59:39","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:10:55 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
09:10:55 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":453,"Free":548,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F8AA784","140003A1","80041496","000000C2"]}}
09:10:55 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5798","IPAddress":"192.168.2.152","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:53:D6:A6","Webserver":2,"WifiConfig":5}}
09:10:55 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_53D6A6","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:10:55 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 13 08:10:55 2018","Local":"Tue Nov 13 09:10:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:18","Sunset":"16:35"}}
09:10:55 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-11-13T09:10:55"}}
09:10:55 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-11-13T09:10:55","Uptime":"0T00:06:21","Vcc":2.762,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":94}}}
```

Jason2866, 2018-11-13:
Cant say if it is a problem with the dual. I didnt have one to test. But i dont think so.
Mhh, gonna flash this version on a S20 and see if it becomes laggy.
It would be the first time ever that i have problems with core 2.3.0....

fragolinux, 2018-11-13:
i can do that, too, but the problem (at least the alexa one) seems related to hue emulation, which is needed for multirelay boards as wemo only supports one... so in s20, you should use hue to test this scenario, and even so you'll end up with a single relay

Jason2866, 2018-11-13:
I am doing a longer testing with the Wemos Board. It works fine with 2 Relais and 1 PWM dimmed channel.
All 3 devices can be controlled via Alexa. Switching on / off and controlling dimming with voice command.
I have a Sonoff 4CH and all 4 relais can be independently switched via Alexa, works fine since about a week when i have installed.
Does every device has a different "Friendly Name"?

fragolinux, 2018-11-13:
> Does every device has a different "Friendly Name"?
sure, they're visible in my comments above, 1st line of the "status 0" command...

Jason2866, 2018-11-14:
Devices (S20 and Wemos mini) are still working no problems...

fragolinux, 2018-11-14:
> Devices (S20 and Wemos mini) are still working no problems...

my dual became completely unaccessible, instead... not available on the network anymore... maybe it's faulty, have to retry, next week, now i'm away

ClauRevolver, 2018-11-14:
mh...as for me I'm back at square one, the T1 gets discovered by Alexa, the 2 channels show up as non dimmable lamps but both trigger "device not responding" error when trying to operate them.

ClauRevolver, 2018-11-14:
> I am doing a longer testing with the Wemos Board. It works fine with 2 Relais and 1 PWM dimmed channel.
> All 3 devices can be controlled via Alexa. Switching on / off and controlling dimming with [...]

@Jason2866 what is your echo model please and which firmware is it running?

Jason2866, 2018-11-14:
Echo Dot 2nd Generation
Firmware self compiled 
https://github.com/arendst/Sonoff-Tasmota/issues/3327#issuecomment-438174316

ClauRevolver, 2018-11-14:
Right. Everything works perfectly also for me using the Echo Dot Gen 2 I have, even running the original amazon firmware. 

The problem I'm facing is with a brand new Echo Gen2 device (not the dot)...

Of course I'm using 2 different amazon accounts to manage those 2 echo's

Jason2866, 2018-11-14:
Amazon changed "how things work" in Echo Gen2
OpenHab (a HA system) has problems with this new device too.
So it needs some reverse engeniering to get it running...

Jason2866, 2018-11-14:
@ClauRevolver 
Do you compile your own versions?
Would you help us if we need some testing or log files?

fragolinux, 2018-11-18:
just upgraded to version 6.3.0.10 and still issues with hue emu on dual r2, using echo 2nd generation

Jason2866, 2018-11-18:
Nothing changed in HueEmulation since 6.3.0...

fragolinux, 2018-11-18:
> Nothing changed in HueEmulation since 6.3.0...

i know, i read the changelogs, just reporting back and adding details trying to help to solve the issues

ClauRevolver, 2018-11-18:
> @ClauRevolver
> Do you compile your own versions?
> Would you help us if we need some testing or log files?

Hey @Jason2866, sorry for my late reply! I’m traveling a lot for business now a days and I overlooked this one. 
I generally do not compile my own version but sure, I’d love to help in any mean!

c0w, 2019-03-20:
Sorry to tag this onto the bottom, hopefully the detail will be helpful. 

I'm having the same issue as above and hoping someone can point me in the direction to address the issue. I've got a loads of Sonoff devices around the house, the single ones work perfect and I've got no issues the newer looking UK devices i'm having issues with Alexa and the device is not responding. I can pull up the web interface tick the toggle and they work absolutely fine. 

I've just upgraded to the latest version in a hope to resolve things, but I'm still suffering with the issue any help/guidance would be most appreciated.

`20:51:30 RSL: STATUS = {"Status":{"Module":29,"FriendlyName":["9","10"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:51:30 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:28","StartupUTC":"2019-03-20T19:41:02","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":129,"SaveAddress":"F9000"}}
20:51:30 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:51:30 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["m00",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["0000A001","280500000100000000000000000000000000","00000020"]}}
20:51:30 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
20:51:30 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-3620","IPAddress":"10.0.0.28","Gateway":"10.0.0.254","Subnetmask":"255.255.255.0","DNSServer":"176.103.130.130","Mac":"60:01:94:97:8E:24","Webserver":2,"WifiConfig":4}}
20:51:30 RSL: STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 20 19:51:30 2019","Local":"Wed Mar 20 20:51:30 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:53","Sunset":"19:02"}}
20:51:30 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-03-20T20:51:30"}}
20:51:30 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-03-20T20:51:30","Uptime":"0T00:10:28","Vcc":3.262,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"m00","BSSId":"0C:F4:D5:09:16:38","Channel":11,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:40"}}}` 

I've done a fresh upgrade and the device is working, Alexa is able to turn the device on as soon as the device comes on via Alexa it drops off the network. 

[image]

Then of course if I ask Alexa to turn the device on it reports its not responding. Its extremely frustrating that these devices don't work the same as the other ones (flawlessly) I've been contemplating putting the original firmware back on until I can make these work well as this controllers covers my kitchen and snug so i'm getting a fair bit of wifey agro.

Happy to run or debug or test anything required, just let me know Since initiating the command to turn on the device has still not returned to the network, if i restart it will or if i wait several hours it will reconnect and in the log will be massages about unable to connect to AP. 

Apologies for the massive wall of text, hopefully its helpful if would be good to get this cleared off once and for wall.

Jason2866, 2019-03-21:
Try Hue Bridge emulation. There is a open issue with Belkin Emulation https://github.com/arendst/Sonoff-Tasmota/issues/5505

c0w, 2019-03-21:
@Jason2866 
This is with the Hue Bridge emulation, i thought it showed up in the Status 0 but I've just checked the debug above and it doesn't show it. Is there anyone with the devices and a working firmware and I'll try that. I've tried about 7-8 different firmware's and all suffer some similar issues.

Thanks