arendst/Tasmota #3328: sonoff basic - flash form espeasy web interface to tasmota

lpt2007, 2018-07-25:
Hi

I flash tasmota (latest sonoff-allsensors.bin) from espeasy web interface. 

At end I get message update success. But now I can not connect to sonoff-basic (no light and if I press button once relay not click). 

I also try all this but not working.
[https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage](url)

Frogmore42, 2018-07-25:
You will likely need to flash using serial.  There are things that could have gone wrong in the update, that once gone wrong can only be fixed by serial flash, which is the only recovery method that almost always works.

ascillato2, 2018-07-25:
Hi,

Have you managed to solve your issue by flasing by wire?

ascillato2, 2018-07-28:
Closing this issue as there is no feedback. Please, ask to reopen if you need anything else. Thanks

closed 2018-07-28 by ascillato2