arendst/Tasmota #3329: Tasmota with Itead Nextion display - How To?

martinmatias, 2018-07-25:
Hi, I'm trying to use Tasmota with itead Nextion display using itead nextion library: https://github.com/itead/ITEADLIB_Arduino_Nextion

I've tried creating a "xdrv_98_user_application.ino" copied from https://github.com/tobox/Sonoff-Tasmota/blob/development/sonoff/xdrv_98_user_application.ino without success. It seems that serial port still used by loggin to serial console and dont know how to disable.

For example, I tried to intializate display and write something simple into a button of the display. below the modified code:
`void` User_Application_Init()
{
 
  snprintf_P(log_data, sizeof(log_data), "User Application Init...1");
  AddLog(LOG_LEVEL_INFO);
  serial_local = 1;
  nexInit();
  Serial.begin(115200);
  delay(1000);
  b0.setText("martin_init");   

}
`
¿Any idea how to solve this problem?
Regards
Martin

renne, 2018-07-25:
Have you tried to set `#define SERIAL_LOG_LEVEL` to `LOG_LEVEL_NONE` in line 78 of `user_config.h`?

martinmatias, 2018-07-26:
Hi renne, I've tried and see that serial port still loggin into serial console. Doesn't solved my problem. Regards
Martin

ascillato, 2018-07-26:
Type in the console seriallog 0

martinmatias, 2018-07-26:
H Adrian, thanks. I've just tested with SerialLog 0 into webconsole an it worked!! How can I disable Serial at compilation instead of runtime?
I've also added:
`   case FUNC_LOOP:
    nexLoop(nex_listen_list);
    break;  `
and now it also receive command (inputs) from display.

curzon01, 2018-07-26:
see user_config.h (and user_config_override.h) and set

`#define SERIAL_LOG_LEVEL       LOG_LEVEL_NONE`

renne, 2018-08-08:
I'm going to use the SerialBridge with Node-Red for a more flexible approach.
```
backlog Baudrate 9600; SerialDelimiter 255; SerialSend3 h0.val=75\0xFF\0xFF\0xFF;
```
does not work.

Does anyone have an idea what's wrong with the command?

ascillato2, 2018-08-22:
@renne Please, can you open a new issue to track that problem? Thanks

ascillato2, 2018-08-22:
Closing this issue as @martinmatias has stated that it is solved. Thanks

closed 2018-08-22 by ascillato2

renne, 2018-08-24:
@martinmatias 
How do you execute the `SerialLog 0` command in your code?

martinmatias, 2018-08-24:
Hi @renne I've didnt code SerialLog 0, I've run it after compiling. It seems to be persistent. Unfortuatelly, I wasn't able to cotinue with my project using it, I hope will return to it soon.
Would be better if:
 `#define SERIAL_LOG_LEVEL       LOG_LEVEL_NONE    // LOG_LEVEL_INFO [SerialLog] (LOG_LEVEL_NONE, LOG_LEVEL_ERROR, LOG_LEVEL_INFO, LOG_LEVEL_DEBUG, LOG_LEVEL_DEBUG_MORE)

worked at compilation time, but it didnt (no error, just didnt worked for me)

regards
Martin

ascillato, 2018-08-24:
That config will work if you reset your device settings by pressing the button for 40 seconds. That config is the default config. If you already have Tasmota installed, your config in your eeprom will be another. To take effect the config changes of your code, you have to reset your config.

curzon01, 2018-08-25:
see also [CFG_HOLDER explained](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#cfg_holder-explained) - **Summarized**

"Reset" command is the fastest and easiest way to make the compiled device settings as active

renne, 2018-08-28:
@martinmatias 
`SetSeriallog(LOG_LEVEL_NONE);` works :-)