arendst/Tasmota #3330: Error while sendind IR command to LG AC

douglaspaulo, 2018-07-25:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff Basic with IR Led
- [ ] Provide the output of command ``status 0`` :

When i sending IR command to my AC LG over the Tasmota I getting the error below:

cmnd/ac/IRSEND {"PROTOCOL":"LG","BITS":28,"DATA":"88C0051"}

stat/ac/RESULT = {"IRSend":"Protocol not supported"}

USE_IR_REMOTE it´s active on user_config.h

Please help me :(

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-07-25:
Tasmota does not support LG

Jason2866, 2018-07-25:
LG TV´s are using NEC protocol...

douglaspaulo, 2018-07-25:
Theo good affternoon

Tasmota supports Samsung Air Conditioned??
My air conditioned is  LG, I will sell and buy a Samsung Air Conditioned....

Enviado do meu iPhone

Jason2866, 2018-07-25:
I dont think Samsung will work. Take a look in wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#irremote

douglaspaulo, 2018-07-25:
Hi Theo and Jason

Sorry for my questions.

On wiki, I observ that SAMSUNG is available 

It’s right? Or AC Category is only IRhvac?

Thanks so much. I use 10 sonoffs in my house and your project are awesome!!!

Enviado do meu iPhone

Jason2866, 2018-07-25:
AC Category is only IRhvac!

douglaspaulo, 2018-07-26:
Thank you for explanation

Have a good day my friends.

Cheers

Enviado do meu iPhone

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2