arendst/Tasmota #3333: Rule in arduino IDE, configure

Sergio-tix, 2018-07-25:
Hi!
I use ESP-12S with firmware Sonoff-Tasmota 6.1.1. I select model type "18 Generic".
All configuration and firmware I make through the Arduino IDE. I manage to successfully flash firmware and use my ESP-12S as Relay through MQTT, with iobroker.
But I have new needs - I want to use RULES. My goal is to get 4 inputs and 4 outputs, completely independent of each other, managed and receiving statuses through MQTT.
I read the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules , but I did not understand.
I made a rule:
`rule 1
rule on switch1#state do publish /stat/remote1/switch1 endon on switch2#state do publish /stat/remote2/switch2 endon on switch3#state do publish /stat/remote3/switch3 endon on switch4#state do publish /stat/remote4/switch4 endon rule 1 endon
`

The wiki says: 

>  To use rules the define USE_RULES has to be enabled in file user_config_override.h

This means that add a library `#define USE_RULES` to the begining file  user_config_override.h ?
It is not clear what and where to add.
In file user_config_override_sample.h is written:

> To modify the stock configuration without changing the user_config.h file:
> *   (1) copy this file to "user_config_override.h" (It will be ignored by Git )
> *   (2) define your own settings below
> for Arduino IDE:
> * enable define USE_CONFIG_OVERRIDE in user_config.h

In file `user_config.h` I removed the slashes for `#define USE_CONFIG_OVERRIDE`

What else should I set up and copy? I translate this into Russian, and all the instructions turn into a sick man's delirium. I need to copy all from the file `user_config.h` to a file `user_config_override.h` ??
Which file and where in the file do I need to write a RULE ?
It is also written:
> // force the compiler to show a warning to confirm that this file is inlcuded
> #warning **** user_config_override.h: Using Settings from this File ****

But Arduino IDE does not display such a message.
I tried four nights, but nothing happened, and there are no more strengths and thoughts.
Rules do not work. I performed in the console: `SwitchTopic 0` and `rule1 1`, there is no success.
By sending a signal to any Switch - The signal is always displayed only on the POWER.

- [ ] `22:33:41 CMD: status 0
22:33:41 MQT: stat/Generic_ESP8266/STATUS = {"Status":{"Module":18,"FriendlyName":["Generic_1"],"Topic":"Generic_ESP8266","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"Generic_ESP8266","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:02:51","StartupUTC":"2018-07-25T21:30:50","Sleep":0,"BootCount":12,"SaveCount":31,"SaveAddress":"F5000"}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-25T23:47:43","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dd",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS4 = {"StatusMEM":{"ProgramSize":517,"Free":484,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000419","0F80E790","00000001","02A0009E","00000000"]}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS5 = {"StatusNET":{"Hostname":"Generic_ESP8266-8888","IPAddress":"192.168.1.31","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:**:**:**","Webserver":2,"WifiConfig":3}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.44","MqttPort":1883,"MqttClientMask":"Generic_1_%06X","MqttClient":"Generic_1_C4****","MqttUser":"******","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS7 = {"StatusTIM":{"UTC":"Wed Jul 25 21:33:41 2018","Local":"Wed Jul 25 22:33:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:14","Sunset":"20:38"}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS10 = {"StatusSNS":{"Time":"2018-07-25T22:33:41","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"OFF"}}
22:33:41 MQT: stat/Generic_ESP8266/STATUS11 = {"StatusSTS":{"Time":"2018-07-25T22:33:41","Uptime":"0T00:02:51","Vcc":3.328,"Wifi":{"AP":1,"SSId":"dd","RSSI":96,"APMac":"AA:BB:CC:DD:EE:FF"}}}
`
How do I get the rules to work? Is there a video, how to set up?

ascillato, 2018-07-26:
Hi,

**Rules are enabled by default** in last release. You don't have to do nothing extra.

All the firmwares on release page have already rules enabled _(except from sonoff_clasic.bin)_

If you compile by yourself, just using the default without changing anything, you will have rules enabled.

Your Status 0 is:

```
{"Status":{"Module":18,"FriendlyName":["Generic_1"],"Topic":"Generic_ESP8266","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"Generic_ESP8266","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:02:51","StartupUTC":"2018-07-25T21:30:50","Sleep":0,"BootCount":12,"SaveCount":31,"SaveAddress":"F5000"},"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-25T23:47:43","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dd",""],"TelePeriod":300,"SetOption":["00008009","55818000"]},"StatusMEM":{"ProgramSize":517,"Free":484,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000419","0F80E790","00000001","02A0009E","00000000"]},"StatusNET":{"Hostname":"Generic_ESP8266-8888","IPAddress":"192.168.1.31","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:**:**:**","Webserver":2,"WifiConfig":3},"StatusMQT":{"MqttHost":"192.168.1.44","MqttPort":1883,"MqttClientMask":"Generic_1_%06X","MqttClient":"Generic_1_C4****","MqttUser":"******","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Wed Jul 25 21:33:41 2018","Local":"Wed Jul 25 22:33:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:14","Sunset":"20:38"},"StatusSNS":{"Time":"2018-07-25T22:33:41","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"OFF"},"StatusSTS":{"Time":"2018-07-25T22:33:41","Uptime":"0T00:02:51","Vcc":3.328,"Wifi":{"AP":1,"SSId":"dd","RSSI":96,"APMac":"AA:BB:CC:DD:EE:FF"}}}
```

And using _decode-status.py_ we can see that:

```
*** decode-status.py v20180725 by Theo Arends ***
Decoding information for device Generic_1 from status report taken at 2018-07-25
T22:33:41

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (ON ) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) Rules until 5.14.0b
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX enabled
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led

Features
  Language LCID = 1049
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  USE_ADC_VCC
  USE_ARILUX_RF
  USE_CONFIG_OVERRIDE
  USE_DHT
  USE_DISCOVERY
  USE_DS18B20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_NOVA_SDS
  USE_PMS5003
  USE_PZEM004T
  USE_RULES
  USE_SERIAL_BRIDGE
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_WEBSERVER
  USE_WS2812
  WEBSERVER_ADVERTISE
```

**_So, Rules are enabled in your firmware_**

-------------------------------

Another thing:

`rule 1 rule on switch1#state do publish /stat/remote1/switch1 endon`

will not work. Need to be rewrited as:

```
rule 1
rule on switch1#state do publish stat/remote1/switch1 %value% endon
```

Sergio-tix, 2018-07-26:
Thanks for the quick response!
I made the rules like this:
`rule 1
rule on switch1#state do publish stat/remote1/switch1 %value% endon on switch2#state do publish stat/remote2/switch2 %value% endon on switch3#state do publish stat/remote3/switch3 %value% endon on switch4#state do publish stat/remote4/switch4 %value% endon rule 1 endon
`
But the rules do not work again.
I think that I write the rules not in the right place. The rules must be in a file `user_config_override.h`, at the end before text `#endif  // _USER_CONFIG_OVERRIDE_H_` ?
Or am I doing something wrong?
Screenshot of iobroker:
[image]
Changes only the value I highlighted in red.

arendst, 2018-07-26:
The rules are only stored when you execute a rule command.

As ascillato tried to tell there is no need to compile anything. It's all done online either using serial, mqtt or http.

Sergio-tix, 2018-07-26:
Now I understand. Thank you!
I entered the rule in the console, When closing and opening contacts S1-S4 I see the:
`09:02:07 CMD: rule1
09:02:07 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","Free":260,"Rules":"on switch1#state do publish stat/remote1/switch1 %value% endon on switch2#state do publish stat/remote2/switch2 %value% endon on switch3#state do publish stat/remote3/switch3 %value% endon on switch4#state do publish stat/remote4/switch4 %value% endon"}
09:02:21 RUL: SWITCH1#STATE performs "publish stat/remote1/switch1 2"
09:02:21 MQT: stat/remote1/switch1 = 2
09:02:22 RUL: SWITCH1#STATE performs "publish stat/remote1/switch1 2"
09:02:22 MQT: stat/remote1/switch1 = 2
09:02:29 RUL: SWITCH2#STATE performs "publish stat/remote2/switch2 2"
09:02:29 MQT: stat/remote2/switch2 = 2
09:02:29 RUL: SWITCH2#STATE performs "publish stat/remote2/switch2 2"
09:02:29 MQT: stat/remote2/switch2 = 2
09:02:35 RUL: SWITCH3#STATE performs "publish stat/remote3/switch3 2"
09:02:35 MQT: stat/remote3/switch3 = 2
09:02:35 RUL: SWITCH3#STATE performs "publish stat/remote3/switch3 2"
09:02:35 MQT: stat/remote3/switch3 = 2
09:02:42 RUL: SWITCH4#STATE performs "publish stat/remote4/switch4 2"
09:02:42 MQT: stat/remote4/switch4 = 2
09:02:42 RUL: SWITCH4#STATE performs "publish stat/remote4/switch4 2"
09:02:42 MQT: stat/remote4/switch4 = 2`
Does this mean that my rules work? In the iobroker now no reaction at all. Driver sonoff does not see these MQTT topics? Need to deal with it?
[image]

arendst, 2018-07-26:
The rules work.

You'll have to tell iobroker to look for the correct MQTT message of ``stat/remote1/switch1`` etc. I know zilch from iobroker sorry.

Sergio-tix, 2018-07-26:
Thank you! No more questions.

Sergio-tix, 2018-07-26:
Again the question.
When I applied the settings and rules which wrote above, at closing I see:
[image]
What is "2" ?

closed 2018-07-28 by Sergio-tix