arendst/Tasmota #3334: Controlling an electric shutter using Amazon Echo

kouske, 2018-07-26:
Hi, 
I am trying to figure out how to control my electric shutters using a sonoff running tasmota and an Amazon Echo. I have everything connected, interlock mode enabled, pulse timer enabled and Hue Bridge emulation. 

My issue is with the Alexa control: I have two friendly names configured: Shutter Up and Shutter Down. If I want to move the shutter, I need to tell Alexa "turn shutter up on" or "turn shutter down on". 
Also, if I want to control my whole house and turn it off ( lights, air condition, TV and shutters) I need to issue the command "Alexa, turn off home". The issue with this is that even if I put both shutter friendly names in a single Amazon Echo group called Shutter, it still won't close the shutter since the "off" command turns the relays off and thus nothing will happen. 

My idea is to add an option to tasmota, specifically the Hue emulation, which will inverse the on/off functionality of one of the friendly names. 

This will essencialy make it so that I can control the group called Shutter with only two commands:
- If I say "shutter off" the relay which corresponds to "down" will be turned on whilst the other one will be turned off
- If I say "shutter on" the relay which corresponds to "up" will be turned on, down will be turned off. 

Also, I thought about adding another functionality which will act as follows:
- If one of the relays is on and I issue the off command, both relays will switch off (this adds stop functionality). 

Note: this is not really a feature request since I can write the code myself, I'm trying to understand if this makes sense and if this is something that is already available or not. Thanks for reading :)

Jason2866, 2018-07-26:
Just my 2 cents. 
TASMOTA  firmware is great, but it is impossible to implement every feature or use case...
Alexa is still too limited to do complex Home Automation.
What you want can be done with a powerful Home Automation system.
For example: OpenHab can control Alexa devices. All you want to do is possible with Open Hab :-)
Take a look at https://community.openhab.org/t/released-openhab2-amazon-echo-control-binding-controlling-alexa-from-openhab2/37844

kouske, 2018-07-26:
Well, I'm trying to avoid adding a 3rd party home automation solution (such as OpenHab, HA etc.).
Ideally, I would like to buy a 15 dollar device that would have an Alexa skill that would control shutters. Unfortunately, I did not yet find one which is why I opened this thread :)

alexeinz, 2018-07-26:
i thought button functionality can be inverted in tasmota , why not just reverse one and set names accordingly ..? 
also i wonder how you control shutter with tasmota ? isnt it one motor with up, down and stop?

kouske, 2018-07-26:
I'm using the Dual R2 to control both directions (each relay corresponds to a different direction).
How can I invert the button functionality? Will this work with voice commands sent over the Hue emulation?

Jason2866, 2018-07-26:
Button function is useless here. You control the device via Alexa. No button involved.
But the idea goes in the right direction. Define the module as generic an invert the relay function.
Look in sonoff_template.h for the Dual definition. Take this seetings and configure your generic module 1:1 and use inverted relay definition.

kouske, 2018-07-26:
My original idea was just to add an inversion when the Hue module parses the "on" command e.g 

`if (device == 1)
 execute reverse switch logic here
else if (device == 2)
 execute normal switch logic here`

I'm guessing this kind of ruins the Hue module so it's not the best solution.
Is there a guide for defining generic modules? Implementing a generic module can produce the same results? I'm not an expert on this, but I don't mind reading a bit :)

Jason2866, 2018-07-26:
Just select (easiest in webpage) module "Generic" instead Dual.
Here the code snipplet from sonoff_tmplate.h
`{ "Sonoff Dual R2",  // Sonoff Dual R2 (ESP8285)
     GPIO_USER,        // GPIO00 Button 0 on header (0 = On, 1 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_REL2,        // GPIO05 Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_USER,        // GPIO09 Button 1 on header (0 = On, 1 = Off)
     GPIO_KEY1,        // GPIO10 Button on casing
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },`
Just use this a blue copy for your setup (change GPIO_REL1 to GPIO_REL1_INV)
Read here
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini

Frogmore42, 2018-07-26:
If you invert one of the relays, isn't lockout going to be confused?

kouske, 2018-07-26:
No idea about the lockout. Since there are going to be two commands sent to each relay then the only thing that can be bad is if the second command stops the first command. Other than that it should be fine.

Jason, what happens to the switch functionality if I reverse using this method? I have an external switch connected to button0 and button1. Will this somehow affect the button functionality or it stays the same?

Jason2866, 2018-07-26:
Connected switch(es) will work. Maybe (i dont think) you have to change switchmode. Nothing special...

kouske, 2018-07-27:
Eventually, I realized that the logic that I was looking for should be done by higher levels.
The easiest solution for this would be using Alexa Routines + Yonomi but for some reason the Routines don't work where I live :(

Still, now it works like I wanted - I had to do a nasty hack in the wemohue module but hey, at least it works (reversing the relay did not play out well since the interlock mode did get confused).

Frogmore42, 2018-07-27:
It is usually best to keep things simple at the lower levels.  Trying to push too much control/logic too low can make it really hard.  But, moving the logic higher makes the system dependent on all the pieces working correctly, which increases the probability of something going wrong.

Jason2866, 2018-07-27:
@kouske 
My first advice was doing this in higher level 😎

nospam2000, 2018-08-20:
@kouske can you please describe in more detail what you have done?

kouske, 2018-08-20:
Eventually I got the Alexa routines working. Solved everything.

ascillato, 2018-08-20:
Hi,

If your issue is solved, please, close it. Thanks

closed 2018-08-20 by kouske

stefanbode, 2018-09-12:
My MOD of tasmoto is now using Alexa Dimmer function (HUE enulation) to control the shutter. You can say "Alexa shutter to 80%" and the shutter moves to 80%. Also 0% and 100% work. OFF/ON is a bit unnatural, but this also works. https://github.com/stefanbode/Sonoff-Tasmota/wiki/Shutter-blinds-working-with-two-relays