arendst/Tasmota #3335: Always ON - Best strategy

cirolaferrara, 2018-07-26:
My POW auto powered off after changed router and after a lot of hours. I don't know why but my fridge was OFF (sigh) in the weekend.
POW was flashed with sonoff.bin release, tasmota 5.14.0  and connected by MQTT to Openhab. PowerOnState was on 1 and Openhab has retain=false

POW won't go in AP mode (4 short presses) and won't RESET (button for over forty seconds) so i flashed by wire.

I compiled tasmota from source and edited this:
```
#define STA_SSID1 "MYWIFI"
#define STA_PASS1 "MYPWD"
#define WIFI_CONFIG_NO_SSID WIFI_RETRY
#define APP_POWERON_STATE POWER_ALL_ALWAYS_ON
``` 
After flashed it in the console I wrote:
`Reset 1`

After i configured MQTT parameters etc. Can I stay safe? My POW will be always powered ON?

Thank you

Jason2866, 2018-07-26:
Change of 
`#define WIFI_CONFIG_NO_SSID WIFI_RETRY`
is not neccessary. You have defined wifi SSID ane Pwd! 
Check if you have
`#define WIFI_CONFIG_TOOL       WIFI_WAIT `

cirolaferrara, 2018-07-26:
Yes I have. It is a default value

ascillato, 2018-07-31:
Hi,

> After i configured MQTT parameters etc. Can I stay safe? My POW will be always powered ON?

With all that you have configured and `wificonfig 5` (wait) that should be ok)

If your question has been addressed, please close this issue. Thanks. :+1:

closed 2018-07-31 by cirolaferrara