arendst/Tasmota #3337: Problem with Sonoff T1 EU

Balanceprof, 2018-07-26:
Hi all i have t1 eu and problem with it. I download last surce and try upload in sonoff, after connect on wi-fi and mqtt server i watch that sonoff on and off power every 1-2 minutes. it`s not mqtt, i try whithout it and on 3 same sonoff t1

```
00:00:00 Project sonoff Sonoff (Topic t001, Fallback DVES_8300BB, GroupTopic sonoffs) Version 6.1.1c-2_3_0
00:00:00 WIF: Connecting to AP2 BP in mode 11N as t001-0187...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on t001-0187.local with IP address 192.168.124.88
00:00:04 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/t001/LWT = Online (retained)
00:00:05 MQT: cmnd/t001/POWER = 
00:00:05 MQT: tele/t001/INFO1 = {"Module":"Sonoff Basic","Version":"6.1.1c","FallbackTopic":"DVES_8300BB","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/t001/INFO2 = {"WebServerMode":"Admin","Hostname":"t001-0187","IPAddress":"192.168.124.88"}
00:00:05 MQT: tele/t001/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: stat/t001/RESULT = {"POWER":"ON"}
00:00:07 MQT: stat/t001/POWER = ON
16:31:39 MQT: tele/t001/STATE = {"Time":"2018-07-26T16:31:39","Uptime":"0T00:00:14","Vcc":3.244,"POWER":"ON","Wifi":{"AP":2,"SSId":"BP","RSSI":100,"APMac":"4C:5E:0C:3D:2B:71"}}
16:31:42 MQT: stat/t001/RESULT = {"POWER":"OFF"}
16:31:42 MQT: stat/t001/POWER = OFF
16:32:25 MQT: stat/t001/RESULT = {"POWER":"ON"}
16:32:25 MQT: stat/t001/POWER = ON
16:33:27 MQT: stat/t001/RESULT = {"POWER":"OFF"}
16:33:27 MQT: stat/t001/POWER = OFF
16:33:58 MQT: stat/t001/RESULT = {"POWER":"ON"}
16:33:58 MQT: stat/t001/POWER = ON
```

Frogmore42, 2018-07-26:
Please provide all of the requested information from the template.  It makes it much easier to rule out the simple reasons for the issue.

How do you have the module configured?

It looks like the device is being told to change it's state.  That can be via MQTT or something is toggling the GPIO.

Frogmore42, 2018-07-26:
Maybe it is very sensitive and it thinks it is being touched when it is not.

Jason2866, 2018-07-26:
Do you have connected anything on GPIOs?

helmar74, 2018-07-27:
Do you have it powered with 230V~ or with USB/Serial Adapter?
If connected via Adapter be sure to disconnect the relais board:
[https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU](url)

> DO NOT attempt to power the device from your programmer when connected to the relay board. This draws too much power and will cause the ESP to reboot frequently and will reset all the configured settings.

Balanceprof, 2018-07-30:
Hi all, sorry for long answer. It was my fail. I test on serial connect, when i pick parts at one and launched at 220v that was ok

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2