arendst/Tasmota #3338: Interest in Chrome Extension?

drbeat, 2018-07-26:
Hi,

I am working on a Chrome Extension hobbywise.
Is there interest that I publish it and make it open?

If so are there some volunteers to help developing it?

For now I am using ReactJS as a base.

Regards

meingraham, 2018-07-26:
Looks useful. I'd like to have it. The extension's screenshots look similar in function to the TasmoAdmin (SonWeb) homepage and autoscan functions. In addition to discovery and a dashboard to every device, what other features would the extension provide?

arendst, 2018-07-26:
And this time the interface to Tasmota uses MQTT so I can finally remove the webserver in Tasmota....

meingraham, 2018-07-26:
@arendst Just for my own edification - if you don't have a web server in TASMOTA, what responds to the http requests made to the device admin web interface?

arendst, 2018-07-26:
Nothing. You should do MQTT requests as is tasmota's core business. Http has been hijacked.

meingraham, 2018-07-26:
@arendst are you saying that this http interface is being retired? What do you mean by hijacked?
[image]

drbeat, 2018-07-26:
I think some kind of a webserver is needed if binaries are already compiled.
As there needs to be a way to set up WiFi+mqtt broker.

I actually meant that a webserver as in TasmoAdmin would not be necessary anymore.

Maybe I am wrong? I am just a frontend dev :sweat_smile:

curzon01, 2018-07-27:
don't forget 
- the smartphone user (I don't think this plugin will ever exist for this number of users) and it's awesome to configure a Tasmota not needed to starting a PC...
- the iP**** user
- the user who do not have chrome
- user without having a mqtt broker
...

reloxx13, 2018-07-27:
i think, theo meant his own devices at home^^

ascillato2, 2018-08-25:
Good Idea!

Looks great!

But why not helping to expand features of the actual [TasmoAdmin](http://github.com/reloxx13/tasmoAdmin) tool that it is already working very well? Just an idea.

curzon01, 2018-08-25:
A _pro Web GUI_ story:
Just had the situation where I tried to upgrade my old fw 5.14.0 sonoffs - after some isolated tests before - to current 6.1.1
Part of them had no reaction on "Upgrade" commands via mqtt -  may be it was/were issue(s) related with the outdated fw but anyway in that case I could use the backup Web-GUI :sunny: - otherwise I had to dismount some from bad accessible places. 

Background: the sonoffs which wouldn't start to upgrade were all up >90 days
manual "Upgrade 1" had no reaction, also no access attempts to Ota Url (observed the network) - and also still not working after sonoff "Restart" :cry: :confused:  . So the only way was manual start with minimal.bin and then manual do the final.bin using the Web GUI :clap:

Long live the GUI :grin:

drbeat, 2018-08-26:
Sorry I guys, I did not have enough time to work on that in the past couple of weeks.
But I will definitely continue work on that.

@ascillato2 
Well the reason is, that I want to control all my Tasmota enabled devices via a dashboard and I dont like to host a whole server with SQL & PHP for that. So my intention was to reduce the amount of effort to have dashboard for all devices. Also if I get familiar with "Ionic" I can bring the whole project to mobile devices as an app.

wongnam, 2018-08-29:
@meingraham `What do you mean by hijacked?` it's mean that if the command send to devices using http method included un-encrypted password on the networking traffic it will easy be captured by http sniffer.

meingraham, 2018-08-29:
@wongnam that was Theo (@arendst) that mentioned hijacking. I'm not sure what that meant. Theo?

arendst, 2018-08-29:
I meant that the main task of tasmota is MQTT and not HTTP. HTTP is used by TasmoAdmin. Others are working on a MQTT implementation where you access by topic and not by ever changing IP addresses...

closed 2018-10-28 by stale[bot]