arendst/Tasmota #3339: Sonoff 4ch Pro R2 in inching mode. Momentary ON from MQTT on Wi-Fi reconnect

alekseyn1, 2018-07-26:
Hello,

I just recently got into Tasmota with Sonoffs so I might be overlooking something. I have seen a few issues on forums related to a similar setup.

I am using Sonoff 4ch Pro R2 as a garage door opener. Flashing it was not and issue. All runs well. 

But: during the boot AND during WiFi reconnection, those channels that are in the inching mode turn on momentarily. This or course triggers the garage doors in my case. 

So after some research it looks like the MQTT saved state is ON for both of the inching devices. Possibly is does not send OFF state after the relay is disengaged?

Here is the log of what's happening

````
21:30:27 WIF: Connected
21:30:27 DNS: Initialized
21:30:27 MQT: Attempting connection...
21:30:27 MQT: Connected
21:30:27 MQT: tele/sonoff4chpror2/LWT = Online (retained)
21:30:27 MQT: cmnd/sonoff4chpror2/POWER = 
21:30:27 MQT: Subscribe to cmnd/sonoff4chpror2/#
21:30:27 MQT: Subscribe to cmnd/sonoffs/#
21:30:27 MQT: Subscribe to cmnd/DVES_A19D37/#
21:30:27 SRC: MQTT
21:30:27 RSL: Received Topic cmnd/sonoff4chpror2/POWER, Data Size 2, Data ON
21:30:27 RSL: Group 0, Index 1, Command POWER, Data ON
21:30:27 MQT: stat/sonoff4chpror2/RESULT = {"POWER1":"ON"}
21:30:27 MQT: stat/sonoff4chpror2/POWER1 = ON
21:30:27 SRC: MQTT
21:30:27 RSL: Received Topic cmnd/sonoff4chpror2/POWER2, Data Size 2, Data ON
21:30:27 RSL: Group 0, Index 2, Command POWER, Data ON
21:30:27 MQT: stat/sonoff4chpror2/RESULT = {"POWER2":"ON"}
21:30:27 MQT: stat/sonoff4chpror2/POWER2 = ON
21:30:27 SRC: MQTT
21:30:27 RSL: Received Topic cmnd/sonoff4chpror2/POWER3, Data Size 3, Data OFF
21:30:27 RSL: Group 0, Index 3, Command POWER, Data OFF
21:30:27 MQT: stat/sonoff4chpror2/RESULT = {"POWER3":"OFF"}
21:30:27 MQT: stat/sonoff4chpror2/POWER3 = OFF
21:30:27 SRC: MQTT
21:30:27 RSL: Received Topic cmnd/sonoff4chpror2/POWER4, Data Size 3, Data OFF
21:30:27 RSL: Group 0, Index 4, Command POWER, Data OFF
21:30:27 MQT: stat/sonoff4chpror2/RESULT = {"POWER4":"OFF"}
21:30:27 MQT: stat/sonoff4chpror2/POWER4 = OFF
21:30:27 CFG: Saved to flash at FA, Count 482, Bytes 3584
21:30:29 APP: Button1 level 0-1
21:30:29 SRC: Button
21:30:29 MQT: stat/sonoff4chpror2/RESULT = {"POWER1":"OFF"}
21:30:29 MQT: stat/sonoff4chpror2/POWER1 = OFF
21:30:29 APP: Button2 level 0-1
21:30:29 SRC: Button
21:30:29 MQT: stat/sonoff4chpror2/RESULT = {"POWER2":"OFF"}
21:30:29 MQT: stat/sonoff4chpror2/POWER2 = OFF
21:30:29 CFG: Saved to flash at F9, Count 483, Bytes 3584
````
Output of command ``status 0`` :
```
21:37:13 CMD: status 0
21:37:13 SRC: WebConsole from XXXXX
21:37:13 RSL: Received Topic /status, Data Size 1, Data 0
21:37:13 RSL: Group 0, Index 1, Command STATUS, Data 0
21:37:13 MQT: stat/sonoff4chpror2/STATUS = {"Status":{"Module":23,"FriendlyName":["4ch","4ch-2","4ch-3","4ch-4"],"Topic":"sonoff4chpror2","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T15:11:00","StartupUTC":"2018-07-26T05:26:13","Sleep":0,"BootCount":21,"SaveCount":483,"SaveAddress":"F9000"}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-07-24T22:30:46","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hassio",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS4 = {"StatusMEM":{"ProgramSize":513,"Free":488,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","B7B61F9E","00000000"]}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS5 = {"StatusNET":{"Hostname":"sonoff4chpror2-7479","IPAddress":"192.168.1.110","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:A1:9D:37","Webserver":2,"WifiConfig":5}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A19D37","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS7 = {"StatusTIM":{"UTC":"Thu Jul 26 20:37:13 2018","Local":"Thu Jul 26 21:37:13 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:15","Sunset":"20:37"}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS10 = {"StatusSNS":{"Time":"2018-07-26T21:37:13"}}
21:37:13 MQT: stat/sonoff4chpror2/STATUS11 = {"StatusSTS":{"Time":"2018-07-26T21:37:13","Uptime":"0T15:11:00","Vcc":3.475,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"hassio","RSSI":74,"APMac":"C8:B3:73:15:2B:01"}}}

```

alekseyn1, 2018-07-27:
Thank you. I had to uninstall and reinstall MQTT to wipe all of the info. Looks like it works now.
It would be great to have a button or something in the HA console to do the task of deleting a database of MQTT as I found no reliable instructions of doing it via SSH on hass.io on raspberry pi.

Frogmore42, 2018-07-27:
You would have to ask that on the HA issues list or forum for the button.
I have never done it, but you should just be able to stop MQTT and then delete its database.  I don't believe HA could do that easily.  So, perhaps the question/request is better directed to the broker provider.

Since your issue is resolved, please close it now.

alekseyn1, 2018-07-27:
Thanks for the support!

closed 2018-07-27 by alekseyn1