arendst/Tasmota #3340: Get current power consumption on POW R2 trough web API

evlo, 2018-07-26:
Hi,

I'm trying to get current power consumption trough web command on POW R2 and can't find a way (nor it is showin in information in web interface)

I tried this http://192.168.1.12/cm?cmnd=Status&payload=8 among other things and did not get anywhere

Is it possible?

```
{"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
```

evlo, 2018-07-26:
Jup, now I see in the web it shows up as "Sonoff Basic Module" but it is real POW R2 :)

Frogmore42, 2018-07-26:
I believe this addresses your issue.  If so, please close this.  If not, please provide all the information requested in the template.

closed 2018-07-26 by evlo