arendst/Tasmota #3342: How to flash a Generic DHT11 ESP01s sensor

vaidyasr, 2018-07-27:
Tried to flash the Generic DHT11 ESP01s sensor and after flashing, I don't know how to configure it. There is no WiFi broadcast-ed from the device. Any help?

I have flashed via esptool using ESP to USB adapter using flash mode.

Frogmore42, 2018-07-27:
Serial log?

Frogmore42, 2018-07-27:
Also, the DHT11 is probably the lowest quality sensor you can buy.  It is not very accurate, nor is it very repeatable for humidity.  For temperature, it has low resolution and accuracy.  I'm not convinced that it is actually better than nothing.  At least with nothing, you know you are guessing what temperature and humidity it really is.  With the DHT11, you might actually think you know and then make decisions on that "knowledge".  Those choices would likely be wrong/bad.

curzon01, 2018-07-27:
Use [Button functionlity](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage) to set an alternate mode you want (WPS - 4 short presses, AP mode - 5 short presses)

On ESP01 use GPIO0 tapped on GND several times to get the mode - I tap the GPIO0 wire in 1/2 second interval and I did that successfully several times in past.

vaidyasr, 2018-07-28:
Not working, tried 4 and 5 short button presses.

vaidyasr, 2018-07-28:
I have selected DHT11/ESP01s, as it was already soldered and sold as a single piece, instead of doing on my own.

Frogmore42, 2018-07-28:
Get a wemos and a shield
https://wiki.wemos.cc/products:d1_mini_shields:sht30_shield

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2