arendst/Tasmota #3344: rule 'once' option does not work for me

qorron, 2018-07-27:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff TH16
- [x] Provide the output of command ``status 0`` :
```
tasmota/TH16_1/cmnd/status 0
tasmota/TH16_1/stat/STATUS {"Status":{"Module":4,"FriendlyName":["TH16_1"],"Topic":"TH16_1","ButtonTopic":"group1","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
tasmota/TH16_1/stat/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"group1","OtaUrl":"http://192.168.2.2:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"1T20:41:32","StartupUTC":"2018-07-25T12:43:10","Sleep":0,"BootCount":8,"SaveCount":29,"SaveAddress":"F7000"}}
tasmota/TH16_1/stat/STATUS2 {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-21T10:12:13","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
tasmota/TH16_1/stat/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"192.168.2.2","LogPort":514,"SSId":["wald","Twilight"],"TelePeriod":300,"SetOption":["00008229","55818000"]}}
tasmota/TH16_1/stat/STATUS4 {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F50A590","00000001","00040096","00000000"]}}
tasmota/TH16_1/stat/STATUS5 {"StatusNET":{"Hostname":"TH16_1-7687","IPAddress":"192.168.2.108","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"5C:CF:7F:9F:1E:07","Webserver":2,"WifiConfig":3}}
tasmota/TH16_1/stat/STATUS6 {"StatusMQT":{"MqttHost":"192.168.2.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F1E07","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
tasmota/TH16_1/stat/STATUS7 {"StatusTIM":{"UTC":"Fri Jul 27 09:24:42 2018","Local":"Fri Jul 27 11:24:42 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:17","Sunset":"21:36"}}
tasmota/TH16_1/stat/STATUS10 {"StatusSNS":{"Time":"2018-07-27T11:24:42","DS18B20":{"Temperature":23.7},"TempUnit":"C"}}
tasmota/TH16_1/stat/STATUS11 {"StatusSTS":{"Time":"2018-07-27T11:24:42","Uptime":"1T20:41:32","Vcc":3.209,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wald","RSSI":70,"APMac":"F0:9F:C2:FA:71:8B"}}}
```

this is my test setup:

```
tasmota/TH16_1/tele/STATE {"Time":"2018-07-27T11:21:47","Uptime":"1T20:38:37","Vcc":3.211,"POWER":"ON","Wifi":{"AP":1,"SSId":"wald","RSSI":70,"APMac":"F0:9F:C2:FA:71:8B"}}
tasmota/TH16_1/tele/SENSOR {"Time":"2018-07-27T11:21:47","DS18B20":{"Temperature":23.6},"TempUnit":"C"}
tasmota/TH16_1/cmnd/power 0
tasmota/TH16_1/stat/RESULT {"POWER":"OFF"}
tasmota/TH16_1/stat/POWER OFF
tasmota/TH16_1/cmnd/rule1 1
tasmota/TH16_1/stat/RESULT {"Rule1":"ON","Once":"ON","Free":426,"Rules":"on DS18B20#Temperature>25 do power 0 endon on DS18B20#Temperature<24 do power 1 endon"}
tasmota/TH16_1/stat/RESULT {"POWER":"ON"}
tasmota/TH16_1/stat/POWER ON
tasmota/TH16_1/stat/RESULT {"POWER":"ON"}
tasmota/TH16_1/stat/POWER ON
tasmota/TH16_1/stat/RESULT {"POWER":"ON"}
tasmota/TH16_1/stat/POWER ON
tasmota/TH16_1/stat/RESULT {"POWER":"ON"}
tasmota/TH16_1/stat/POWER ON
tasmota/TH16_1/stat/RESULT {"POWER":"ON"}
tasmota/TH16_1/stat/POWER ON
tasmota/TH16_1/stat/RESULT {"POWER":"ON"}
tasmota/TH16_1/stat/POWER ON
.....
```
final goal is:
th16 with a DS18B20 attached to a PSU it controls.
the rule should turn the power off if a certain temperature is reached to prevent the PSU from overheating.

this setup (to toy around with rules):
th16 with a DS18B20 on my desk and a LED light bulb attached to it.

the 'once' option appears to be set and yet, the power on message is repeated every second.

what am I doing wrong?

**(Please, remember to close the issue when the problem has been addressed)**
will do. promise!

arendst, 2018-07-27:
See development changelog. Fixed in 6.1.1b

qorron, 2018-07-27:
yes, that. I just found the closed bug #3226. thanks

closed 2018-07-27 by qorron

vel2000, 2018-09-05:
I think it's not fixed.
Using Sonoff-Tasmota 6.2.0.1 and this message repeats every second

``09:37:14 RUL: DS18B20#TEMPERATURE<32 performs "power1 on"
09:37:14 RSL: RESULT = {"POWER1":"ON"}
09:37:14 RSL: POWER1 = ON``

rule is: 
`rule on DS18B20#Temperature>35 do power1 off endon on DS18B20#Temperature<32 do power1 on `endon``