arendst/Tasmota #3347: Support for Sonoff B1 R2

bipsendk, 2018-07-27:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X] Hardware used :  Sonoff B1 R1
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

Instructions for how to flash Sonoff B1 R2 desired. The PCB has apparently changed layout, 3.3V, GND RX and TX are clearly visible. Required remaining pin(s) need to be identified, and circuit maybe examined to find out whether the existing B1 flavor of the firmware can be used.

Image of the PCB has been attached. 
 
[image]

Jason2866, 2018-07-27:
Here is a pinout of the ESP8266 IC
[image]

 just check with a ohmmeter if the GPIO0 pin goes to a solder point.

Jason2866, 2018-07-27:
If you found all pin just flash Tasmota and see if it works.
If you save the original flash before you can flash it back if Tasmota doesnt work...

hannesfischer, 2018-08-01:
If you look closely there are too more solder Pads next to the ESP8266 chip. One is unmarked, the other one is marked "SDA". The unmarked pin is GPIO0 (picture).
[image]

maxemoose, 2018-08-16:
I can confirm that Tasmota works fine on the Sonoff B1 R2; Just flashed three of these new PCB bulbs this morning using _exactly_ the same process that I used for the B1 R1.

1: Connect 3.3v, Tx, Rx, and Gnd to your USB-to-Serial solution.
2: Connect GPIO0 (pictured above) to Gnd **before** connecting to a computer.
_**NOTE**: The connected Gnd can be located wherever you can find one. (I.E. on your USB-to-Serial solution as opposed to the B1 PCB). Choose a Gnd location that makes this process as easy as possible._
3: Plug the USB-to-Serial solution into a computer.
3: Count to 10 and then disconnect GPIO0 from the Gnd.
4: Flash Tasmota using whichever solution works for you.

Managed to do this without using any solder or extra hands!