arendst/Tasmota #3348: Question:Sending MQTT Sensor on Value Change or Value hits threshold

helmar74, 2018-07-27:
Hi,

just a short question.
Is it possible to send a value of a connected sensor, when it's value changes.
Two examples:
A 18B20 1wire temperature sensor is connected to a Tasmota Device.

1. Every time the temperature changes a mqtt message is send to broker.
2. If the temperature is lower or higher a (self defined) threshold a mqtt message is send to broker.

Is such a function existing in tasmota (e.g realisiation via rules)?

Thanks a lot 

helmar

Frogmore42, 2018-07-27:
Rules are very powerful, but new.  You will need to invest some time into learning them.  The wiki has a lot of examples and there are several existing issues with other examples.  I believe you could achieve why you want with rules.

But, why are you trying to minimize MQTT traffic?
MQTT is designed to be lightweight, so I don't understand what you are trying to save.

DS18B20 devices are relatively slow to respond so if you set the teleperiod 10s, you will get information much faster than you are likely to need it.  You can slow it down to 30s and probably still be fine.  Any reasonable system can handle hundreds of sensors running at 10s.  You can choose to not do anything with the data at a higher level.

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2