arendst/Tasmota #3349: WiFi still fails with Sonoff Basic R2

razor109, 2018-07-27:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Sonoff Basic R2
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

I'm still stuck running the Sonoff Basic R2 on 6.1.1. Wont connect to my Wifi with the full version. No problem on the Minimal version. Flashing Full version (fresh build with Wifi credentials same as in Minimal) makes the Sonoff not connecting. 

Error log states time out to connect to AP.

What i tried after some reading:

- change wifi network to all lowercase letters
- change CFG-HOLDER
- enabled WIFI-MANAGER
- flashed with 2.4.1 and 2.3.0 core
- flashed with Platform.io and Arduino

Installing 5.14 fixes the problem connects then without a problem.

Config:
#define STA_SSID1              "UNKNOWN"                // [Ssid1] Wifi SSID
#define STA_PASS1              "removed"                // [Password1] Wifi password
#define STA_SSID2              "unknowniot"                // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "removed"                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_MANAGER    // [WifiConfig] Default tool if wifi fails to connect

Log gives only a connection to AP1 and AP2 failed, time out. Starting Wifi network but then the wifi network is not being found by either my android phone of windows laptop. 

Any last advice/tips or should i just stick to 5.14 and hope a new version fixes the wifi trouble?

Frogmore42, 2018-07-27:
Since it works with 5.14 and also works with 6.1.1 Minimal, that seems to indicate that it is probably something related to you environment and the state of Wi-Fi in the core.  There is a new release of the core scheduled for the next week or so, but that would need to get integrated/made available from PlatformIO.  Until that happens, your choice is simple.

1. Use 5.14, since it works for you

1. Try some things to see if you can make it work with 6.1.1.  I haven't updated to the 6.x released code yet, since it seems people are having more issues with it than the older stuff.  If you NEED something that is only in 6.x, then you will need to figure out how to make it work in your environment.  I would start by disabling things until you get it working.  I would start by disabling Hue Emulation, it has caused problems for people in the past

Jason2866, 2018-07-28:
Maybe it is the change in hardware design -> power supply is changed.
@arendst is thinking in this direction. With v.6.x the ESP chip needs more current than in earlier versions...
It needs more bugtracking what exactly causes this issue. In general v.6.x works on all devices.
But some devices fail.

wasi54, 2018-07-28:
Just want to let you know that v6.1.1a works fine on my side.
But with v6.1.1b, Wifi connections do not appear on devices with poor signal (less than 90%). Some time it also fails with signal strength of 100%.
Furthermore there are reboots by watchdog.
The difference i found is between these two versions:
[image]

I compiled with 
;platform = espressif8266@1.7.3
; *** Esp8266 core for Arduino current version
;platform = espressif8266
; development espressif8266 @ d90b917
platform = https://github.com/platformio/platform-espressif8266.git  

and USE_WPS disabled.

v6.61b works also with platform 2.3.0 but not with higher versions.

I tryed to analize the source, but i am still a beginner :-(, sorry

Frogmore42, 2018-07-28:
Core 2.4.0 is known to be bad and 2.4.1 is known to have strange issues on some devices/environments. 2.4.2, which is not released yet, is supposed to address them.  I have heard good things about it in other projects, but I don't know how it will work with Tasmota.

wasi54, 2018-07-30:
For compiling tasmota 6.1.1c  i suggest the **bold** settings. Wifi is stable on sonoff basic, TH, Dual and Dual V2

**platform = espressif8266@d90b917**

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  **-DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH**

[image]

arendst, 2018-07-30:
See #2745

BladeRunner68, 2018-08-01:
Is there a known issue with the Sonoff Basic devices now?

ALL my Sonoff Basic devices are failing when upgrading from 5.13 to either 5.14 or 6. 

Pinging the devices responds in very random OK's and the web interface is not reachable at all on any of them.

No problem on S20 or TH devices.  

I'm going to have to rethink how I am using these basics as one is under my floorboard controlling my hall light and it is a real ball-ache getting to it 

:(

Frogmore42, 2018-08-01:
Anytime you have devices deployed in a location that is hard to reach, you need to be extremely cautious about why, when, what and how you update them.  People generally only learn this after they have experienced the consequences of not doing that.

Is there a particular reason you decided to update them?
I have devices that are still running on 5.12c.  They are working as well as they ever did.  At least one is in a somewhat difficult location to access (need a ladder).

I am waiting till core 2.4.2 comes out and has proven to be good before updating anything, since there have been lots of people who have experienced issues with 2.4.0 and 2.4.1.  The 6.x versions have some interesting new capability, but I don't need that in my already deployed devices, so I am staying with the version that has proven to be stable.

I do have an issue with most/all of my devices not syncing with NTP after being powered without a connection to the internet for a long period.  This happens every time the power goes out.  The generator starts up 30s later, but if the internet connection is gone, there is no NTP server.  Hours later, the internet comes back, but it seems the underlying NTP code has given up trying to get the correct time.  This is not an issue with the Tasmota code as the call it makes returns 0 for the time.  I am hoping that the change to lwip 2 will address this issue, but so far it has come at the cost of not being very reliable.  I have heard that core 2.4.2 with the high bandwidth option makes things much better in general.

BladeRunner68, 2018-08-01:
well i have managed to update the hidden one to 5.14 minimal build.   the wifi stayed connected long enough to upload. For my use case i can just use the minimal build as I just trigger the  relay on or off via a web url call.
this is the last time i update because as you suggest, if it ain't broke...

Frogmore42, 2018-08-01:
@BladeRunner68 not really.  I assume you mean heap when you say 20k.  That is an okay value.  The non-minimal build has a lot of stuff running.  Hue emulation has been known to cause issues in some locations/networks, but works fine in others.  The amount of activity, noise, and other factors likely play into the stability (or lack thereof).  Some people have not had good luck even with the minimal build.

If all you are doing is turning on/off the relay and don't need to persist any settings, the minimal will probably work okay.  But, it was really designed to be small enough to load and allow loading a bigger than 50% size build.

Looking at the Arduino esp8266 project all the 2.4.2 issues have been closed:
https://github.com/esp8266/Arduino/milestone/13?closed=1

Today is the day that 2.4.2 is supposed to release, so it may be available for usage soon.  I don't think I will have time to try it till the weekend.  But from the list of issues that have been fixed, it sounds like it will be something quite positive.

himswim28, 2018-08-06:
I had a similar issue.  I compiled with 6.1.1 and Arduino IDE 1.8.4 with #define STA_SSID1 and STA_PASS1 set in user_config.h and serial flashing to a new sonoff basic, that never had tasmota installed (tried 3 different basics) and all failed to connect; the same as parents issue, serial monitor showed the correct SSID name, but timed out. It would then show it going to AP mode for 90 seconds, but no wireless AP showed up on windows to attempt to connect to the basic.  I had 3 other basics, with identical SSID settings in the user_config.h and using 5,14.0 they had no issues. When those basics were upgraded from 5.14.0 to 6.1.1 with the exact same bin file as failed on the new basics, they all connected immediately to my wireless AP without issues. When I took the 3 new Basics and flashed them with 5.14.0 and the hardcoded SSID1 and PASS1, then updated them to 6.1.1 they all connected to wifi, then I could see in serial that each complained about couldn't connect to MQTT (I had never done any configuring of any settings on these devices) it would then reboot approximately every minute. I had the same issue flashing back to 5.14.0. I then hard-coded my MQTT, and WIFI settings in the user_config.h file, and set serial settings to erase all flash, flashed 5.14.0; After this, then I could update to 6.1.1 without it rebooting.

ascillato2, 2018-08-22:
@razor109 

Have you solved your issue?

razor109, 2018-08-22:
@ascillato2

I installed 6.1.1c and now WiFi works. 

With the precompiled binairies of 6.1.1 still no luck but doesn't matter anymore!

closed 2018-08-22 by razor109

SniFManS, 2018-09-09:
I flash the Sonoff basic R2 witch Tasmota 6.2.1 and used Arduino - but no wifi is transmitting. 
Also with my wifi settings its not connecting.

Jason2866, 2018-09-09:
See #3678 and the remark from Theo at precompiled relase page.