arendst/Tasmota #3350: MCP23017 not updating status

MadDoct, 2018-07-27:
in 6.1.1c, using the mcp23017, the status of the pins don't get updated when they change although I set the pins using sensor29 7,2,1, for example...
The wiring was correct because I tried espeasy and the mcp works perfectly...

Frogmore42, 2018-07-28:
Huh?
You'll need to provide the information listed in the template, in particular, a clear description of the problem.
I certainly don't understand what is written.

The only thing I can think of, is you changed the module type and the pins definitions at the same time.  When you change the module type in the web UI, it resets the pin definitions to the default for the module.  You can/have to then change the definition for the pins.

MadDoct, 2018-07-28:
"Solved" it by enabling the pull-up resistors for all pins! Maybe that should be written in the wiki and/or set by default...

andrethomas, 2018-07-28:
`"Solved" it by enabling the pull-up resistors for all pins! Maybe that should be written in the wiki and/or set by default...`

I do not think one wants to enable pull-up by default - by default, this would be done externally by whatever is connected to the pin so the option to enable it is only for specific uses like connecting a button directly to a pin and would not be the default use for any particular user. Hence it defaults to disabled but can easily be enabled as you have done :)

MadDoct, 2018-07-28:
From what I could test, no enabling the pull-up meant that the values were constantly floating!
What other uses could you have for the mcp, the way it is implemented - as a senao -, other than switches or binary sensors?

andrethomas, 2018-07-28:
For example, I have them connected to PIR sensors that pull the pin high by default - if the PIR is triggered it pulls the pin LOW... if the wire is cut or the power removed from the PIR the pin goes LOW which would not happen if internal pull-up was enabled and lead to not detecting a PIR which wires have been snipped...

The default of the chip on power-up is input without pull-up according to the datasheet so this also needs to be taken into consideration insofar that normal usage of the chip would require you to specifically enable pull up resistors.

andrethomas, 2018-07-28:
I guess adding the pullup for `sensor29 reset` by default wouldn't hurt either

MadDoct, 2018-07-28:
Ok, that way I understand, but it should be noted in the wiki that some kind of pull-up or pull-down should be used, internal or external, because the values keep floating... Also, I had to enable the pull-up on all to start getting the intents... Enabling on that specific pin had no result!

MadDoct, 2018-07-28:
Or adding an option to enable all easily...

andrethomas, 2018-07-28:
You mean something like

```
sensor29 reset   // Reset all pins to input with pull-up disabled (datasheet default)
sensor29 reset1  // Reset all pins to input without interrupt and with pull-up enabled
sensor29 reset2  // Reset all pins to input with interrupt on change change and with pull-up enabled
sensor29 reset3  // Reset all pins to input with interrupt on LOW and with pull-up enabled
sensor29 reset4  // Reset all pins to input with interrupt on HIGH and with pull-up enabled

```

That will cost another 80 bytes of flash.

MadDoct, 2018-07-28:
That would be a great idea, but above all, there should be a heads up in the wiki and maybe an investigation to understand why the intent is only sent when all, or at least, several pins have the pull up activated...

andrethomas, 2018-07-28:
The interrupts are not intended to update the Sensor 10 command's output - The interrupts provide MQTT outputs for each instance on any given pin... So I guess I still don't understand what you mean.

MadDoct, 2018-07-28:
I know that. I only talked about status 10 to show you that the mcp was connected andthe pins were changing as they should... The problem is that, unless the pull-ups are activated on ALL the pins (or at least several), the intents aren't generating the MQTT outputs they should...

andrethomas, 2018-07-28:
I'm unable to reproduce your experience - interrupts work for me whether pull-ups are enabled or not...

MadDoct, 2018-07-28:
That's strange... Maybe a problem on my mcp... I'm using one with a breakout board... Did you wire something besides de GND, VCC, SDA and SCL? (besides the GPIOs, obviously..)

MadDoct, 2018-07-28:
Mine doesn't generate the interrupt if only one or two are used...

andrethomas, 2018-07-28:
To recreate

```
20:30:38 CMD: sensor29 reset
20:30:38 SRC: Serial
20:30:38 RSL: Group 0, Index 29, Command SENSOR, Data reset
20:30:38 MQT: stat/sonoff/RESULT = {"Sensor29":{"D":99,"MODE":1,"PULL-UP":0}}

```
This sets all pins to input without pull-up but pinmode = 1 so there's no interrupt.

I then set pin 0 to pinmode 2 (interrupt on change) without pull-up
```
20:31:18 CMD: sensor29 0,2,0
20:31:18 SRC: Serial
20:31:18 RSL: Group 0, Index 29, Command SENSOR, Data 0,2,0
20:31:18 MQT: stat/sonoff/RESULT = {"Sensor29":{"D":0,"MODE":2,"PULL-UP":0}}

```

I get nothing until I put a dupont wire into the pin on the breadboard which obviusly causes it to oscillate based on interference picked up from my finger and the absense of a pull-up or pull-down setup and produces something like this:

[image]

So this is generating interrupts on A0 on the MCP23017 and pushing it out on MQTT whilst all the other pins are still pinmode = 1 (no interrupt) and no pull-ups enabled

If I enable pull-up or externally pull up or pull down the pin then it only occurs as and when I connect to the relevant voltage line.

andrethomas, 2018-07-28:
I've added a section in the wiki to cover use of interrupts on pins which are stateless/floating.

MadDoct, 2018-07-28:
Strangely, for me, I get floating values even without Dupont wires attached... (Maybe a problem with my mcp - are you using one with a break out board or stand alone?) 
But that would only increase the number of intents... My problem is the opposite, the values change but the intents aren't producing a result... The intents only work after setting the pull-ups for all or, at least, several pins...
I'm using 6.1.1c, with core 2.3. What version are you using? Could that be the problem?

andrethomas, 2018-07-29:
> Strangely, for me, I get floating values even without Dupont wires attached... (Maybe a problem with my mcp - are you using one with a break out board or stand alone?)

Standalone ESP8266 with a MCP on breadboard. I have two, one wired for MCP23008 and one wired for MCP23017

[image]

[image]

> Did you wire something besides de GND, VCC, SDA and SCL? (besides the GPIOs, obviously..)

Yes, the reset pin on the MCP is tied to VCC (3.3V)

> But that would only increase the number of intents... My problem is the opposite, the values change but the intents aren't producing a result (maybe too many changes simultaneously?!)... The intents only work after setting the pull-ups for all or, at least, several pins...
> I'm using 6.1.1c, with core 2.3. What version are you using? Could that be the problem?
> 

I use 2.4.1 but I have tested it with 2.3.0 and 2.4.0 with the same result as expected.

andrethomas, 2018-07-29:
> (maybe too many changes simultaneously?!)

Too many would not make it not cause an interrupt - perhaps missing interrupts in between is possible as the register is only read every 50 milliseconds - The register can only report on the last interrupt which occurred, but if at least one was set in the 50 milliseconds since the last check it will still report it via MQTT - Also remember that the interrupts are not reported to telemetry data sent via MQTT based on logging interval configured for your module and will also only the current state of a particular pin (not the interrupt) on telemetry data and/or status 10. Interrupts are ONLY reported via MQTT output as follows:

`13:23:44 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T13:23:44","MCP230XX_INT":{"Pin":"D0", "State":1}}`

andrethomas, 2018-07-29:
For a more scientific approach I did this as a test

`sensor29 0,2,0`

Then create an arduino UNO (5V) with the following sketch and connected D2 output pin to pin 0 on mcp23017

```
void setup() {
  pinMode(2,OUTPUT);
}

void loop() {
  digitalWrite(2,LOW);
  delay(1000);
  digitalWrite(2,HIGH);
  delay(1000);
}
```

I get the expected output on the Tasmota firmware:

```
14:14:00 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:14:00","MCP230XX_INT":{"Pin":"D0", "State":0}}
14:14:01 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:14:01","MCP230XX_INT":{"Pin":"D0", "State":1}}
14:14:02 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:14:02","MCP230XX_INT":{"Pin":"D0", "State":0}}
14:14:03 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:14:03","MCP230XX_INT":{"Pin":"D0", "State":1}}
14:14:04 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:14:04","MCP230XX_INT":{"Pin":"D0", "State":0}}
14:14:05 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:14:05","MCP230XX_INT":{"Pin":"D0", "State":1}}

```

Lets try interrupt on LOW
`sensor29 0,3,0`

```
14:16:14 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:16:14","MCP230XX_INT":{"Pin":"D0", "State":0}}
14:16:16 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:16:16","MCP230XX_INT":{"Pin":"D0", "State":0}}
14:16:18 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:16:18","MCP230XX_INT":{"Pin":"D0", "State":0}}
14:16:20 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:16:20","MCP230XX_INT":{"Pin":"D0", "State":0}}
14:16:22 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:16:22","MCP230XX_INT":{"Pin":"D0", "State":0}}

```

Goes LOW every 2 seconds as expected and interrupt is reported via MQTT

Lets try interrupt on HIGH
`sensor29 0,4,0`

```
14:17:33 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:17:33","MCP230XX_INT":{"Pin":"D0", "State":1}}
14:17:35 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:17:35","MCP230XX_INT":{"Pin":"D0", "State":1}}
14:17:37 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:17:37","MCP230XX_INT":{"Pin":"D0", "State":1}}
14:17:39 MQT: stat/sonoff/RESULT = {"Time":"2018-07-29T14:17:39","MCP230XX_INT":{"Pin":"D0", "State":1}}
```

Again, as expected, interrupt provided via MQTT on HIGH

In all these cases the pull-up was disabled as the Arduino UNO dictated the state of the pin.

andrethomas, 2018-07-29:
Hmm, hang on, I think I found something weird - will report back

andrethomas, 2018-07-29:
@MadDoct Please change line 149 in xsns_29_mcp230xx.ino from
switch (Settings.mcp230xx_config[+(mcp230xx_port*8)].pinmode) {
to
switch (Settings.mcp230xx_config[intp+(mcp230xx_port*8)].pinmode) {

It was missing the intp picked up from the for loop which would have caused the code to incorrectly reference the pin configuration. This would cause int to only work if it was set in the corresponding bit for portB on the MCP.

I've made a PR for it here https://github.com/arendst/Sonoff-Tasmota/pull/3369 but you may apply the change directly until the PR is committed to development.

Please test again after and revert back - Thanks for the testing so far... without it I would not have detected the error as my "working as a device" where the PIR's are connected have all pins to pinmode 2.

andrethomas, 2018-07-29:
PR committed to development.

MadDoct, 2018-07-29:
I'm happy to confirm that it is now working flawlessly!
Glad I could help and thanx for the fix! ;)

MadDoct, 2018-07-29:
I made the change in my compilation and will try...

MadDoct, 2018-07-29:
tested and working... now to try rules...

andrethomas, 2018-07-29:
Change 164 to

`snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s,\"MCP230XX_INT\":{\"D%i\":%i}"), mqtt_data, intp+(mcp230xx_port*8), ((mcp230xx_intcap >> intp) & 0x01));`

Should work - have an example rule?

MadDoct, 2018-07-29:
I changed it by myself before and it works as expected, but the truth is that it isn't necessary when using rules because rules don't care about the MCP2300XX_INT, they just care if D0 is 1 or 0 (MCP2300XX) (when I asked you to change it I wasn't thinking clearly!)
By the way, I was able to use in rules to activate a relay successfully using:
rule on mcp230xx#d0=0 do power1 on endon on mcp230xx#d0=1 do power1 off endon
rule 5
It works like a charm! ;)
Now we really have additional inputs!!!! ;)

andrethomas, 2018-07-29:
Cool, well done and thanks again for testing.

MadDoct, 2018-07-29:
Can that driver be used to have some pins as input and some as output?

andrethomas, 2018-07-29:
Yup - it just adds pinmode 5 (for output) and sensor29 commands 
`sensor29 pin,ON`
`sensor29 pin,OFF`

The rest is the same as the one existing in the development branch here.

You can conditionally include or exclude it using the #define USE_MCP230xx_OUTPUT compiler directive in the config file

closed 2018-07-29 by MadDoct

MadDoct, 2018-07-29:
Just tested it and it works like a charm! Thanx!

andrethomas, 2018-07-29:
Good stuff!

Also, earlier change proposed has been commited to development and mcp230xx_output
```
snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s,\"MCP230XX_INT\":{\"D%i\":%i}"), mqtt_data, intp+(mcp230xx_port*8), ((mcp230xx_intcap >> intp) & 0x01));
```

andrethomas, 2018-07-29:
The rules only respond to the pin state reported by telemetry (same as status 10) - I need to find a way to publish the interrupt in such a way that it can be picked up by rules then it will work, I think.

MadDoct, 2018-07-29:
No, the rules work perfectly as is! The rule I put above worked very well... So, as trigger, it is perfect. The problem is using the outputs in an action using rules...

MadDoct, 2018-07-29:
The rule would probably be best used inverted, I mean on when 0 and off when 1... ;)

andrethomas, 2018-07-29:
Yeah, damn chinese with their inverted relays :)

MadDoct, 2018-07-29:
Could you add a third option? Besides on and off, there could also be a toggle... That would be perfect!

andrethomas, 2018-07-29:
Good idea, will work on it during the week... off to bed now 11:32pm :|

MadDoct, 2018-07-29:
I didn't mean the relays... If you activate the pull-ups, 1 will be the base and 0 will be the "activated" for the switch...

andrethomas, 2018-07-29:
Gotcha

MadDoct, 2018-07-31:
I've been doing some tests and have several sugestions for the output mode:
1 - When you disconnect the nodemcu from power and reconnect (not a software restart), the output pins always start LOW and keep LOW independently of how they were before... That's pretty bad if you're using active low relays...
2 - The telemetry should be changed to indicate that the pin is being used as output. The way it is, it makes the telemetry about that pin totally irrelevant and maybe misleading... Currently there's no info about that, unless you query the specific pin.
3 - Currently there's no way to know if the pin is on or off. The command sensor29 1,?, should return, if the pin is an output, the mode and on or off...

andrethomas, 2018-07-31:
Sure, all in good time - remember output mode is not officially supported by Tasmota.

I've added sensor29 pin,T for toggle to the mcp230xx_output fork but  have not yet tested it as I do not have hardware with me where I am right now.

The other stuff you mention is on the todo list... just not a priority as only the input is formally part of the Tasmota development codebase... output is still experimental and there are other things I am also adding which won't even require you to use rules to execute commands... but all in good time.

andrethomas, 2018-07-31:
@MadDoct Response has been update to include pin state with PR https://github.com/arendst/Sonoff-Tasmota/pull/3384

Change already comitted to mcp230xx_output

andrethomas, 2018-07-31:
@MadDoct 

To address your issue on output pins being low on startup for the mcp230xx_output fork I dual purposed the pullup option to set the default reset/power-up state of a pin.

This means
```
sensor29 0,5,0  // Pin will be low on reset/power-up
sensor29 0,5,1  // Pin will be high on reset/power-up
```

Trust that solves your inverted relay problem... so you can do `sensor29 pin,5,1` to specify that the pin will go high on reset/power-up

Note this is only applicable to the mcp230xx_output branch from here https://github.com/andrethomas/Sonoff-Tasmota/tree/mcp230xx_output

andrethomas, 2018-08-01:
@MadDoct Added PR https://github.com/arendst/Sonoff-Tasmota/pull/3391 which allows you to write rules that will respond immediately to interrupts and not based on the output provided by telemetry JSON data.

andrethomas, 2018-08-01:
@MadDoct Nah, it was getting it from the telemetry json...

See comments from Theo on the previous commit https://github.com/arendst/Sonoff-Tasmota/pull/3370

andrethomas, 2018-08-01:
> What I think is necessary, if you can, is to add a way for the output to be the same it was before powering off..

I will add code to the output branch/fork to adhere to the setoption0 setting, also defined in your config file as 
`#define SAVE_STATE             1`

MadDoct, 2018-08-01:
Believe me, it was working immediately! I've tested it allot!
Thanx for the latest commit! And for all of them!!!

andrethomas, 2018-08-01:
Cool, anyway the SAVE_STATE setting is used globally by the Tasmota firmware to save the state of relays configured the traditional way without using the MCP so its only logical that the output fork should adhere to this... it will save the state of the pin each time it changes, when it powers up it will first consult the default setting substituted by the pullup parameter and then look at the savestate if it is enabled and apply accordingly.

So saved state will always override default setting but if saved state is not enabled in the firmware (it is by default anyway) it will use the default set by the pullup parameter.

Will report back once I've implemented it on the output branch

andrethomas, 2018-08-01:
@MadDoct https://github.com/andrethomas/Sonoff-Tasmota/tree/mcp230xx_output now adheres to save_state / setoption0

MadDoct, 2018-08-03:
I've been testing and now, for example, 7,2,1 isn't activating the pull-up althout it replies as pull-up:1, the pin fluctuates...

MadDoct, 2018-08-03:
I'm probably wrong, but why would you put this? Is this the problem?
#if**n**def USE_MCP230xx_OUTPUT
      if (Settings.mcp230xx_config[idx+(mcp230xx_port*8)].pullup) {
        reg_gppu |= (1 << idx);
      }
#endif

Although USE_MCP230xx_OUTPUT is defined, it doesn't mean that we don't want to use input...
(My knowledge of c++ is VERY limited...)

andrethomas, 2018-08-03:
Apparently, your C isn't as limited as you think :)

Will make a PR to fix shortly.

andrethomas, 2018-08-03:
The defined caused the pre-processor to completely disable pull-up on all commands whereas it should be selective on whether or not a specific pin is in output mode (pinmode 5)

So we want pull-ups to work but not for pinmode 5

Grab a fresh copy of https://github.com/andrethomas/Sonoff-Tasmota/blob/mcp230xx_output/sonoff/xsns_29_mcp230xx.ino

Should be fixed now.

Nice find, Thanks!

MadDoct, 2018-08-03:
It would probably be a good idea to set pinmode 0 when you reset...

andrethomas, 2018-08-03:
> It would probably be a good idea to set pin mode 0 when you reset...

Reset adheres to datasheet's default power-up which is all pins floating input without the pull-up.

MadDoct, 2018-08-03:
Also, I added to my code two more reset modes, to start them all with pull-up enabled and another one to set pinmode 0 for all...
I've noticed that, in practive, pinmode 1 and 0 are the same, because it keeps returning the state of the pin even when set to 0, when you use Status 10

andrethomas, 2018-08-03:
Correct - 0 was originally used to "disable" a pin but has since been deprecated since not needing to use a pin having it in pin mode 1 is all the same anyway.

[46 more replies]