arendst/Tasmota #3351: Can't acess webserver and random disconnect

grandchamp, 2018-07-28:
Hi.
I'm experiencing a weird problem with 2 different Sonoff Basic. I have a sonoff connected to a power plug that connects to a lamp. I use it to control the light.
I had configured both Sonoffs on RaspberryPI and everything runs ok (looking at the logs), but when i plug it on the power plug, the device keeps restarting and don't serve the webserver (i even tried to ping, and the result was more than 70% of packet loss). Sometimes the webserver shows up on browser, but then turns offline again.

I've managed to get one info output:
```

Program Version | 6.1.1
-- | --
2018-07-23T11:18:24
2_3_0/1.5.3(aec24ac9)
0T00:03:59
37 at F7000
22
Power on
Luz Escada
 
MY SSID (56%)
luzescada-6332
192.168.0.17
192.168.0.1
255.255.255.0
187.2.216.34
B4:E6:2D:15:78:BC
 
xxxx.duckdns.org
1883
DVES_1578BC
DVES_USER
luzescada
sonoffs
cmnd/luzescada/
 
None
Enabled
Web Server
 
1407164
1327198
1024kB
1024kB
537kB
464kB
14kB
```

Then, i get this same Sonoff and put it on the power plug near the router (Wifi 100%) and it looks like it is running fine.
I really don't know what is the problem.

Frogmore42, 2018-07-28:
Based on the information you provided, the simple answer is that it doesn't work unless the signal strength is good.

You didn't answer any of the questions from the template, nor provide the details of what you are running.  There are many things that could be causing the issue you are seeing.  It will require effort to figure out, but it looks like you might be having the same issue that many other reports are having.

grandchamp, 2018-07-28:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : Sonoff Basic
- [ ] Provide the output of command ``status 0`` :
```
12:01:44 SRC: Serial
12:01:44 RSL: Group 0, Index 1, Command STATUS, Data 0
12:01:44 MQT: stat/luzentrada/STATUS = {"Status":{"Module":1,"FriendlyName":["Luz Entrada"],"Topic":"luzentrada","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:01:44 MQT: stat/luzentrada/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:01","StartupUTC":"2018-07-28T10:59:43","Sleep":0,"BootCount":16,"SaveCount":39,"SaveAddress":"F5000"}}
12:01:44 MQT: stat/luzentrada/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-23T11:18:24","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:01:44 MQT: stat/luzentrada/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":4,"SysLog":3,"LogHost":"","LogPort":514,"SSId":["Grandchamp",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
12:01:44 MQT: stat/luzentrada/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","23B6179E","00000000"]}}
12:01:44 MQT: stat/luzentrada/STATUS5 = {"StatusNET":{"Hostname":"luzentrada-1306","IPAddress":"192.168.0.15","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"187.2.216.34","Mac":"DC:4F:22:2C:C5:1A","Webserver":2,"WifiConfig":3}}
12:01:44 MQT: stat/luzentrada/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.duckdns.org","MqttPort":1883,"MqttClientMask":"DVES_55623","MqttClient":"DVES_55623","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:01:44 MQT: stat/luzentrada/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 28 11:01:44 2018","Local":"Sat Jul 28 12:01:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:18","Sunset":"20:35"}}
12:01:44 MQT: stat/luzentrada/STATUS10 = {"StatusSNS":{"Time":"2018-07-28T12:01:44"}}
12:01:44 MQT: stat/luzentrada/STATUS11 = {"StatusSTS":{"Time":"2018-07-28T12:01:44","Uptime":"0T00:02:01","Vcc":3.064,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX","RSSI":100,"APMac":"BC:2E:48:D2:BE:FC"}}}
12:02:00 MQT: tele/luzentrada/UPTIME = {"Time":"2018-07-28T12:02:00","Uptime":"0T00:02:17"}
```

grandchamp, 2018-07-28:
Ok. I've connected to my raspberry downstairs to see what happens.
So far, the miniterm is full of:

```
13:02:18 DNS: Initialized
13:02:18 MQT: Attempting connection...
13:02:18 MQT: Connected
13:02:18 MQT: tele/luzentrada3/LWT = Online (retained)
13:02:18 MQT: cmnd/luzentrada3/POWER =
13:02:31 DNS: Initialized
13:02:31 MQT: Attempting connection...
13:02:31 MQT: Connected
13:02:31 MQT: tele/luzentrada3/LWT = Online (retained)
13:02:31 MQT: cmnd/luzentrada3/POWER =
13:02:43 MQT: Attempting connection...
13:02:43 MQT: Connected
13:02:43 MQT: tele/luzentrada3/LWT = Online (retained)
13:02:43 MQT: cmnd/luzentrada3/POWER =
13:02:44 DNS: Initialized
13:02:55 DNS: Initialized
13:02:55 MQT: Attempting connection...
13:02:55 MQT: Connected
13:02:55 MQT: tele/luzentrada3/LWT = Online (retained)
13:02:55 MQT: cmnd/luzentrada3/POWER =
13:03:06 WIF: Connecting to AP1 XX in mode 11N as luzentrada3-1306...
13:03:10 WIF: Connected
13:03:10 DNS: Initialized
13:03:10 MQT: Attempting connection...
13:03:11 MQT: Connected
13:03:11 MQT: tele/luzentrada3/LWT = Online (retained)
13:03:11 MQT: cmnd/luzentrada3/POWER =
```

And the same thing happens, the web ui can't be accessed.

grandchamp, 2018-07-28:
It looks like that with the -KNX version on release it worked. I dont know why.

closed 2018-07-28 by grandchamp