arendst/Tasmota #3354: HTTP/Web UI Authentication

joearkay, 2018-07-28:
Hi,

Is web ui auth in the pipeline? It would be nice to be able to lock this down with a username and password. 

Further, it would be great to have http basic auth for the http api. This would provide a good level of protection as well as allow simpler devices to still function under basic auth.

curzon01, 2018-07-28:
use url auth - it's on same unsecure level as using http basic auth (user password is plain readable) - the difference is only the basic auth is visible in data stream instead of url - this is apparently more secure. For TLS the device is to small to implement.

ascillato2, 2018-07-29:
Hi. It is already an admin password and if you define it, that need to be used for web commands. Please read the wiki

closed 2018-07-29 by ascillato2

joearkay, 2018-07-30:
I've worked this out and updated the page above. Thanks for the comments :)

ascillato, 2018-07-30:
Hi @joearkay

Sorry about that. A lot of people left theirs issues opened after receiving answers. There are lots of issues opened and that makes very difficult to give a following up to every each one so as to solve Tasmota software bugs. So, when there is a little of time, we try to solve and close as many issues as possible. Sorry if the time between the answer and the closing was not enough for you.

Happy to know that your issue is solved.