arendst/Tasmota #3355: AP Mode Don't provide IP address

romelemperado, 2018-07-28:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : SOnoff basic
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

I flashed the latest sonoff.bin and everything was okay but I cannot connect to the AP(sonoff-xx) AP it is stuck on obtaining IP address. I did several restart and also reflashed the firmware but no luck...

Any hint about this problem?

Jason2866, 2018-07-28:
You didnt gave any useful information to help.
With version do you use? Self compiled? Release version? Which version of module?
There are many similar issues here. Try a search. Look for wifi problems.

romelemperado, 2018-07-28:
HI @Jason2866 sorry for the lack of information provided.. I've edited my post..

Im using the sonoff basic and flashed it using the bin file v6.1.1.. I've browsed  several wifi problems in the issues tab but I cannot seem to find similar topic which is stuck on obtaining IP address.. The AP is not giving me IP Address..

romelemperado, 2018-07-28:
`Romels-Air:esptool-master romel$ ./esptool.py erase_flash
esptool.py v2.5.1-dev
Found 2 serial ports
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:af:ea:b7
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.8s
Hard resetting via RTS pin...`

`Romels-Air:esptool-master romel$ ./esptool.py write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.5.1-dev
Found 2 serial ports
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:af:ea:b7
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 550112 bytes to 374770...
Wrote 550112 bytes (374770 compressed) at 0x00000000 in 33.0 seconds (effective 133.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

Still no luck....Stuck in obtaining ip address.. I even tried on my laptop same problem..

[image]

romelemperado, 2018-07-28:
Update firmware v5.12.0 20180209 worked I can not reach the web config page..

Jason2866, 2018-07-28:
Do you use arduino core 2.3.0? If not i know where your problems do come from.
Versions 2.4.x are known to make wifi problems....

romelemperado, 2018-07-28:
Hi

Im only using Esptool and nothing more.. Im not sure What is arduino core
because I can't remember using that..

Thanks,
Romel

Jason2866, 2018-07-28:
Which version do you have downloaded? Latest from this site? Please provide exact info what file do you use.

romelemperado, 2018-07-28:
Hi

This is the summary 👍 
- flashed sonoff.bin(v6.1.1) using Esptool to SOnoff basic R2 v1.0 and I cannot obtain IP from AP
- Flashed sonoff.bin(v5.12.0) everything works fine now.

yes files were downloaded in this repo

akpuggy, 2018-07-28:
I had similar AP problems with the IP address not getting assigned. As a workaround to the issue, I used my phone to configure custom IP (192.198.4.1) for the device and it connected well.

ascillato2, 2018-07-29:
Hi, I could not reproduce your issue. Have you tried as @akpuggy said?

romelemperado, 2018-07-29:
How would I make my phone configure custom IP for the device that I can't even make a connection?

Maybe he meant assign my device a static IP?

akpuggy, 2018-07-29:
My apologies, @romelemperado , I said that very wrong. So yeh, you would need to click the Sonoff Wifi SSID, go to advanced settings, Change IP Settings from DHCP to Static, and assign IP address of 192.168.4.1 and gateway as 192.168.4.0

cdonjenkins, 2018-07-31:
i have started to use Termite to set the initial values.  See:
https://www.compuphase.com/software_termite.htm
you can send a string of commands all at once setting everything you need.
Example Backlog Command for Termite to enter Tasmota Commands: 
Backlog SSID1 xxxxxx; Password1 yyyyyy; MqttHost brokerIP; MqttUser xxxxxxx; MqttPassword yyyyyyy; GPIO14 09; Hostname sonoff_name; MqttClient unique_sonoff_name; Topic sonoff_name; FriendlyName1 Sonoff-Name
Dr Zzs has a great video:
https://www.youtube.com/watch?v=KMiP9Ku71To&t=102s

Termite even returns the IP address obtained from you AP.

andrethomas, 2018-08-02:
Could not reproduce dhcp problem compiling current development codebase without changes to user_config.h onto a previously unflashed ESP8266 board.

Initial serial output on first boot:
```
0:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_102FD4, GroupTopic sonoffs) Version 6.1.1c-2_4_1
00:00:00 WIF: WifiManager active for 3 minutes
00:00:01 HTP: Web server active on sonoff-4052 with IP address 192.168.4.1
```

Then connected with phone to sonoff-xxx access point and it opened the captive page asking for wireless configuration - entered configuration and the device reconfigured and rebooted success

```
00:01:04 WIF: Hostname %s-%04d, SSId1 Wireless, SSId2 
00:01:06 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_102FD4, GroupTopic sonoffs) Version 6.1.1c-2_4_1
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-4052...
17:13:32 WIF: Connected
17:13:32 DNS: Initialized
17:13:32 HTP: Web server active on sonoff-4052.local with IP address 192.168.47.138
```

ascillato2, 2018-08-22:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-08-22 by ascillato2

stbx117, 2021-02-21:
I have the same problem with esp12-f module
mem is winbond 25vq32btig (4Mb)
flashing by tasmotizer / esptool