arendst/Tasmota #3356: Tutorial Sonoff RF Bridge 433+Synology+MQTT+Domoticz

OTiby, 2018-07-28:
Hi all,

I really need some help or a good tutorial or manual for the next combination:
- Sonoff RF Bridge 433
   - with: Tasmota
- Synology NAS DSM 6.2
   - with: Domoticz 4.97
              Mosquitto 1.4.14-19

I installed everything; Sonoff Bridge with Tasmota is working. RF remotes are working and the 16 buttons do their job.

IP-adresses (subjective)
Sonoff Bridge 192.168.1.69
NAS: 192.168.1.200

NAS-account:
account: admin
pw: 1234

Installed Mosquitto on Synology by: https://youtu.be/b3A1RJdDf-w

Sonoff MQTT parameters:
Host: 192.168.1.200
Port: 1883
User: admin
Password: 1234

Sonoff Domoticz parameters:
I really don't understand what to set here, cause what does these "sensor idx 1 "Temp" eg. mean? I want to add the Idx for Rfkey 1, 2 etc.

On the Synology NAS, in Domoticz I added:
Idx: 2
MQTT Client Gateway with LAN interface
Name: SonoffBridge-MQTT
Remote-address: localhost
Port: 1883
Username: admin
Password: 1234
Publish Topic: Out

Idx: 3
Dummy (Does nothing, use for virtual switches only) 
Name: SonoffBridge-Dummy

Virtual Sensors:
Idx: 1
Hardware: SonoffBridge-Dummy
Id: 00014051
Name: SonoffBridge-1
Type: Light/Switch
SubType: Switch

But what now?

If I go to switches in Domoticz and I turn on the light, nothing happens.
Tried to fill in all the Idx sensors in the sonoff with the Idx: 1.
How can I switch different rfkeys from domoticz?

I googled a lot, trying to find a good tutorial but cannot find anything.
I am note sure if mosquitto is still running well..
By Putty mosquitto won't accept commands, or how can I try this?
The text file of the Mosquitto youtube video of above is now different. In the past it looked the same.
But mosquitto is 'running' in the package centre of the synology.

Please help me.

R4C3R, 2018-07-28:
@OTiby In the case of the RF Bridge it works slightly different. You don't need to set the idx in the Sonoff - Domoticz config.

What you do instead is that after you have created a virtual switch in Domoticz, you look for the Switch in the Domoticz switch panel and then click "edit".

Now, you have the option to define actions on what to do when the button is pressed to turn on and turn off.

To use the different RFkeys you can use the following command:

http://ip.off.sonoff.rfbridge/cm?cmnd=RfKey1
...
http://ip.off.sonoff.rfbridge/cm?cmnd=RfKey16

So in other words, you use http commands to manage the RF-Bridge

If this is not clear enough, please let me know.

OTiby, 2018-07-28:
Yes! It is working! It was also a problem with google chrome. When I tried to edit the switch in chrome, it gave 'undefined', in Edge it is editting normally.

OTiby, 2018-07-28:
Do you have any idea how it works with a door sensor?

If I open en close de sensor I'll get the next values:

17:54:44 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14090,"Low":470,"High":1360,"Data":"8D0B0E","RfKey":"None"}}
17:54:44 MQT: domoticz/in = {"idx":3,"nvalue":0,"svalue":"9243406","Battery":64,"RSSI":10}
17:54:47 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14030,"Low":470,"High":1360,"Data":"8D0B0E","RfKey":"None"}}
17:54:47 MQT: domoticz/in = {"idx":3,"nvalue":0,"svalue":"9243406","Battery":64,"RSSI":10}

How can I implement this in Domoticz?

R4C3R, 2018-07-28:
You have to assign the Sync, Low, High, Data and RfKey values to the a button of the RF bridge.

This can be done via http commands, just enter the http commands in your browser like this:

Programming sequences into Sonoff RF (don't enter the _comments_!)
http://ip.of.sonoff.rf-bridge/cm?cmnd=RfSync 14090      _// set Sync_
http://ip.of.sonoff.rf-bridge/cm?cmnd=RfLow 470           _// set Low_
http://ip.of.sonoff.rf-bridge/cm?cmnd=RfHigh 1360        _// set High_
http://ip.of.sonoff.rf-bridge/cm?cmnd=RfCode 9243406 _// set Code / data (=hex 8D0B0E )_
http://ip.of.sonoff.rf-bridge/cm?cmnd=RfKey1 4              _// save to key 1_
http://ip.of.sonoff.rf-bridge/cm?cmnd=RfKey1 5              _// show saved settings for key 1_

R4C3R, 2018-07-28:
Did you do programm also the closing signal? (only difference to the opening signal is the RfSync value of 14030 instead of 14090) 

To my knowledge you can't see the actual status of an RF device.

You can only capture and send RF signals (please correct me if I'm wrong), but if you "miss" a signal or the status is changed without Domoticz registering a signal, then the status in Domoticz doesn't correspondwith the actual status.

You can create a switch as virtual device again just like you did before. If you edit the switch there is a drop down box that says "Switch Type", you can choose "contact".

And you can provide on / off actions just like before.

OTiby, 2018-07-28:
Ok, I'll try.

I have now this in the Sonoff console when I open and close the contact:
20:03:23 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14080,"Low":470,"High":1360,"Data":"8D0B0E","RfKey":5}}
20:03:23 MQT: domoticz/in = {"idx":4,"nvalue":0,"svalue":"9243406","Battery":64,"RSSI":10}
20:03:31 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14010,"Low":470,"High":1350,"Data":"8D0B0A","RfKey":6}}
20:03:31 MQT: domoticz/in = {"idx":4,"nvalue":0,"svalue":"9243402","Battery":64,"RSSI":10}

Now, I'm gonna try to get it registered in domoticz. Or is it impossible to get it from the sonoff bridge to Domoticz?

R4C3R, 2018-07-28:
The bridge will "see" the RF signal from the door contact and post a telemetry message accordingly via MQTT. Domoticz will see the message and change the status accordingly. However you can never be 100% sure that it is correct. e.g. when the battery wears out, the signal might become to weak for the RF bridge to notice.

OTiby, 2018-07-28:
I see, but I'm still not sure that the MQTT is working..

I installed Mosquitto on the synology and stuff. But which parameters should I include where?

IP-adresses (subjective)
Sonoff Bridge 192.168.1.69
NAS: 192.168.1.200

NAS-account:
account: admin
pw: 1234

Installed Mosquitto on Synology by: https://youtu.be/b3A1RJdDf-w

Sonoff MQTT parameters:
Host: 192.168.1.200
Port: 1883
User: mosquitto
Password:

On the Synology NAS, in Domoticz I added:
Idx: 2
MQTT Client Gateway with LAN interface
Name: SonoffBridge-MQTT
Remote-address: localhost
Port: 1883
Username: mosquitto
Password: 
Publish Topic: Out

Idx: 3
Dummy (Does nothing, use for virtual switches only)
Name: SonoffBridge-Dummy

Virtual Sensors:
Idx: 1
Hardware: SonoffBridge-Dummy
Id: 00014052
Name: Doorcontact
Type: Light/Switch
SubType: Switch
Data: Closed

Data from Sonoff console:
20:17:02 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14010,"Low":470,"High":1360,"Data":"8D0B0A","RfKey":5}}
20:17:02 MQT: domoticz/in = {"idx":4,"nvalue":0,"svalue":"9243402","Battery":64,"RSSI":10}
20:17:05 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14080,"Low":470,"High":1360,"Data":"8D0B0E","RfKey":6}}
20:17:05 MQT: domoticz/in = {"idx":4,"nvalue":0,"svalue":"9243406","Battery":64,"RSSI":10}

OTiby, 2018-07-28:
I use this door/windows contact with open/close code 433Mhz
https://www.aliexpress.com/item/5pcs-433MHz-EV1527-two-way-wireless-intelligent-door-window-sensor-APP-control-wifi-door-detector-for/32642367411.html

OTiby, 2018-07-29:
Aaah.. no...

I also learned another remote as rfkey 1, this one needs to switch a relay..
But also this one activates now the Idx: 2 :(
Same as the door contact..

08:26:47 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":13560,"Low":440,"High":1300,"Data":"55DD00","RfKey":1}}
08:26:47 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"5627136","Battery":64,"RSSI":10}

And in Domoticz i see only 'off'

Date | Data
-- | --
2018-07-29 09:31:31 | Off
2018-07-29 09:31:23 | Off
2018-07-29 09:31:11 | Off
2018-07-29 09:31:06 | Off
2018-07-29 09:30:54 | Off
2018-07-29 09:27:48 | Off
2018-07-29 09:26:48 | Off
2018-07-29 09:22:48 | Off
2018-07-29 09:22:42 | Off

OTiby, 2018-07-30:
@R4C3R, you already helped me a lot! Thanks! Do you (or somebody else) know how to deal with the Mqtt question? Then I'm finished! :)

R4C3R, 2018-07-30:
@OTiby just looked into it and saw that the received signals are not by default relayed as mqtt message (although they appear in the log / console). I would have to look into that, but I don't have the time right now. Maybe someone else can be of help.

ascillato2, 2018-09-04:
Hi,

As there is no activity in this issue, I'm closing it. Please, ask to reopen if you have more information/questions. Thanks

OTiby, 2018-09-07:
Please reopen. I'm still looking for the solution to get different mqtt signals for open and close

ascillato, 2018-09-07:
If you have different RF codes for open and for close, you can set rules to publish the mqtt message you want, please see the wiki for examples.

ascillato, 2018-09-09:
Hi,

Have you managed to solve your issue?

OTiby, 2018-09-09:
No not yet. I don't get it. In Domoticz I receive only the same message as said earlier.

But also this one activates now the Idx: 2 :(
Same as the door contact..

08:26:47 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":13560,"Low":440,"High":1300,"Data":"55DD00","RfKey":1}}
08:26:47 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"5627136","Battery":64,"RSSI":10}

And in Domoticz i see only 'off'

Date	Data
2018-07-29 09:31:31	Off
2018-07-29 09:31:23	Off
2018-07-29 09:31:11	Off
2018-07-29 09:31:06	Off
2018-07-29 09:30:54	Off
2018-07-29 09:27:48	Off
2018-07-29 09:26:48	Off
2018-07-29 09:22:48	Off
2018-07-29 09:22:42	Off
In sonoff I learned:
rfkey1: RC1
rfkey2: RC2

rfkey5: door contact open
rfkey6: door contact closed

And all the 4 are sending through Idx 2 to Domoticz, where I only see 'OFF' as Data regardless of which controller I activate..

ascillato, 2018-09-11:
Can you show the output of Tasmota Console when receiving the OPEN and when receiving the CLOSE ?

and also a screenshot of your Tasmota domoticz config webpage?

ascillato, 2018-09-13:
Please, can you show the output of the Tasmota's Console when receiving the OPEN and when receiving the CLOSE RF commands ?

and also a screenshot of your Tasmota's domoticz config webpage?

OTiby, 2018-09-13:
Yes, I'll.  But I'm at holiday till next week.

ascillato2, 2018-09-17:
Hi,

When you have time, please, show the output of the Tasmota's Console when receiving the OPEN and when receiving the CLOSE RF commands ?

and also a screenshot of your Tasmota's domoticz config webpage?

Thanks

OTiby, 2018-09-17:
[image]

ascillato, 2018-09-17:
Hi,

Thanks for sharing all the information.

Can you post again the **Sonoff log when opening and closing door sensor** but before doing that use the command `weblog 4` ?

OTiby, 2018-09-17:
Sonoff log after weblog 4

20:38:47 CMD: weblog 4
20:38:47 MQT: stat/sonoff/RESULT = {"WebLog":4}
20:38:47 CFG: Opgeslagen in flash op F8, Aantal 148, Bytes 3584
20:39:01 WIF: Controleer verbinding...
20:39:01 WIF: Verbonden
20:39:21 WIF: Controleer verbinding...
20:39:21 WIF: Verbonden
20:39:41 WIF: Controleer verbinding...
20:39:41 WIF: Verbonden
20:40:02 WIF: Controleer verbinding...
20:40:02 WIF: Verbonden
20:40:11 MQT: tele/sonoff/STATE = {"Time":"2018-09-17T20:40:11","Uptime":"0T03:50:34","Vcc":3.249,"Wifi":{"AP":2,"SSId":"Boerderij de Heerlijkheid","RSSI":90,"APMac":"B0:4E:26:E5:2A:DA"}}
20:40:12 SER: Ontvangen A4 36 B0 01 D6 05 50 8D 0B 0A 55
20:40:12 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14000,"Low":470,"High":1360,"Data":"8D0B0A","RfKey":5}}
20:40:12 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"9243402","Battery":64,"RSSI":8}
20:40:13 HTP: Opdracht
20:40:13 SRC: WebCommand from 192.168.178.100
20:40:13 RSL: Ontvangen topic /RfKey4, Data lengte 0, Data 
20:40:13 RSL: Groep 0, Index 4, Opdracht RFKEY, Data 
20:40:13 MQT: stat/sonoff/RESULT = {"RfKey4":"Learned sent"}
20:40:13 SER: Ontvangen A0 55
20:40:13 SER: Ontvangen A4 36 B0 01 D6 05 50 8D 0B 0A 55
20:40:22 WIF: Controleer verbinding...
20:40:22 WIF: Verbonden
20:40:27 SER: Ontvangen A4 36 B0 01 D6 05 46 8D 0B 0E 55
20:40:27 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14000,"Low":470,"High":1350,"Data":"8D0B0E","RfKey":6}}
20:40:27 MQT: domoticz/in = {"idx":2,"nvalue":0,"svalue":"9243406","Battery":64,"RSSI":8}
20:40:27 HTP: Opdracht
20:40:27 SRC: WebCommand from 192.168.178.100
20:40:27 RSL: Ontvangen topic /RfKey4, Data lengte 0, Data 
20:40:27 RSL: Groep 0, Index 4, Opdracht RFKEY, Data 
20:40:27 MQT: stat/sonoff/RESULT = {"RfKey4":"Learned sent"}
20:40:27 SER: Ontvangen A0 55
20:40:43 WIF: Controleer verbinding...
20:40:43 WIF: Verbonden
20:41:03 WIF: Controleer verbinding...
20:41:03 WIF: Verbonden
20:41:23 WIF: Controleer verbinding...
20:41:23 WIF: Verbonden
20:41:44 WIF: Controleer verbinding...
20:41:44 WIF: Verbonden

ascillato, 2018-09-17:
Thanks again for all the tests.

So, from Tasmota seems that is receiving the 2 codes correctly:

```
20:40:12 SER: Ontvangen A4 36 B0 01 D6 05 50 8D 0B 0A 55
20:40:12 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14000,"Low":470,"High":1360,"Data":"8D0B0A","RfKey":5}}
```
and
```
20:40:27 SER: Ontvangen A4 36 B0 01 D6 05 46 8D 0B 0E 55
20:40:27 MQT: tele/sonoff/RESULT = {"RfReceived":{"Sync":14000,"Low":470,"High":1350,"Data":"8D0B0E","RfKey":6}}
```

So, may be the configuration issue is in Domoticz.

I can't help you with Domoticz since I don't use it. Sorry.

Let's ask.

@arendst Can you see the configuration issue here?

OTiby, 2018-09-18:
Thanks for helping me!

What I see is that sonoff send both messages to idx '2'. And domoticz don't seperate these.

R4C3R, 2018-09-18:
The link below shows a way how to deal with the two different RFkey signals in Domoticz.

http://www.domoticz.com/forum/viewtopic.php?t=19226

ascillato2, 2018-09-18:
@R4C3R 

Thanks a lot for sharing this information.

@OTiby 

Please, when you have time, tell us if now it is working for you. Thanks.

closed 2018-09-18 by ascillato2

OTiby, 2018-11-17:
Ok, it took a while cause it is at my parents.

Today I tried the link of R4C3R, but this blog is about Domoticz with RFXCOM...And not with Tasmota (Sonoff RF bridge) and Mosquitto and that you get everything through Idx / 1 sensor...
I only get the message 'off'. Not two different messages. The message that the contact send to Tasmota is different but it comes all through MQTT bij Idx 2 :S

OTiby, 2018-11-28:
The door contact sends different svalues for the status of 'open' and 'closed', however it is both sended by the same Idx (In my case Idx 5 (in Tasmota under "Sensor idx 6 Count/PM1") to Domoticz.

In the Log of Domoticz I'll get the following messages:
CODE: SELECT ALL

2018-11-18 15:18:49.967 MQTT: Topic: domoticz/in, Message: {"idx":5,"nvalue":0,"svalue":"9243402","Battery":64,"RSSI":8}
2018-11-18 15:18:52.276 MQTT: Topic: domoticz/in, Message: {"idx":5,"nvalue":0,"svalue":"9243406","Battery":64,"RSSI":8}
9243402 = Open
9243406 = Closed

But if I look in the Log of the Dummy switch with Idx 5 i'll see when I open ánd close the sensor:
CODE: SELECT ALL

2018-11-18 15:18:52	Off
2018-11-18 15:18:49	Off
Both messages are "Off"

So I want to make a switch that gives the right status.

I tried things with a LUA script, but I'm a noob :P So I don't understand it.

Can somebody help me with making the script, so I'll get a Dummy switch that distinguishes the different svalue and translate it to the right status of the dummy switch?

maiaz, 2019-01-03:
i have similar issue with RF bridge with tasmota installed. i have found no other approach to manage this, except to process MQTT message externally (rewrite with NodeRed, and put it back to MQTT) so Domoticz unedrstand it.

so by steps:
1. rf bridge send mqtt message under topic: stat/XXXX/RESULT
2. NodeRed picks it up and rewrites, then put back with topic domoticz/in
3. domoticz picks it up - everyone is happy.

in my case difficulty is that if i'm pressing same button on remote and receiving not the same signal every time :((

data= { Sync: 5740, Low: 180, High: 550, Data: 'DDD55D' }
data= { Sync: 5740, Low: 180, High: 560, Data: 'DDD55F' }
data= { Sync: 5730, Low: 170, High: 560, Data: 'DDD55F' }
data= { Sync: 5750, Low: 180, High: 560, Data: 'DDD55F' }
data= { Sync: 5750, Low: 180, High: 560, Data: 'DDD544' }
data= { Sync: 5740, Low: 170, High: 560, Data: 'DDD55F' }
data= { Sync: 5740, Low: 170, High: 560, Data: 'DDD55F' }
data= { Sync: 5750, Low: 180, High: 560, Data: 'DDD544' }
data= { Sync: 5750, Low: 180, High: 560, Data: 'DDD55F' }