arendst/Tasmota #3359: wifi trigger

motaz9k, 2018-07-28:
Tasmota has implemented Mqtt#Connectedand and Mqtt#Disconnected triggers but there is no Wifi#Connected and Wifi#Disconnected, 
I would like to ask why?

I have this rule: 
`on Mqtt#Disconnected do backlog PowerOnState 3;ButtonTopic 0 endon on Mqtt#Connected do backlog PowerOnState 4;ButtonTopic sonoff_touch_4 endon`

and it works great if the connection to mqtt is lost and restored but if you lose your wifi connection, these triggers do not work. 
is there any other method to check by rule if there is a connection to wifi?

Jason2866, 2018-07-28:
What is your goal do you want to reach? You get no info back if you have no wifi connection.

arendst, 2018-07-28:
Working on it. About to release.
```
00:00:00 APP: Boot Count 196
00:00:00 SRC: Restart
00:00:00 Project sonoff Ring2 (Topic ring2, Fallback DVES_15568C, GroupTopic sonoffs) Version 6.1.1c-2_4_1
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as ring2...
00:00:00 RUL: WIFI#DISCONNECTED performs "power 1"
00:00:00 SRC: Rule
00:00:00 RSL: Group 0, Index 1, Command POWER, Data 1
00:00:00 RSL: stat/ring2/POWER = {"POWER":"ON"}
00:00:00 RSL: stat/ring2/POWER = ON
00:00:01 CFG: Saved to flash at FA, Count 611, Bytes 3584
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on ring2 with IP address 192.168.2.191
00:00:05 RUL: WIFI#CONNECTED performs "power 0"
00:00:05 SRC: Rule
00:00:05 RSL: Group 0, Index 1, Command POWER, Data 0
00:00:05 RSL: stat/ring2/POWER = {"POWER":"OFF"}
00:00:05 RSL: stat/ring2/POWER = OFF
00:00:05 CFG: Saved to flash at F9, Count 612, Bytes 3584
```

motaz9k, 2018-07-28:
wow, I tested it and it works as expected 👍 
thank you @arendst !

closed 2018-07-28 by motaz9k