arendst/Tasmota #3360: Question: Sonoff Touch reconfigure functionality

firzanarmani, 2018-07-28:
Hi,

I would like to know if it is possible to reconfigure the button/touch functionality, i.e the button usage as per https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage.

My case scenario is this. When I single-press/touch the button, instead of toggling the relay, I would like it to simply send a MQTT message (instruction to trigger another device or something). When I double-press the button, it will perform the regular relay toggle with the appropriate status change.

Can anyone point me in the correct direction of how to perform this?

Jason2866, 2018-07-28:
Take a look here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Control-other-devices

ascillato2, 2018-07-29:
Hi, if your question has been addressed, please close this issue. Thanks.

closed 2018-08-25 by ascillato2