arendst/Tasmota #3361: Commandlline restart sonoff via mqtt

lscosta90br, 2018-07-28:
I am using sonoff 6.1.1 software on the mini d1 hardware and am loosing a web interface after a few days.
The mqtt commands can be fine. I would use a command line using the mosquitto to restart the hardware.
Could someone help me in command?

  mosquitto_sub -h <host> -p 1883 -u ha -P <password> -d -t "cmnd / garage / Restart 1"

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

wasi54, 2018-07-28:
mosquitto_pub -h mymqtt -t 'cmnd/garage/restart' -m '1'

replace mymqtt with your host

wasi54, 2018-07-28:
See https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

ascillato2, 2018-07-28:
Closing this issue as was answered. Please, ask to reopen if you need anything else. Thanks

closed 2018-07-28 by ascillato2