arendst/Tasmota #3362: wifi range very poor with sonoff basic/tasmota 6.1.1, works perfect with 5.12,5.11...

hudecitydave, 2018-07-28:
After using tasmota version 5.11 and 5.12 for about 6 sonoff basic yesterday I flshed a new one with tasmota 6.1.1. 
The wifi range under 6.1.1 was very poor (under 3 meters away from fitz!box or repeater). Changing back to 5.12 or 5.13 brought back the "known" wifi "power". 

Hardware used : sonoff basic

Status0:
15:58:50 MQT: stat/wasserspiel/STATUS = {"Status":{"Module":1,"FriendlyName":"wasserspiel","Topic":"wasserspiel","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:58:50 MQT: stat/wasserspiel/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":46,"SaveCount":136,"SaveAddress":"FA000"}}
15:58:50 MQT: stat/wasserspiel/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018.01.07 21:51:30","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:58:50 MQT: stat/wasserspiel/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"skynet","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
15:58:50 MQT: stat/wasserspiel/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
..... ,"Webserver":2,"WifiConfig":3}}
15:58:50 MQT: stat/wasserspiel/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.151","MqttPort":1883,"MqttClientMask":"wasserspiel","MqttClient":"wasserspiel","MqttUser":"ASU19MQTT","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
15:58:50 MQT: stat/wasserspiel/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 28 14:58:50 2018","Local":"Sat Jul 28 15:58:50 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
15:58:50 MQT: stat/wasserspiel/STATUS10 = {"StatusSNS":{"Time":"2018.07.28 15:58:50"}}
15:58:50 MQT: stat/wasserspiel/STATUS11 = {"StatusSTS":{"Time":"2018.07.28 15:58:50","Uptime":0,"Vcc":3.186,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","RSSI":22,"APMac":"xxxxx"}}}

arendst, 2018-07-28:
MQTT MAX_PACKET_SIZE is also way to small so I guess some libraries have not been update in between...

hudecitydave, 2018-07-28:
Very sorry. My Status 0 took you the wrong way. It is from the actual 5.11.1... Not very helpful I think.
I used the 6.1.1 minimal, and Thema the DE Edition as .bin download leading to my problem...

ascillato2, 2018-07-29:
Hi, I could not reproduce your issue. Please share status 0 of your v6.1.1 device

mosix76, 2018-07-31:
Hello,
I think I have the same problem. I've flashed via OTA a sonoff basic passing from 5.14 to v6.xx (the last one downloaded on July 30th). Before flashing the connectivity was good, but since I've flashed the sonoff, it doesn't want to connect to the router. On the router system log I can see the request from the mac address of the sonoff, and the ip address assignment by the router, but It's like the sonoff does not accept the ip.
Instead, if I use a 4g mobile router (very close to the sonoff) everything works perfectly. It's making me crazy... 
More over, I tried to downgrade via OTA the sonoff to go back to the version 5.14... After an error on the uploading, the device didn't respond any more: no led flashing, no action pushing the button. So I tried again uploading the firmware by Platformio and ft232rl usb adapter. The downgrade went well, but the sonoff is like dead... doesn't respond. Any idea?

curzon01, 2018-07-31:
@mosix76 - what's the router manufacturer, model where the problem occurs? I'm asking because e. g. some older AVM models causes the same problem here and deleting my ESP device from the known network device list solved the problem.

mosix76, 2018-07-31:
TP-Link TL-MR6400 Router 4G LTE
A sonoff basic with TASMOTA 5.14 works perfectly with it

curzon01, 2018-07-31:
was only a thought - did you tried a reset (`Reset 1` or `Reset 2` to load default params) on your sonoff basic you flashed from 5.14 to 6.xx and then a reconnect using WPS or whatever?

mosix76, 2018-07-31:
Just this morning Theo suggested me to do so. I'll try after work. I hope it can solve the problem. I'll let you, thanks.

murphyslaw05, 2018-08-03:
I have also had this issue happen on 6.1.0 and 6.1.1. When upgraded the sonoff won't connect to wifi while powered by mains. If it is connected to 3.3v from my ftdi adapter it connects nearly every single time(same issue with multiple sonoff upgrades.) Eventually after the upgrade and subsequent fiddling I can get it to connect while on mains, but generally only after having booted it with the 3.3v, I have moved back to 5.14 for almost everything due to these issues.

GasTurbineMan, 2018-08-05:
I had a Sonoff Basic flashed with an earlier version (maybe 7 days ago) of 6.1.1c and it worked great, no issues.  I did an OTA flash of the most recent 6.1.1c and post flash, it would not connect to my WiFi.  I removed the unit and did a serial flash of the same latest 6.1.1c and it still would not connect.  I then did a serial flash of the prior 6.1.1c (7 days older) and it connects to my Wifi just fine.  

Just to be clear, I am downloading the .zip files from GitHub and doing my own builds.  For the most recent 6.1.1c, I copied the same user_config.h file from the earlier 6.1.1c to the latest 6.1.1c, so maybe that was the issue, not sure.

Just an observation.

pcdiks, 2018-08-21:
I have this problem as well. I just flashed from 6.1.1c to 5.1.2 (had this version already on my laptop). Went from no connectivity on 6.1.1c to full connectivity on 5.1.2. I actually bought an additional Sonoff Basic because i thought it was broken when i had 6.1.1c on it. I flashed the new unit with 6.1.1c and it just cannot connect through WiFi. Flashed it with 5.1.2 and it works without a problem.

ascillato, 2018-08-21:
Hi,

Please, try to erase all flash before installing last version so as to reset all the wifi config of the sdk.

`esptool.py --port COM5 erase_flash`

murphyslaw05, 2018-08-21:
The repeats flash tool erases it iirc. This is what I use to flash, so I’m not sure it’s tied to that.

ascillato2, 2018-08-25:
Hi,

Have you managed to solve your issue?

pcdiks, 2018-08-26:
I’m nog sure if you are waiting for my feedback but since it started working on 5.1.2 I left it on that version. But to make it clear, I use Visual Studio Code to flash. Also, i flashed version 6.1.1c on a Sonoff basic with stock firmware and it did not connect to WiFi. Only after flashing 5.1.2.

ascillato, 2018-08-26:
@pcdiks

That is ok. Have you tried to erase all flash before upload? 

@hudecitydave

Have you managed to solve your issue?

ascillato2, 2018-08-30:
Hi,

Please, share your results when you have time. Thanks.

closed 2018-08-30 by ascillato2

Damontir, 2018-09-22:
same problem here with 6.2.1. 
No connection after flashing it via OTA -> minimal -> german.

Thought i could get it back to work with an wired flash via plattformIO but same problem after flashing it. No connection to the wlan.

mattpurnell, 2019-01-14:
I am also having these problems with Sonoff Basic since maybe 5-6 months ago.

Interesting - my T1 lights and Sonoff POW work great..!

ascillato, 2019-01-14:
Hi,

Please try the solutions proposed in the wiki for Wifi Issues: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

Thanks