arendst/Tasmota #3364: CCS811 stops working

Chaot1, 2018-07-29:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : 
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
stat/sonoff/STATUS = {"Status":{"Module":19,"FriendlyName":["Buero"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Thank you for implementing the CCS811 . But i have a small Problem:
When i start the CCS811 it reads the correct status an i can see him on the SonoffDEV website. But after 1 or 2 Hours the CCS811 disappears. When i restart  the Sonoff it appears again.

Connected on GPIO 4 and 5
Bridge between WAK and GND (on CCS 811 to activate)
No other items connected

martin072, 2018-07-29:
HI,

I have seen strange behaviour with the CCS on the ESP. In the end this worked for me:

./Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_si2c.c"
Line 74: Changed twi_setClockStretchLimit(230) to twi_setClockStretchLimit(460);

It does not seem to affect other i2c devices I use (Oled and BME260), maybe this helps?

reference: [https://github.com/AKstudios/CCS811-library/issues/3#issuecomment-336360650](https://github.com/AKstudios/CCS811-library/issues/3#issuecomment-336360650l)

gemu2015, 2018-07-30:
in Tasmota Adafruit Lib Clockstretchlimit is already set to 1000 =>
#ifdef ESP8266
	Wire.setClockStretchLimit(1000);
#endif

I use this Sensor since a couple of month without any problems.
also the new Tasmota implementation runs without problems.
however i use a very stable 3.3 V power supply with an Esp07, (no sonoff) may be it is a power supply issue?
is use these pins =>
GPIO12 = SDA (d6)
GPIO14 = SCL (d5)

ascillato2, 2018-08-22:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks

Chaot1, 2018-09-02:
Ok, i think this Sensor is not usable for me.
Exactly after 50 Minutes it stops working and i have to restart the whole board to get actual data.
I think it is a problem on the "Awake" signal. I use a wire to hold it on GND. Maybe it must be switched every n-minutes

moneybag, 2018-10-01:
Same problem here, after a couple of hours the sensor is gone. /awake to ground.
Interesting: i have to connect and disconnect to usb power a few times until the sensor is recognized.

pesone, 2019-01-25:
I have got it working now for more then 10 Days without any reset or power reboot.

What i did is to increase the value for updating CCS811 to 30 seconds in 

sonoff/xsns_31_ccs811.ino > line 43

as my teleperiod is set 30 and i dont need no more measurements meanwhile.

`#define EVERYNSECONDS 30

void CCS811Update(void)  // Perform every n second
{`

and to adjust > line 73

      } else {
        // failed, count up
        ecnt++;
        if (ecnt > 2) { 
          // after 60 seconds, restart 
          ccs.begin(CCS811_ADDRESS);
        }

I have set sleep to 250.

qingz2004, 2019-01-25:
I tried it.
It does not freeze any more, but the data I got is not realistic.
[image]

pesone, 2019-01-25:
I have two of them. Both with similar values

`22:04:23 MQT: tele/Test/SENSOR = {"Time":"2019-01-25T22:04:23","CCS811":{"eCO2":1309,"TVOC":138}}`

`22:13:53 MQT: tele/Test/STATE = {"Time":"2019-01-25T22:13:53","Uptime":"11T00:22:33","SleepMode":"Dynamic",...`

If open a window for a couple of minutes values goes near 0

qingz2004, 2019-01-26:
Flash back to original firmware, I get this.
[image]

qingz2004, 2019-01-26:
Grabbed a new Wemos, erased and flashed Tasmota with updated CCS811 driver. Still frozen after a few minutes, and the data read out was still bad before frozen.

> 12:48:18 CMD: i2cscan
> 12:48:18 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x5a"}
> 12:53:13 MQT: tele/sonoff/SENSOR = {"Time":"2019-01-26T12:53:13","CCS811":{"eCO2":7426,"TVOC":1070}}
> 13:00:32 CMD: I2cscan
> 13:00:32 RSL: stat/sonoff/RESULT = {"I2CScan":"Error 4 at 0x01"}
>

mike2nl, 2019-01-31:
@Chaot1 @martin072 @gemu2015 @ascillato2 @moneybag
- depending on the sensor breakout board you have to check if the WAKE 
input is on ground = 0V
- This sensor uses I2C address 0x5A. make sure with a s2cscan that the address has no conflicts
- if you have the Adafruit board checkj the on the 3V3 output is enough voltage. They have two versions. One with a older model of regulator and when you power it with 3V3 only via the I2C bus it can be the voltage issue. A known one.
I have a breakout board from Adafruit with the newer regulator and have to ground the WAKE input.
I can be that some breakout boards have done it with a 10K resistor.

- Check this in any case. When that input have a bad level it can be that the sensor is stoped tp messaure

andrethomas2, 2019-01-31:
re-opening on request from @Jason2866

mike2nl, 2019-01-31:
Update:
there is possible a I2C bus freezing after a certain time. Yes you read it right...
I came to that because with another CCS811 sensor i got hat event from all the users too.
So far i have seen there are some changings in the core 2.5.0-beta 3 too. When i
understand that right and possible that brings other things up.

I found things in the adafruit cpp file in wire.begin. About the use of i2c timeing stretching it can freeze the bus. A certain wwit time (delay) is needed for that to overcome that issue.

And i won't believe but i found it in the Sparkfun library. Yeah so as always you can't trust all ada libs for 100%.

I will test that tomorrow with the other library from Sparkfun with some changes in the code i think. When wthis works i will rewrite the whole driver without using a library at all.

Next part we miss a value. Named millis, see data sheet. Have to investigate what that means.
We have eCO2 in ppm and TVOC in ppb. So there are more changes to get this straight.

But 1st the timing issue fix. Possible we can fix it shortly (cross my fingers)

andrethomas, 2019-01-31:
The datasheet seems to suggest that the host controller needs to have clock stretching enabled.

The behaviour could be because this is not being done and that the CCS811 is holding the I2C bus ransom.

I see it is in the library for the CCS811 as follows

```
void Adafruit_CCS811::_i2c_init()
{
	Wire.begin();
	#ifdef ESP8266
	Wire.setClockStretchLimit(1000);
	#endif
}
```

Its not uncommon for slow I2C sensors to require clock stretching so this is fine, but it may be worth playing with a higher value... I cannot find anything specific in the datasheet which should have specified the clock stretch period required and I do not have this sensor so I cannot hook it up to a scope to see if the conditions are met.

The timing mentioned by @mike2nl also seems to be a bit picky as outlined in the datasheet

[image]

I'm sure @mike2nl has taken all the above into consideration so I'm actually just posting for posterity purposes with the view that it might spark other ideas of what could be going wrong.

pesone, 2019-01-31:
Tasmota Version | 6.4.1(sonoff)
2_4_2/2.2.1(cfd48f3)

### This is status of my CCS811 adafruit breakout board on a sonoff basic
```
21:29:03 MQT: tele/Abluft/STATE = {"Time":"2019-01-31T21:29:03","Uptime":"16T04:17:42","SleepMode":"Dynamic","Sleep":250,"LoadAvg":3,"POWER":"OFF",...
21:29:03 MQT: tele/Abluft/SENSOR = {"Time":"2019-01-31T21:29:03","CCS811":{"eCO2":530,"TVOC":19}}
```
I can't do any measuring, because it is too much effort to remove tit from its place.
But i remember that i have to connect 
- WAKE to GND 
on the Adafruit Board, otherwise it won't start.

### Status of a other breakoutboard CJMCU-8118 see above.

- WAKE-10K-GND (INPUT is STABLE Measured between VCC = 3.26V)

- ADD-GND

**As im not a developer and i don't know what im doing, i can only put in the
History of my observations:**

- First i bought the adafruit board and it won't work. The IC2 freezes. Sensor disapeared and to get it work for another couple of minutes i have to do a power reboot.

- I tought the expensive adafruit breakoutboard could be broken, and i orderd another cheap Breakoutboard CJMCU-8118 

- But the same behaviour as with adafruit

- i've started playing around with clockstretchlimit, with no luck (i had the impression higher values faster freezing)

- i have done 100 testings for sure and finally i've got  them working

what i've done is

- **I've compiled with no sensors except CCS881 in my_user_config**
- **Clockstretchlimit 500 (i have read about this somewhere else)**
- and see post above (But this shouldn't have impact because all values bigger then 5 worked for me)

**It can't be just a coincidence that two different boards start working with no issues, 
and if i flash back to "standard-tasmota-master" they show the freezing behaviour.**
This is reproducable at least for me.

andrethomas, 2019-01-31:
@pesone Thanks for sharing - I'm sure @mike2nl will find it useful... really wish I had one of these to help out though.

mike2nl, 2019-02-03:
@pesone @andrethomas 
Update for today:
- Code for detecting sensor type, config and start complete re-written without using any library. 
- No use of a Library at all. All defines in code.
- No stretching will be used for first test today.

Have to do some things in code to test the first test steps. When the debugging tells me what i wish to see i will look further in it.
Also i found out that some of the CCS811 breakout boards have a NTC on board for the temperature compenstation. There is a register to set and i think i have found a way to check for a NTC.
Next step will be read the values and after that a define will be given to use a BME280 which is possible used in the same setup.
So lots of work but i hope i get it done. 
BTW: Andre has looked ones over the first idea of code. He means give it a go.

gemu2015, 2019-02-03:
@mike2nl 
sorry that i can not contribute. i bought this sensor about a year ago together with the sgp30, adapted both drivers and from the beginning i did not have any issues.  ( the ccs811 is compensated with a sht3x) i then sealed the sensors with hot glue so that i can not even verify the exact hardware parts involved.
comparing both parts at the same location gave very similar curves but sometimes shifted values for both signals.

mike2nl, 2019-02-05:
@gemu2015 - no issue at all ;-)
@pesone @gemu2015 @ascillato2 @andrethomas 
Short Update:
- first test wit a library from Maarten Pennings looks very prommising. made a test over 2 hours and it works. If an error is happen on the bus (esp8266 worst stretching mechanism) he shows it.
- the lib is currently filled with a lot of print functions, so she is too big. That i will change to get much smaller. The library looks much more stable to me then the Adafruit. The Adafruit is missing things and has done it not 100%.
- Maarten has also some currently changes to the I2C driver in ESP8266/Arduino. These changes are not in beta 3 now. But i have added his changes and the bus is responding faster and much better with the CCS811 sensor. Brings me to the SHT and TSL issue too. Will tast this with the i2c code changings.
- Jason gave me the helping link to the issue at the esp8266/arduino git. Thank You @Jason2866 for that. I had overseen that link so i think we have a solution about all the issues.
- In any case i will made a new driver with that library from Maarten. Possible add all the lib stuff we need to the driver code. @gemu2015 and me we hate tu much libs at all, haha.

Last but not least we have progress now!

qingz2004, 2019-03-04:
Any news on this?

Frogmore42, 2019-03-12:
Most likely this sensor has the same issues I ran into on the SCD30.  I believe the underlying issue is poor handling of clock stretching by the esp8266.  Even core 2.5.0 does not handle this correctly.  Many of the implementations of sensors ignore CRC and other errors which can completely lock up the i2c bus.  The good news is that it is possible to recover from that.  The library I created for the SCD30 shows one way of checking and reporting errors.  The implementation in sensor 42 shows one way of handling the errors and resetting the i2c bus when things get really bad.

gemu2015, 2019-03-12:
there must be sensor hardware differences too. my ccs has absolutely no issues. it now ran for 178 days without any reset. 

Tasmota 6.1.1.7, core 2.3, sleep 0

Frogmore42, 2019-03-12:
Could be.  I believe it is a timing issue.  When I had the sensor set to deliver updates every 2 seconds and checked every second it was much more likely to run into the issue.  When I had it update every 30 seconds it never happened.  When I ignored CRC errors it happened more frequently.  When I soft reset the SCD30 after a CRC error I actually got more issues than if I just waited till the next second to check again.

Both sensors use clock stretching and the esp8266 implementation of i2c has known issues, so it is not surprising the sensors have issues on the esp8266.

gemu2015, 2019-03-14:
according to maarten pennings the CSS811 issues on ESP8266 can be solved by a wait just before Wire.requestFrom in the CSS811 read sequence in file Adafruit_CCS811.cpp (see below)
since i can not reproduce the issue, may be someone with the issue may test this fix.

void Adafruit_CCS811::read(uint8_t reg, uint8_t *buf, uint8_t num)
{
	uint8_t value;
	uint8_t pos = 0;

	//on arduino we need to read in 32 byte chunks
	while(pos < num){

		uint8_t read_now = min((uint8_t)32, (uint8_t)(num - pos));
		Wire.beginTransmission((uint8_t)_i2caddr);
		Wire.write((uint8_t)reg + pos);
		Wire.endTransmission();
                delayMicroseconds(50); // maarten pennings suggestion
		Wire.requestFrom((uint8_t)_i2caddr, read_now);

		for(int i=0; i<read_now; i++){
			buf[pos] = Wire.read();
			pos++;
		}
	}
}

qingz2004, 2019-03-16:
Tried. No luck.

Frogmore42, 2019-03-16:
The code show is doing no error checking, so it is not surprising that it is not working.  Many examples in the Arduino world are quite simplistic and work great as long as everything goes down the "happy path" (where everything just works all of the time).  The CCS811 and the SCD30 are NOT sensors that work with the esp8266 in the "happy path".  If you want success with them, the code has to handle the things that can and do happen that are outside the "happy path".

The specific code sample above is only waiting 50 uSec for data to be available before requesting it.  That is likely not enough time for the CCS811 to prepare it (at least some of the time).  When it does request the data, it doesn't bother to see if the request was actually fulfilled.  It just blindly reads the data.

This was what I found in the available libraries for the SCD30.  The official library for the SCD30 from Sensiron was a little better, but it had too many dependencies to be easily usable with Tasmota.  So, I wrote a new library for it.  The code window is not working for me, so look at this:
[image]

The key thing here is checking the result from requestFrom().  The check for available() is not really necessary, since the result from requestFrom() pretty much assures there is data available.

But, this is the section that made it start working for me:
[image]

The key piece is waiting (1ms worked fine for me) between sending the command to the sensor which requests it to send data and actually trying to read data.  The examples I saw for Arduino did no waiting and all had issues working with the esp8266.  I believe this is due to the underlying issue in the libraries for the esp8266.  Waiting seems to work around the issue, at least most of the time.

The code for the SCD30 and the sensor code has additional retry methods and the really big hammer of resetting the whole i2c bus.  When I slowed things down for the SCD30, I rarely have to use the really big hammer.  Before getting the code working well, I would have to reset every few min at the most.  Now, it can go for days without needing to reset the i2c bus.

I don't have a CCS811 and given what I have seen of it and the SCD30, I have no desire to have one.  But, the technique I used for the SCD30, should also work the CCS811.

jpduhen, 2019-03-23:
@Frogmore42 Since I’m not a programmer, could you please show the particular file to edit and what lines need to change for the CSS811 to get working, when using Platformio to compile Tasmota?

>

Frogmore42, 2019-03-23:
@jpduhen that would not be easy.  The code for the CSS811 does not follow best practices for dealing with hardware (it takes quite a few shortcuts, not a problem … until it is).  A quick look at the code shows this function:
`void Adafruit_CCS811::read(uint8_t reg, uint8_t *buf, uint8_t num)`

add a delay between:
`Wire.endTransmission();`
and 
`Wire.requestFrom((uint8_t)_i2caddr, read_now);`

I used delay(1);
this was enough for the scd30 most of the time.  To really solve the problem is going to take someone who is really interested in solving it who has one and enough time and patience to figure out exactly what is going wrong.  Making i2c work reliably is not easy in the best of circumstances.  Since it seems there are issues with the esp8266 implementation/support of clock stretching, it might be better to get a different sensor.

jpduhen, 2019-03-23:
@Frogmore42 I’ll 1st try suggestion one, and let you know if it fixes things, if not, I’ll go for suggestion 2 ;-)

jpduhen, 2019-03-23:
@Frogmore42 SUCCESS!! Now 4 hrs stable. I think you nailed it!
[image]

Frogmore42, 2019-03-23:
Glad it worked out.  I suspect that it will still have issues, but they won't be as frequently.

qingz2004, 2019-03-23:
Does not work form me. Disappeared in 3 hours.

jpduhen, 2019-03-23:
14 hours here and still going OK. Do you use 6.5.0 with the 2.3.0-core?

qingz2004, 2019-03-23:
6.4.1.22 with 2.4.2 core

jpduhen, 2019-03-23:
I had massive problems with core 2.4.2. Half a year of instability in Wifi connections from my sensors and switches, socket-errors in mosquitto-MQTT. Glad the 6.5.0 release is on core 2.3.0 again. Perhaps you should try it also.

qingz2004, 2019-03-23:
Just tried.
Still no luck.
Less than 18 minutes.
[image]

Jason2866, 2019-03-24:
@qingz2004 you can try i2c clock streching fix for arduino core https://github.com/maarten-pennings/I2C-tool/blob/master/I2Ctest8266/README.md#how-to-fix

jpduhen, 2019-03-24:
You can also try this fix, worked for me earlier:
[image]

qingz2004, 2019-03-24:
@Jason2866 @jpduhen Thanks for the suggestions, but I'm using PlatformIO.

joba-1, 2019-03-24:
Pio uses the same esp cores, just search for the file yourself. For me it
is somewhere below $HOME/.platformio
-- 
iphone so i typo

Jason2866, 2019-03-24:
@qingz2004  i use Platformio too...  Search where @joba-1  wrote. It is there ;-)

jpduhen, 2019-03-24:
Yes, found it too:

[image]

Frogmore42, 2019-03-24:
Doesn't really do anything since this code follows that:
'void Adafruit_CCS811::_i2c_init()
{
	Wire.begin();
	#ifdef ESP8266
	Wire.setClockStretchLimit(1000);
	#endif
}'

jpduhen, 2019-03-24:
OK, then this is not a possible solution.

Frogmore42, 2019-03-24:
If you want it longer, just change it in the CCS811 code.  The SCD30 code I wrote uses a VERY large number.  It sort of helps but does not solve the problem always.  Maybe when the underlying issue is resolved in the core things will work better.  But, errors happen.  Not dealing with them doesn't make them go away 😉.  Example code rarely deals with errors.  That is left as an exercise for the reader.

jpduhen, 2019-03-24:
CCS811 still working on my TH10 by the way,  1 day and 14 hrs now:

23:23:29 MQT: tele/TH10/STATE = {"Time":"2019-03-24T23:23:29","**Uptime":"1T13:55:34"**,"....
23:23:29 MQT: tele/TH10/SENSOR = {"CCS811":{"eCO2":1872,"TVOC":224},"PressureUnit":"hPa","TempUnit":"C"}

qingz2004, 2019-03-26:
@Jason2866 Tried i2c clock streching fix for arduino core https://github.com/maarten-pennings/I2C-tool/blob/master/I2Ctest8266/README.md#how-to-fix.
Tried both core 2.3.0 and 2.50.0 with PlatformIO, none of them works for me.

jpduhen, 2019-03-26:
@qingz2004 That’s frustrating. If you like I can try to build a tasmota-version for you on Arduino IDE, maybe that will fix things for now?

qingz2004, 2019-03-26:
@jpduhen Sure, thanks! Please let me try your load, and see if the problem is in the load or my CCS811 module. I already tried the CCS811 on different NodeMCU.

jpduhen, 2019-03-26:
@qingz2004 Try this one: [https://www.dropbox.com/s/vmbp7d5ylwbcdkm/nodeMCU-CCS811.bin?dl=0](url)

qingz2004, 2019-03-26:
@jpduhen Thanks! It worked for less than 18 minutes. So not the firmware.
Have to wait for the driver from @mike2nl or get another CCS811.

jpduhen, 2019-03-26:
OK, that’s so frustrating !

gemu2015, 2019-03-28:
@qingz2004 
mine is running since 200 days without any error or stopping.
i compiled a version thats logging the 3 version parameters for:

hardware version
firmware version
app version

my numbers are => CCS811 Vers: HW:18 FW:16 APP:17

you may give it a try

1 mb flash =>https://www.dropbox.com/s/uoweb4ad6y2ft3u/ccs811.bin.zip?dl=0

qingz2004, 2019-03-28:
@gemu2015 Thanks! Will definitely try it when I get home.

qingz2004, 2019-03-28:
@gemu2015 Flashed. How to check the version number?

qingz2004, 2019-03-29:
Good news.
After 7 and half hours, still good.
[image]
[image]

gemu2015, 2019-03-29:
@qingz2004 

OK, first lets check your versions.
version is logged to the console in debug mode on initializing the chip.
setup=>configure logging=>web log level=debug
then restart the module and check the console log

there are NO differences against the Tasmota main version of the driver, BUT
from the very beginning i used the chip in "compensation mode" that is i used the sht31 for humidity and temperature compensation.

having that in mind i only changed ONE thing in the current driver.
when there is no compensation from another chip i did set the compensation to default.
that forces the ccs811 to reevaluate its internal calculations on every reading.
may be that this makes it that stable as in my case. you can simply change the code of the standard tasmota driver with my code snippet below to prove this.

in xsns_31_ccs811.ino

replace
if (global_update) { ccs.setEnvironmentalData((uint8_t)global_humidity, global_temperature); }

with
if (global_update) {
      ccs.setEnvironmentalData((uint8_t)global_humidity, global_temperature);
} else {
      // these are the default values
      ccs.setEnvironmentalData(50,25.0);
}

qingz2004, 2019-03-29:
@gemu2015 14 hours, still good.
Here is my version. Same as yours.
CCS811 Vers: HW:18 FW:16 APP:17

qingz2004, 2019-03-29:
@gemu2015 Tried your code with Tasmota 6.5.0.1 and core 2.5.0. CCS811 stopped working after one and half hour.
There must be something else.

gemu2015, 2019-03-29:
it seems to be a very strange interaction.
probably try again with core 2.3 as i do, there are significant changes in i2c in 2.5
or take my fork and adapt it to your needs. it is a bit outdated, but i only bring it up to date from time to time. => https://github.com/gemu2015/Sonoff-Tasmota

[51 more replies]

closed 2019-05-19 by stale[bot]