arendst/Tasmota #3367: TelePeriod - less than 10 seconds

Rapemortality, 2018-07-29:
Is there any possibility to reduce telemetry period to less than 10 seconds?

There is many things in automation that need quick reaction form automation system - and according to that actual data form sensors. Possibility to reduce "TelePeriod" to ~1-3 sec would be very helpfull.

ascillato2, 2018-07-29:
Hi,

Sorry but no. Teleperiod is just for telemetry. 
For fast response you need to use MQTT or rules or http or knx.

closed 2018-07-29 by ascillato2

Klagio, 2018-10-27:
I would need fast updates, not fast response

osulivanDE, 2022-04-18:
Is there an update? I also would need more frequent updates from my sensors. Why this 10s limit?

barbudor, 2022-04-18:
No and there won't be.
If you want to publish some data more frequently, use Rules and the Publish command

sfromis, 2022-04-18:
If the sensors are energy monitoring, check the `PowerDelta` command for faster responses.