arendst/Tasmota #3368: WebSocket

wschnell, 2018-07-29:
Hi,
I would like to use WebSockets to get push data from the device, so not using MQTT at all.

in node.js:
var socket = new WebSocket('ws://<user>:<password>@<devicename>/cm');
socket.onmessage = function(event) { console.log('WEBSOCKET: ', event); };

currently the Tasmota Software does react, but does not upgrade the http protocol to websockets; I get an "Error during WebSocket handshake: Unexpected response code: 200" when a 101 return would be the positive result.

thanks
Walter

arendst, 2018-07-29:
WebSocket is not supported by Tasmota.

ascillato, 2018-07-29:
Duplicated issue #2749

closed 2018-07-29 by ascillato2

dale3h, 2019-02-14:
I apologize for bringing up an old issue, but this is not a duplicate of #2749. #2749 is referring to MQTT via websockets, but like @wschnell I was hoping to be able to control the Tasmota via raw websockets. Regardless, being that the use cases requiring websockets are unique, I do not see it being added to the core.

The reason I was hoping websockets were already supported is because I am deploying a Sonoff 4CH Pro R2 in my Jeep to control various lights and accessories. Two-way non-polling local control is simply not possible without deploying a local MQTT broker, which is something I was hoping to avoid.