arendst/Tasmota #3371: MHZ19B causes exceptions on Sonoff Basic

Jason2866, 2018-07-29:
MHZ19B causes exceptions after a few minutes

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues and in the wiki
- [x ] Hardware used : Sonoff Basic with MHZ19B
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
`19:26:57 CMD: status 0
19:26:58 MQT: stat/sonoff-71C254/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff134"],"Topic":"sonoff-71C254","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:26:58 MQT: stat/sonoff-71C254/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/web/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:02:38","StartupUTC":"2018-07-29T17:24:20","Sleep":1,"BootCount":71,"SaveCount":91,"SaveAddress":"F9000"}}
19:26:58 MQT: stat/sonoff-71C254/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-07-29T19:13:59","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:26:58 MQT: stat/sonoff-71C254/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00008009","55C18000","00000000"]}}
19:26:58 MQT: stat/sonoff-71C254/STATUS4 = {"StatusMEM":{"ProgramSize":451,"Free":552,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F8AA784","0C000001","C00C0496","00000000"]}}
19:26:58 MQT: stat/sonoff-71C254/STATUS5 = {"StatusNET":{"Hostname":"sonoff-71C254-0596","IPAddress":"192.168.2.134","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"5C:CF:7F:71:C2:54","Webserver":2,"WifiConfig":5}}
19:26:58 MQT: stat/sonoff-71C254/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-71C254","MqttClient":"sonoff-71C254","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:26:58 MQT: stat/sonoff-71C254/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 29 17:26:58 2018","Local":"Sun Jul 29 19:26:58 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"05:43","Sunset":"20:56"}}
19:26:58 MQT: stat/sonoff-71C254/STATUS10 = {"StatusSNS":{"Time":"2018-07-29T19:26:58","MHZ19B":{"CarbonDioxide":928,"Temperature":28.0},"TempUnit":"C"}}
19:26:58 MQT: stat/sonoff-71C254/STATUS11 = {"StatusSTS":{"Time":"2018-07-29T19:26:58","Uptime":"0T00:02:38","Vcc":3.158,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":62,"APMac":"00:A0:57:1E:69:50"}}}`

**To Reproduce**
Connect a MHZ19B to a Sonoff Basic.
5v from Sonoff to Vin, Gnd to Gnd, RX-TX, TX-RX
If "disconnected" in Config Page on Webconsole GPIO1 and GPIO3 to none there are no exceptions anymore 

![image](https://user-images.githubusercontent.com/24528715/43368930-2e2b4416-9365-11e8-9b36-e91a6285a536.png)

![image](https://user-images.githubusercontent.com/24528715/43368932-4152983c-9365-11e8-9801-464425b428d8.png)

![image](https://user-images.githubusercontent.com/24528715/43368937-51489b1a-9365-11e8-8bd1-8cacbe3fabd0.png)

Jason2866, 2018-07-29:
[image]

Jason2866, 2018-07-29:
[image]

Jason2866, 2018-07-29:
[image]

arendst, 2018-07-29:
Workaround: swap tx and rx to use hardware serial.

arendst, 2018-07-29:
See #2659 too.

closed 2018-07-29 by Jason2866