arendst/Tasmota #3373: rules listen to a mqtt topic or send the color value not as a json

Amenophis86, 2018-07-29:
Is it possible to add listening to topics with rules. For example I want to listen to a topic /test/command so if I publish 1 as value then a rule is actived and can answer with another publish like 2 for example. 
Or as another possibility I want to publish the color which is set on a new topic not as a json. Third possability, how can i publish a mqtt msg with a rule after special color is set?

Why do I need this? I have another controller which can not list to ja json as an answer and needs only the value published. So after I set 4C190000 as an color I want to answer him either with the value not as a json or he sends a mqtt msg and i want to answer with my own mqtt msg.

Frogmore42, 2018-07-29:
Where are you running your MQTT broker?
Most devices that can run a broker can also run NodeRED.  NodeRED can easily do what you want and a whole lot more and is probably easier to understand, document, and support.

While rules are impressive in what they can do, when you start wanting to use them to do things across devices, you are crossing into the territory of things that should be done by a higher level.

This is, of course, my opinion, but also one shared by many other people.  I think of this like the parable about the man who just got a hammer and now thinks everything is a nail.  While you can use a hammer to put a screw into a wall, it is not the best tool for the job.  If it is the only tool you have it might be fine, and it is certainly better than using a frying pan for the job 😀

roguestreak, 2018-07-29:
I've asked a similar question to try to solve issue #3308 but no response.  In my case it's trying to figure out a way to distinguish whether a Timer has operated the power or whether it's a button press; if I set buttontopic 1 it triggers sends the cmnd mqtt on button press, whereas the timer doesn't send this, so an mqtt rule trigger could be used to detect which is which.
Also seems strange that there's no rule Trigger that is triggered by the stock button press.

Amenophis86, 2018-07-30:
Well Frogmore42, Thanks for your Statement. I know about NodeRed, pens and other Stuff ;) And I use FHEM. The reason why I ask is a fallback if FHEM does Not work.

Frogmore42, 2018-07-30:
Since you provided no context for your request and answered none of the questions from the issues template, nor the feature request one, it seemed that you did not know all that was possible with Tasmota.

Having two things control the same thing can be done, but it is more complicated to set it up in such a way that they cooperate instead of just fight each other.  It is similar to the old saying, "A man with one clock always knows what time it is, but one with two is never real, sure."

Since you have said that you know about "other Stuff", perhaps you could take the time to figure out the way rules are currently implemented and then figure out how you would add the functionality you are requesting.  Or, even how you would expect a user to specify the behavior you are requesting.

It's my opinion that adding complexity to rules and having Tasmota subscribe to other topics is probably outside of the things that are reply valuable.  People frequently talk about improving reliability by doing things that are not likely to actually improve it.  Any scenario that involves MQTT is going go be dependent on the broker being up.  That is going to depend on the device running the broker being on and having network connectivity.  I would bet that the reliability of NodeRED is at least as good as a random node running Tasmota, and that is no disrespect to Tasmota.  I think Tasmota is great, but it is dependent on a stack of software that has been proven to be less than very reliable.

Remember that Rules were added relatively recently, perhaps Theo will see the value of your suggestion and decide to implement them, if you ask very nicely.

Amenophis86, 2018-07-30:
Well, like I said. There is more than one way and more then one possability. Or in your saing.: There is more than one way to Rom. I asked for a featuer and Theo might implement it, or not. Guess that is what feature requests are for :)

Frogmore42, 2018-07-30:
Actually, that is why the feature request template has all those questions that you seem to have missed answering.  When someone ignores the basics there are two ways to interpret it, ignorace or disrespect.  Since I don't know you, I chose not to assume disrespect.

Amenophis86, 2018-07-30:
I like you :) You are kind of funny. Well maybe it was the first time I had a feature request and did not notice the template or did not exacatly know how it works an just tried it this way. 

And you missed a saying in your last post, dont get inconsistent ;)

Frogmore42, 2018-07-30:
Fair enough.  Life is like a box of chocolates.  You never know what you're gonna to get. 😄

Frogmore42, 2018-07-30:
But, seriously, you seem convinced that having even more flexibility in the rules is a good thing.  I am not convinced it is a bad thing, but I am trying to understand how you would use it and why you feel that is better than the alternatives. 

>For example I want to listen to a topic /test/command so if I publish 1 as value then a rule is actived and can answer with another publish like 2 for example

This seems like a natural thing for something that is easily configured and very flexible, like NodeRED.  How would you describe this in the rules "language"?
Would you expect it to be hard coded with all those values, would you use a regx?
What real value does this provide over using NodeRED?  Is it easier to understand?  Is it easier to implement as a user?  Is it easier to debug?  Is it more reliable?  If yes to any of the above, how/why?

>Or as another possibility I want to publish the color which is set on a new topic not as a json

This, once again exactly the kind of thing for which many people use NodeRED.  It has the power, flexibility, and usability to do this.

How would you describe this in the rules "language"? 

>Third possability, how can i publish a mqtt msg with a rule after special color is set?

For this one, it looks like you might need something from rules for this.  So, how would you describe what you want in the rules "language", if the feature your are requesting were implemented?

Since you like sayings, here is one I saw over a decade ago describing the registry.
"It's powerful … It's complex ... and it's largely undocumented"

ascillato, 2018-07-31:
Hi

@Amenophis86 

> Is it possible to add listening to topics with rules. For example I want to listen to a topic /test/command so if I publish 1 as value then a rule is actived and can answer with another publish like 2 for example.

Yes, It is already implemented.

Example:

Send a MQTT message to `cmnd/sonoff/EVENT ON` and `cmnd/sonoff/EVENT OFF`

And the rules to listen to that could be:

rule 1
rule on EVENT#ON do power1 on endon
rule on EVENT#OFF do power1 off endon

_(As this is just sending the command EVENT SOMETHING to your Tasmota, the limitation is that the first part need to be `cmnd/[yourTOPIC]/EVENT`)_

ascillato, 2018-08-22:
Hi,

have you solved your issue?

closed 2018-08-25 by ascillato2

HobbytronicsPK, 2019-03-19:
@ascillato 

How can we implement subscription to a certain MQTT topic. It is not clear from your example as you told 

> Yes, It is already implemented

How can we define the subscription topic in rule's event?

ascillato, 2019-03-19:
There is not a command for subscription. Tasmota uses the topic and Tasmota is subscribing its own topic into the broker. Then, if you want to send from your broker to Tasmota a command or trigger an event, you need to use that topic.

Example, 

From your broker you want to trigger an event in Tasmota.

Send to tasmota

cmnd/sonoff/event yourevent

And in Tasmota make the rule for that event like:

Rule on event#yourevent do ......

If you need further assistant, please find us in the Tasmota support chat. Thanks

HobbytronicsPK, 2019-03-20:
@ascillato 

> From your broker you want to trigger an event in Tasmota.
> Send to tasmota
> cmnd/sonoff/event yourevent

so it means we cannot define our own subscription topics and have to follow the prefix cmnd/sonoff_device_topic/event.

marksmanaz, 2020-07-10:
I want to retrieve a value from a remote sensor through the mqtt broker to display on a local matrix display but using event the rule would see each new value as a separate event, am I correct?