arendst/Tasmota #3375: The Hard Thermostat merge

mcc05, 2018-07-30:
Been testing out the Hard Thermostat #481, code and it seems to work great, however it predates timers which would make it excellent, and chance of a merge, or similar web driven functionality, that uses the rules functinallity introduced.

ascillato, 2018-07-31:
Hi,

Yes, would be great to have a webmenu for a chrono-thermostat but there is not much flash memory to add much more webmenus. By now, you can use the thermostat example that is in the wiki at the chapter rules.

The User Interfacing can be done in your home automation software. Your HA can send mqtt messages to give new temperature setpoints.

May be in the future we can find a better way.

andrethomas, 2018-07-31:
@mcc05 Using a HA really adds a lot more functionality - I have node-red running monitoring bedroom temperature using a BMP180 sensor connected to an ESP8266 Tasmota firmware board which sends temperature to node-red on 15 second intervals through telemetry logging. Node-red then decides what to do sending either on or off commands to a sonoff basic to turn the heater on and off.

I also have my electric blanket connected to a sonoff on full setting all day and it goes on for a few minutes when I shut down in my lab and warms up the bed whilst I am on my way past the kitchen... then switches to an interval based on the temperature in the room which usually results in something around 50 seconds out of every 5 minutes.

Further to this I can schedule in node-red and also change settings/parameters using messaging from my phone using Telegram (similar to Whatsapp, perhaps just less popular but it has a open API)

[image]

Looks like the temp in the room is still dropping at around 0.1 deg C/ 30minutes so it should kick in automagically when it reaches 19.0 degrees and then it will bounce between that and 19.2 throughout the night and eventually switch off sometime in the morning.

mcc05, 2018-07-31:
I agree with you both. But would be nice to create rules without console just like timers. a thermostat is one very good use case for Sonoffs in general.

I agree with the HA approach for complex interactions, but I would prefer this for augmenting a bullet proof 'just do this' until told different without Wi-Fi etc.
I will try the rules based example on the wikki to see how it compares.

tbagro, 2018-08-10:
how do i use a standalone command for dht sensor? is possible?

mcc05, 2018-08-12:
All,
So I've played with the thermostat example from the wiki, and firstly it doesn't work as expected, changing mem1 without the switch had no effect, so I have re-written it to: 

rule1 
on system#boot backlog do ruletimer1 30 endon 
on Switch1#State do event toggling1=%mem1% endon 
on event#toggling1=0 do mem 1 endon 
on event#toggling1=1 do mem 0 endon 
on rules#timer=1 do backlog var1 0; ruletimer1 30; power1 0; endon 
on tele-AM2301#temperature do backlog var1 1; ruletimer1 30; event ctrl_ready=1; event temp_demand=%value%; endon 
on event#ctrl_ready>%mem1% do var1 0 endon 
on event#temp_demand>%mem2% do power1 0 endon 
on event#temp_demand<%mem3% do power1 %var1% endon

The next problem is timers, so I have the following:
rule2 
on Clock#Timer=1 do mem 1 endon on Clock#Timer=2 do mem 0 endon
on Clock#Timer=3 do mem 1 endon on Clock#Timer=4 do mem 0 endon 
on Clock#Timer=5 do mem 1 endon on Clock#Timer=6 do mem 0 endon 
on Clock#Timer=7 do mem 1 endon on Clock#Timer=8 do mem 0 endon 
on Clock#Timer=9 do mem 1 endon on Clock#Timer=10 do mem 0 endon 
on Clock#Timer=11 do mem 1 endon on Clock#Timer=12 do mem 0 endon 
on Clock#Timer=13 do mem 1 endon on Clock#Timer=14 do mem 0 endon

which does equate to controlling 1 on and off time period per day of the week, however most thermostats at least allow for 3 schedules per day, morning / afternoon / nighttime, but with only 16 timers this is not possible it would need 21. 

Is there a better way to do timers or a way to increase the numbers? It would be nice if each timer supported a start/stop time, then in effect for the 16 we would get 32 timeslices?

Mark

mcc05, 2018-08-13:
I've updated the wiki with a working example, I've been running it for a day now with no issues.

closed 2018-08-13 by mcc05

mcc05, 2018-08-16:
So the thermostat rules do not work on d1 mini?? works on sonoff basic and that's not a th which it said it was for

tbagro, 2018-08-16:
I could not make wemos D1 work, just the timer function, I did not test the thermostat rules in sonoff, because mine is a basic sonoff.
But I confess that I still do not understand the rules for the thermostat when using a DHT11.