arendst/Tasmota #3377: KNX Address add button disabled

mizrachiran, 2018-07-30:
Hi, 
I'm trying to setup KNX on my device. it seems like the add button in the lower section is disabled (Group Addresses to Receive Data from). See image below.
What am I doing wrong?

Thanks!

[image]

- [ X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [  X ] Hardware used : Custom
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

mizrachiran, 2018-07-30:
Using the latest firmware 6.1.1

mizrachiran, 2018-07-30:
just checked my debug logs. It seems that i have 10 malformed records 

21:47:03 KNX: CB: 10
21:47:03 KNX: CB #1: 1/1/81 to <null>
21:47:03 KNX: CB #2: 0/0/0 to <null>
21:47:03 KNX: CB #3: 0/0/0 to <null>
21:47:03 KNX: CB #4: 0/0/0 to <null>
21:47:03 KNX: CB #5: 0/0/0 to <null>
21:47:03 KNX: CB #6: 0/0/0 to <null>
21:47:03 KNX: CB #7: 0/0/0 to <null>
21:47:03 KNX: CB #8: 0/0/0 to <null>
21:47:03 KNX: CB #9: 0/0/0 to <null>
21:47:03 KNX: CB #10: 0/0/0 to <null>

ascillato, 2018-07-30:
Hi,

Yes, that is the problem. The limit is 10 slots for KNX. If you have 10, the Add button get disabled.

That is a problem in your config, but there is a solution. KNX Driver has a startup check routine. To make it to reset the KNX config just change the module type or change some GPIOs to none and then restart.
After restart you can add your GPIO again.

mizrachiran, 2018-07-31:
Thanks.
I tried to change the module configuration and it solved the issue.
I think it worth not saving illegal records or show them in the UI so it will be possible to remove them manually.

@ascillato, thanks again for the assistant and for developing the KNX feature in Tasmota.

closed 2018-07-31 by mizrachiran

ascillato, 2018-07-31:
Yes, you are right. I will add another startup check for setting. Thanks