arendst/Tasmota #3378: Sonoff S31 Factory Firmware OTA?

meingraham, 2018-07-30:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the SonOTA forum
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)

Sonoff S31 with ITead factory firmware version 1.0.0.

My sense is that the S31 firmware is different that then Sonoff Basic firmware and thus there is no correlation to the 1.6 version I have on the Basics I purchased directly from ITead in the same order.

I know that as of 1.6, ITead removed the ability to OTA update (although I've seen recent online noise that 2.0.4 may allow OTA again).

I can't quite seem to find a definitive answer for is whether I can OTA update an S31 with ITead 1.0.0 firmware. I found a few posts on several forums where it stated that SonOTA would not enter the final phase. But there was no confirmation that indeed it is not possible.

I'm just trying to get confirmation before I crack my units open that the only means to flash these devices is to upload new firmware via the serial interface.

digiblur, 2018-07-31:
Did you try it?  Doesn't take but 20-30 seconds to know if it works or not.

meingraham, 2018-07-31:
Since I've never had the option of OTA from factory firmware (all my Sonoff devices have newer firmware), I haven't gone through the install, configure, learn, test of SonOTA. Thus the time to test is likely considerably longer than just "giving it a try".

Having said that, the consensus seems to be that the 1.0.0 in the S31 as well as the POW do not allow OTA. It looks like I'll have to solder a header, flash TASMOTA, and remove the header (so it fits back in the case). A project for another day.

closed 2018-07-31 by meingraham

digiblur, 2018-07-31:
Only took me a short time to download the release and double click the file to test it on the ifan02. Super easy even for a person who never did it.