arendst/Tasmota #3379: Timer enabled

Courty40, 2018-07-30:
Not a show stopper but Just upgraded a bunch of Sonoff Basic’s to 6.x from 5.x and all seem to upgrade ok but end up with the timer enabled but settings not set. Seems to cause the odd glitch on the relay at midnight.

arendst, 2018-07-31:
Should be fixed in 6.1.1c

Courty40, 2018-07-31:
Just checked, I was 6.1.1c !

arendst, 2018-07-31:
Pls provide output of command ``timers`` to verify if any timer was set.

Courty40, 2018-07-31:
11:32:11 CMD: Timers
11:32:11 MQT: stat/Node4/RESULT = {"Timers":"OFF"}
11:32:11 MQT: stat/Node4/RESULT = {"Timers1":{"Timer1":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer2":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer3":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer4":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
11:32:11 MQT: stat/Node4/RESULT = {"Timers2":{"Timer5":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer6":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer7":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer8":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
11:32:11 MQT: stat/Node4/RESULT = {"Timers3":{"Timer9":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer10":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer11":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer12":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
11:32:11 MQT: stat/Node4/RESULT = {"Timers4":{"Timer13":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer14":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer15":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer16":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}

Paul Court
Mobile - 07956 200232

Courty40, 2018-07-31:
This is just one of the units, can dump another if that would help.

arendst, 2018-07-31:
So they're all off. What was the glitch you noticed?

Courty40, 2018-08-01:
Thought it may have been causing a relay twitch once in a while but I'm still seeing that (couple of ms click while on, enough to make an LED lamp flick)

Seems to coincide with a retry to MQTT as well, will keep an eye on it.

Paul

ascillato, 2018-08-22:
@Courty40 

Hi,

Have you solved your issue?

if not, please provide the output of `status 0` command.

I'm guessing, but your `wificonfig `is 4 (retry)? If it is, change to `wificonfig 5`

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

closed 2018-08-27 by ascillato2