arendst/Tasmota #3380: SmartConfig AP Password?

meingraham, 2018-07-30:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)

I have flashed a Sonoff Basic with 6.1.1b which I built from TASMOTA source and uploaded. All is well and I am able to connect to it and configure and operate it as expected.

I am trying to learn some of its admin features; particularly WIFI_MANAGER and WIFI_SMARTCONFIG. In my situation, I believe that my options set it to SmartConfig as the only available option (SSID/Password in user_config_override.h, USE_WPS not defined) by default. In looking at the [Button usage page](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage), I'm putting the device into WI-Fi configuration mode via four button presses. When I do, I can see the ITEAD-1000xxxx AP being broadcast. I could not find in the documentation what the password for the AP is. Please advise.

Thanks.

Mike

ascillato, 2018-07-30:
Hi Mike,

The AP from Tasmota is **Sonoff-XXXX** where **XXXX** is a number.

If you see **ITEAD-1000xxxx** that could means:

* Your Sonoff still have the original firmware from Itead

Or

* your Sonoff device still have some information in Flash from Itead. If this is the case, It is better to erase all flash and re flash again.

meingraham, 2018-07-30:
OK, I thought it was strange.

TASMOTA is definitely on the device (i.e., I connect to the web admin to manage the device). So, the flash memory still has the old ITead information, for sure.

Having said that, is there a reason why, since TASMOTA is on the device, the Sonoff-XXXX AP did not broadcast?

I'm using the Arduino IDE. I've seen it, but I can't find it now - where do I tell it to erase the contents of the flash memory as well as uploading the new firmware? Separately, can I tell it to erase flash memory during an OTA update when using the file upload method?

ascillato, 2018-07-31:
> Having said that, is there a reason why, since TASMOTA is on the device, the Sonoff-XXXX AP did not broadcast?

Yes, because of some mix wifi data in the flash memory. The ESP8266 Flash map has several parts besides the part that is stored Tasmota. There is the WIFI managing, and other things. So, if the flash process you use, just change the _user_software_ part, there will be some wifi config remaining in the flash.

> where do I tell it to erase the contents of the flash memory as well as uploading the new firmware?

Arduino have the _erase-flash-before-upload_ option only for esp8266 lib v2.4.1. For earlier versions (like the recommended v2.3.0) you will need to use the software **esptool.py**. How to download, install and use it is on the wiki.

> Separately, can I tell it to erase flash memory during an OTA update when using the file upload method?

No as far as I know.

meingraham, 2018-07-31:
@ascillato Adrian,

Thanks. I'm not going crazy. I knew I'd seen the erase flash setting.

I'll grab esptool.py and do some cleanup on my devices.

Regards.

Mike

closed 2018-07-31 by meingraham

mileper2nd, 2020-02-10:
I have had the same issue, the password for the SSID ITEAD-1000xxxx was 12345678. After the WiFi connection was established, I have went to 192.168.4.1, it opened some page where I have entered my WiFi SSID and password. Sonoff Bridge rebooted and connected to my normal WiFi.