arendst/Tasmota #3382: Yunshan Relay STATUS output every 5 seconds

meingraham, 2018-07-31:
TASMOTA 6.1.1b
Yunshan Relay Module

I stumbled across this when I was working on setting up my new Yunshan relay. I do not know what triggers it to start doing this. In the console, a 'STATUS' output is occurring approximately every 5 seconds. I went through a 'Reset 2' and reconfigured the device and it seemed to have stopped. However, sometime a few minutes later I notice the STATUS loop was back.

> 01:23:02 MQT: stat/{MQTT_Topic}/STATUS = {"Status":{"Module":33,"FriendlyName":["{FriendlyName}"],"Topic":"{MQTT_Topic}","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:40","StartupUTC":"2018-07-31T05:13:22","Sleep":250,"BootCount":6,"SaveCount":16,"SaveAddress":"F4000"}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS2 = {"StatusFWR":{"Version":"6.1.1b","BuildDateTime":"2018-07-30T23:04:44","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["{SSID}",""],"TelePeriod":300,"SetOption":["00008109","55818000"]}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS4 = {"StatusMEM":{"ProgramSize":424,"Free":576,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0F002B90","00000001","00000096","00000000"]}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS5 = {"StatusNET":{"Hostname":"{hostname}","IPAddress":"{IP},"Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"{MAC}","Webserver":2,"WifiConfig":5}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS6 = {"StatusMQT":{"MqttHost":"{mqtthost}","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_{MAC_ID}","MqttUser":"{mqttuser}","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 31 05:23:02 2018","Local":"Tue Jul 31 01:23:02 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"06:35","Sunset":"20:30"}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS10 = {"StatusSNS":{"Time":"2018-07-31T01:23:02"}}
01:23:02 MQT: stat/{MQTT_Topic}/STATUS11 = {"StatusSTS":{"Time":"2018-07-31T01:23:02","Uptime":"0T00:09:40","Vcc":3.140,"POWER":"ON","Wifi":{"AP":1,"SSId":"{SSID}","RSSI":96,"APMac":"{APMAC}"}}}

ascillato, 2018-07-31:
Do you have TasmoAdmin ?

meingraham, 2018-07-31:
My Bad - TasmoAdmin!!!!!!!!!

meingraham, 2018-07-31:
You beat me to it

closed 2018-07-31 by meingraham