arendst/Tasmota #3383: h801 Channel / Dimmer expected behaviour

nosaj66au, 2018-07-31:
Hi All this is not a problem more for me to understand how to work with the expected Tasmota operation.
I have many H801's controlling may individual light basically using them as 5 individual channels. The module is set in default H801 i.e Dimmer and Channel not PWM . From Node Red sends MQTT channelx nn commands to change a brightness of each channel individually. However at times it changes the levels other lights on the same H801.  I think its todo with the dimmer level changing but cant track it down.
Should I just change to PWM mode?

andrethomas, 2018-08-01:
^^ @renne

nosaj66au, 2018-08-06:
Hi @renne  can you please help / comment?

renne, 2018-08-06:
@nosaj66au 
I haven't seen this yet but you can switch to PWM. I'm going to configure the following Node-Red setup:
```
Slider(Kelvin) + Slider(Lux) -> Function(Kelvin2RGB * Lux) = PWM1, PWM2, PWM3
```
If you also use WW and CW stripes you can add some if-statement in Kelvin2RGB to use PWM4/PWM5 and disable PWM1/PWM2/PWM3 if the Kelvin range is between WW and CW. I'm also planning to use the APDS-9960 sensor for closed-loop brightness and color temperature control but haven't got gesture detection and RGBC sensing working at the same time, yet.

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks.

closed 2018-08-27 by ascillato2