arendst/Tasmota #3385: Sonoff reboots every 10 Mins constantly with exception

andreaslink-de, 2018-07-31:
Hej, I've installed/tested several versions from GitHub and am finally running Tasmota with v6.1.1c. I nearly deactivated all packages beside web-server, MQTT and ws2812b support as I want to switch on an 20A power supply with a Sonoff Basic which is driving a WS2812B led stripe. GPIO 14 is via 220Ohm connected to the Data-IN pin of the stripe. Currently the stripe is unplugged and only the power supply is connected to the output of my Sonoff Basic. WiFi connection works immediately and also a connection to my local broker (mosquitto) is successful.

Then I realized the relais flickering from time to time and via MQTT and corresponding I can see these (slightly varying) _RestartReasons_:

```
RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x227b3a22 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"

RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x73756168 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```

So there is for sure something wrong, which cannot be identified, so I turned the relais off and monitored the same exception (mainly the first one). 

I finally added a date/time to the cycles to measure the timeslots between the exceptions and reconnections and realized, it's more or less exactly 10 minutes, which indicates a watchdog or some other processing running into a limit, but I have no clue what this could be?

Between these 10 minute slots everything is fine and I can use everything pretty fast and responsive, also toggle relais on and off multiple times, so no issues at all.

**Anyone an idea, what makes my device "reboot" or at least "reconnect" every 10 minutes?** As it's so constant there must be something triggering this, which leads to the reboot exception.

Please find an extract of my logged monitoring here:
```
2018-07-31 00:24:54.286 sonoff/ws2812b/LWT offline
2018-07-31 00:25:03.225 sonoff/ws2812b/LWT online
2018-07-31 00:35:11.507 sonoff/ws2812b/LWT offline
2018-07-31 00:35:20.404 sonoff/ws2812b/LWT online
2018-07-31 00:45:28.696 sonoff/ws2812b/LWT offline
2018-07-31 00:45:35.643 sonoff/ws2812b/LWT online
2018-07-31 00:55:43.973 sonoff/ws2812b/LWT offline
2018-07-31 00:55:50.867 sonoff/ws2812b/LWT online
2018-07-31 01:05:59.191 sonoff/ws2812b/LWT offline
2018-07-31 01:06:08.145 sonoff/ws2812b/LWT online
2018-07-31 01:16:16.438 sonoff/ws2812b/LWT offline
2018-07-31 01:16:23.416 sonoff/ws2812b/LWT online
2018-07-31 01:26:31.786 sonoff/ws2812b/LWT offline
2018-07-31 01:26:38.729 sonoff/ws2812b/LWT online
2018-07-31 01:36:47.035 sonoff/ws2812b/LWT offline
2018-07-31 01:36:53.964 sonoff/ws2812b/LWT online
2018-07-31 01:47:02.281 sonoff/ws2812b/LWT offline
2018-07-31 01:47:11.216 sonoff/ws2812b/LWT online
2018-07-31 01:57:19.554 sonoff/ws2812b/LWT offline
2018-07-31 01:57:28.526 sonoff/ws2812b/LWT online
2018-07-31 02:07:36.841 sonoff/ws2812b/LWT offline
2018-07-31 02:07:43.784 sonoff/ws2812b/LWT online
2018-07-31 02:17:52.100 sonoff/ws2812b/LWT offline
2018-07-31 02:17:58.998 sonoff/ws2812b/LWT online
2018-07-31 02:28:07.360 sonoff/ws2812b/LWT offline
2018-07-31 02:28:14.262 sonoff/ws2812b/LWT online
2018-07-31 02:38:22.589 sonoff/ws2812b/LWT offline
2018-07-31 02:38:31.562 sonoff/ws2812b/LWT online
2018-07-31 02:48:39.857 sonoff/ws2812b/LWT offline
2018-07-31 02:48:48.801 sonoff/ws2812b/LWT online
2018-07-31 02:58:57.157 sonoff/ws2812b/LWT offline
2018-07-31 02:59:04.038 sonoff/ws2812b/LWT online
2018-07-31 03:09:12.377 sonoff/ws2812b/LWT offline
2018-07-31 03:09:19.306 sonoff/ws2812b/LWT online
2018-07-31 03:19:27.636 sonoff/ws2812b/LWT offline
2018-07-31 03:19:36.573 sonoff/ws2812b/LWT online
2018-07-31 03:29:44.885 sonoff/ws2812b/LWT offline
2018-07-31 03:29:51.843 sonoff/ws2812b/LWT online
2018-07-31 03:40:00.179 sonoff/ws2812b/LWT offline
2018-07-31 03:40:07.119 sonoff/ws2812b/LWT online
2018-07-31 03:50:15.436 sonoff/ws2812b/LWT offline
2018-07-31 03:50:22.388 sonoff/ws2812b/LWT online
2018-07-31 04:00:30.729 sonoff/ws2812b/LWT offline
2018-07-31 04:00:39.659 sonoff/ws2812b/LWT online
2018-07-31 04:10:47.982 sonoff/ws2812b/LWT offline
2018-07-31 04:10:56.939 sonoff/ws2812b/LWT online
2018-07-31 04:21:05.244 sonoff/ws2812b/LWT offline
2018-07-31 04:21:12.125 sonoff/ws2812b/LWT online
2018-07-31 04:31:20.445 sonoff/ws2812b/LWT offline
2018-07-31 04:31:29.332 sonoff/ws2812b/LWT online
[... 100 lines cut ...]
**Provide the output of command ``status 0`` :**
```
sonoff/ws2812b/command/status 0
sonoff/ws2812b/STATUS {"Status":{"Module":1,"FriendlyName":["Kinderhaus-Relais","Kinderhaus-WS2812B"],"Topic":"sonoff-kinderhaus-led","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

sonoff/ws2812b/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.253/arduino/firmware/esp8266/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:09:52","StartupUTC":"2018-07-29T17:49:36","Sleep":0,"BootCount":148,"SaveCount":204,"SaveAddress":"FA000"}}

sonoff/ws2812b/STATUS2 {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018.07.29 00:31:55","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}

sonoff/ws2812b/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MyPersonalWiFi",""],"TelePeriod":300,"SetOption":["00008089","55A18000","00000001"]}}

sonoff/ws2812b/STATUS4 {"StatusMEM":{"ProgramSize":428,"Free":572,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0000A7B0","00000000","00000096","00000000"]}}

sonoff/ws2812b/STATUS5 {"StatusNET":{"Hostname":"Kinderhaus-WS2812B","IPAddress":"192.168.0.152","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.253","Mac":"5C:CF:7F:57:85:0A","Webserver":2,"WifiConfig":2}}

sonoff/ws2812b/STATUS6 {"StatusMQT":{"MqttHost":"192.168.0.253","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_57850A","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}

sonoff/ws2812b/STATUS7 {"StatusTIM":{"UTC":"Sun Jul 29 17:59:28 2018","Local":"Sun Jul 29 18:59:28 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}

sonoff/ws2812b/STATUS10 {"StatusSNS":{"Time":"2018-07-29T18:59:28"}}

sonoff/ws2812b/STATUS11 {"StatusSTS":{"Time":"2018-07-29T18:59:28","Uptime":"0T00:09:52","Vcc":3.475,"POWER1":"OFF","POWER2":"OFF","Dimmer":26,"Color":"424242","HSBColor":"0,0,26","Channel":[26,26,26],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"MyPersonalWiFi","RSSI":78,"APMac":"8A:8A:20:4C:C0:15"}}}
```

/Andreas

Jason2866, 2018-07-31:
Did you erase the flash completly before you flashed Tasmota? ->  Esptool erase_flash
Fragments left can make trouble. If this didnt help change the module. Could simply be defect.
I had bad experience with arduino core 2.4.0 and 2.4.1. 
For reliable setups there is just one choice Arduino core 2.3.0

andreaslink-de, 2018-07-31:
Thanks for the fast response, and yes I fully cleared it before flashing (using Arduino IDE and "All Flash contents"). I read somewhere else of downgrading to 2.3.0 in relation to WiFi issues, but I cannot believe this might trigger these issues? Everything is fine and stable between the reboots, so no issues at all, which makes it so weird.

Jason2866, 2018-07-31:
Try 2.3.0 😎  I am pretty sure your problems will left. 
2.4.1 makes funny problems especially wifi.

Jason2866, 2018-07-31:
Just my 2 cents. For bigger projects like Tasmota Arduino IDE is a pain...
Try Platformio or Visual Studio Code. Setting up both for Tasmota is described in wiki.
You will never use Arduino for Tasmota and other projects with more source files...

andreaslink-de, 2018-07-31:
I think, I identified the issue now. Being back on v2.4.1, WiFi connect takes longer, but finally connects, so there is no problem. 
I just flashed it and did not do any changes to the setup, so in the current setup, there is no specific GPIO14 setup done. When I keep it as it is, it runs stable for at least have an hour now, so no 10 minute crashes anymore, which is interesting.

BUT the moment I activate this setup and set WS2812 for GPIO 14, which triggers a reboot...

[image]

...I get the typical static every 10 minutes reboots/resets. So could there something be wrong with that library? Is a watchdog ignored, when this library is activated and what could be wrong? Could anyone else test his in his setup?

This is how I chose the modules in "user_config.h", before compiling:

```
[...]
/*********************************************************************************************\
 * END OF SECTION 1
 *
 * SECTION 2
 * - Enable a feature by removing both // in front of it
 * - Disable a feature by preceding it with //
\*********************************************************************************************/

// -- Localization --------------------------------
  // If non selected the default en-GB will be used
//#define MY_LANGUAGE            bg-BG           // Bulgarian in Bulgaria
//#define MY_LANGUAGE            cs-CZ           // Czech in Czech
#define MY_LANGUAGE            de-DE           // German in Germany
//#define MY_LANGUAGE            el-GR           // Greek in Greece
//#define MY_LANGUAGE            en-GB           // English in Great Britain. Enabled by Default
//#define MY_LANGUAGE            es-AR           // Spanish in Argentina
//#define MY_LANGUAGE            fr-FR           // French in France
//#define MY_LANGUAGE            hu-HU           // Hungarian in Hungary
//#define MY_LANGUAGE            it-IT           // Italian in Italy
//#define MY_LANGUAGE            nl-NL           // Dutch in the Netherlands
//#define MY_LANGUAGE            pl-PL           // Polish in Poland
//#define MY_LANGUAGE            pt-BR           // Portuguese in Brazil
//#define MY_LANGUAGE            pt-PT           // Portuguese in Portugal
//#define MY_LANGUAGE            ru-RU           // Russian in Russia
//#define MY_LANGUAGE            uk-UK           // Ukrainian in Ukrain
//#define MY_LANGUAGE            zh-CN           // Chinese (Simplified) in China
//#define MY_LANGUAGE            zh-TW           // Chinese (Traditional) in Taiwan

// -- Wifi Config tools ---------------------------
//#define USE_WPS                                  // Add support for WPS as initial wifi configuration tool (+33k code, 1k mem (5k mem with core v2.4.2+))
//#define USE_SMARTCONFIG                          // Add support for Wifi SmartConfig as initial wifi configuration tool (+23k code, +0.6k mem)

// -- OTA -----------------------------------------
#define USE_ARDUINO_OTA                          // Add optional support for Arduino OTA (+13k code)

/*-------------------------------------------------------------------------------------------*\
 * Select ONE of possible three MQTT library types below
\*-------------------------------------------------------------------------------------------*/
  // Default MQTT driver for both non-TLS and TLS connections. Blocks network if MQTT server is unavailable.
#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT   // Use PubSubClient library
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT    // Use TasmotaMqtt library (+4k4 code, +4k mem) - non-TLS only
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_ESPMQTTARDUINO // Use (patched) esp-mqtt-arduino library (+4k8 code, +4k mem) - non-TLS only

// -- MQTT ----------------------------------------
#define MQTT_TELE_RETAIN     0                   // Tele messages may send retain flag (0 = off, 1 = on)

// -- MQTT - Domoticz -----------------------------
//#define USE_DOMOTICZ                             // Enable Domoticz (+6k code, +0.3k mem)
  #define DOMOTICZ_IN_TOPIC    "domoticz/in"     // Domoticz Input Topic
  #define DOMOTICZ_OUT_TOPIC   "domoticz/out"    // Domoticz Output Topic

// -- MQTT - Home Assistant Discovery -------------
//#define USE_HOME_ASSISTANT                       // Enable Home Assistant Discovery Support (+2k code)
  #define HOME_ASSISTANT_DISCOVERY_PREFIX "homeassistant"  // Home Assistant discovery prefix

[... 115 lines cut ...]

/*********************************************************************************************\
 * Select KNX without Emulation to save space
\*********************************************************************************************/

//#define USE_KNX_NO_EMULATION                     // Create sonoff-knx with KNX but without Emulation (See sonoff_post.h)

/*********************************************************************************************\
 * Compile a minimal version if upgrade memory gets tight ONLY TO BE USED FOR UPGRADE STEP 1!
 *   To be used as step 1 during upgrade.
 *   Step 2 is re-compile with option BE_MINIMAL commented out.
\*********************************************************************************************/

//#define BE_MINIMAL                               // Minimal version if upgrade memory gets tight (-45k code, -2k mem)

/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

#endif  // _USER_CONFIG_H_
```

Frogmore42, 2018-08-01:
How are you powering your led strip?

I have a NodeMCU board that probably has a defective voltage regulator.  It reboots all the time.  Sometimes it will stay up for 175 hours, at other time it can't even stay up for an hour.

Jason2866, 2018-08-01:
I am with @Frogmore42 
Change the module, it is defect
I have a WS2812 Strip running on a Arilux Module without any restarts since last update to 6.1.1b
(16 days)

andreaslink-de, 2018-08-01:
Hmm, as I said I use a 20A power supply for the LEDs, but currently nothing of it is attached during my tests. I've only my naked Sonoff Basic, flashed via Arduino according to the setup as posted above. It is stable until I set "WS2812" for GPIO14 and I'm pretty confident, this is a software issue as it's so strict and precise with the 10 minutes reboot slots and only triggered, when this setup is changed.

It would be great if @arendst might give a hint here, I've seen some watchdog handling in the code, but it's pretty complex tied into the code. Could there be some requirement I need in relation to the used WS2812 library?

Never the less, I can also flash it on another new one, I should have another one lying around. Probably I can give a try today in the evening.

Frogmore42, 2018-08-01:
I would suggest trying one of the released binaries and capturing the complete serial log from the device.  You said you removed lots of things.  It is possible that you removed too much by changing flags in a way that no one else has tried before.  I usually use the all sensors or its equivalent.  That way I can use the same firmware on all my devices and don't have to make and keep track of several versions.

You are correct that if it is exactly at 10min that is probably a soft/firmware issue.

But, I know many people are using the WS2812 library and I have not heard of anyone else having this issue.

So, try the released version and see if it has the same issue.  It should be pretty easy to tell, since the failure seems consistent.

andreaslink-de, 2018-08-01:
> You said you removed lots of things. It is possible that you removed too much by changing flags in a way that no one else has tried before.

And here I fully agree, this is also my major concern. I removed a lot as the Sonoff Basic only has 1M of flash and I wanted to be below 500kB in total just to be sure, to be able to deploy OTA updates later. But based on the instability and probably unforseen dependencies between modules, it's not that wise to go that way. So just put everything in and live without future OTA updates. I'll consider this approach as a possible solution.

But never the less I would be interested in, where and what the root cause is being responsible for the 10 min reboot :-).

Jason2866, 2018-08-01:
@andreaslink-de 
I fired up my wemos mini with the 24 pixel ring running scheme 11 (latest 6.1.1c dev with core 2.3.0)

[image]

Frogmore42, 2018-08-01:
There was a time where I worried about how big the binary was, but then Theo implemented the two step upgrade with the minimal bin.  That has enough room to last a little while.  I believe there will come a time when it will be necessary to make some choices about exactly which sensors I need.

The proper way (always works) is to warp the #under in a #ifdef as shown in the wiki/sample file.  If you know the value is defined, because of other choices you made, you omit the #ifdef.  If you omit the #undef, and try to redefine it, you will get a compiler error.

Until you try the full released binary, there is no way to know if the problem was that or really a hardware issue or possibly something else.  If it is a software modules inclusion issue, it will take someone really motivated to figure out the issue.  Perhaps that is you 😃.

Jason2866, 2018-08-01:
@andreaslink-de 
Yes, user_config.h is fully untouched. If i download a new version i just have to unpack and copy
just the file user_config_override.h and change in platformio.ini the entry
`; -DUSE_CONFIG_OVERRIDE`
 to   
`-DUSE_CONFIG_OVERRIDE`

and hit compile button. Thats it.  Its done in 1 minute.

andreaslink-de, 2018-08-02:
So I finally followed @Frogmore42 suggestion:
> So, try the released version and see if it has the same issue. It should be pretty easy to tell, since the failure seems consistent.

And it turned out, this is the ONLY way to get rid of the 10 minutes consitently dying Sonoff :-(. So somehow a success (on the same hardware by the way), but no real explanation; I took the sonoff-DE.bin and successfully flashed it (without deleting my setup) and I have no hick ups anymore. 

So I did the same on my own again, cloned the repository, did no modifications compiled and uploaded and I'm back with the continously 10 minute restarts. What the hack is going wrong? I'm fully following the suggested Arduino IDE flash recommended params. Why are my compiles or flashes different? I even copied all libraries again from the cloned repro into my lib folder.

So now I think it's an issue with my compile (running newest Arduino IDE unter Linux Mint), what could I oversee?

arendst, 2018-08-02:
Just for my curiosity have a look at a hex dump of your compiled binary and check how byte alignment is being used. The below hexdump is taken almost at the end of the binary where the data table is stored.

[image]

The left part shows a correctly aligned binary compiled in Windows. The right part shows a mis-aligned binary compiled on Linux by Travis. 

Using the mis-aligned binary users encountered difficult wifi connections and probably corrupt webpages. I guess the WS2812 routines also won't like mis-alignment...

I expect there must be a xtensa cross-compiler directive to use correctly byte alignment but as of now I don't know what it is.

Jason2866, 2018-08-02:
So everything fits together. I never had problems because i use Windows environment...

andrethomas, 2018-08-03:
@Jason2866 @arendst I guess being a 32bit cpu the byte alignment should be at least 4 bytes but it may also be flash dependent. 

The alignment is not specific to a variable but rather as to where the blocks defined in the linker script forms the boundaries in the binary when performing the linking process.

I had to do this when I was writing code for STM32F4 microcontrollers otherwise I would get random failures on variables defined in PROGMEM because the functions that made provision for the data to be read from F had to read the content in 4 byte blocks (e.g. 4 byte word) so having a PROGMEM start with an offset of -1 or +1 on a 4 byte read caused exceptions.

Taking a peek at the eagle.app.v6.common.ld which seems to be included from the linker script its interesting to note that some areas are 4 byte, 8 byte and even 16 byte aligned - go check the . = ALIGN (); statements added to the subsections.

A logical explanation for why the sonoff-DE.bin version worked for @andreaslink-de _could_ be that the PROGMEM's just happen to align correctly insofar that a segmentation fault isn't created.

autofrank, 2018-08-06:
I had a RFBridge (V1, no screws) that rebooted every 5 mins. A reflash of the firmware (over serial) and it solved the issue.

ascillato2, 2018-08-22:
Hi,

Any updates on this issue or we can close it?

andreaslink-de, 2018-08-22:
@arendst You are fully right, I've checked your HEX example and compared it with one I've compiled on my Linux machine "as I usually do". I tried to find nearly the same section as seen on your screenshot and I can proof mine is also "slimmer" than your Windows one (spacing with two "dots" (0x00) instead of one "dot":
[image]

This is a major new info for me, as I always thought compiled is compiled and the OS does not influence it as I expect the Arduino IDE and "hidden" ESP8266 compiler flags and linker details to be equal over systems. That there are now differences moves everything into a fully new light and could have several other consequences on other projects I did/compiled.

So thanks for this info, that at least answers and explains my initial request - even though I need to find a solution under Linux. Probably with this learning it could be wise to add a comment either in the wiki or somewhere meaningful, so that others are pointed on this challenge with Linux or compiler/linker flags being used, do you agree?

@andrethomas your experience and suggestions with PROGMEM sounds interesting. Is there anything I can do to prove this or make somewhere changes to my linker? I did not exactly get, where I can find the `ALIGN();` part? My Linux machine is running a common i7 64bit CPU.

Beside this, I guess I can close this topic now. Thank you all for your effort in analyzing this.

PS: Sorry for late response, had been ill some days in bed; this one was not forgotten :).

closed 2018-08-22 by andreaslink-de