arendst/Tasmota #3386: No STATE or SENSOR via MQTT - RSL instead of MQT

hofismp, 2018-07-31:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : gosund SP1 aka BlitzWolf SHP2
- [x] Provide the output of command ``status 0`` :
```
20:25:28 RSL: home/pow001/stat/STATUS = {"Status":{"Module":45,"FriendlyName":["pow001"],"Topic":"pow001","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:25:28 RSL: home/pow001/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:33:20","StartupUTC":"2018-07-31T18:52:08","Sleep":0,"BootCount":11,"SaveCount":56,"SaveAddress":"F4000"}}
20:25:28 RSL: home/pow001/stat/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-30T21:13:53","Boot":4,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
20:25:28 RSL: home/pow001/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxx",""],"TelePeriod":60,"SetOption":["00008009","55818000"]}}
20:25:28 RSL: home/pow001/stat/STATUS4 = {"StatusMEM":{"ProgramSize":571,"Free":368,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
20:25:28 RSL: home/pow001/stat/STATUS5 = {"StatusNET":{"Hostname":"pow001-3367","IPAddress":"192.168.178.110","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"BC:DD:C2:xx:xx:xx","Webserver":2,"WifiConfig":4}}
20:25:28 RSL: home/pow001/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.xxx.xxx","MqttPort":1883,"MqttClientMask":"pow001","MqttClient":"pow001","MqttUser":"pow001","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:25:28 RSL: home/pow001/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jul 31 19:25:28 2018","Local":"Tue Jul 31 20:25:28 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:22","Sunset":"20:30"}}
20:25:28 RSL: home/pow001/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:25:28 RSL: home/pow001/stat/STATUS10 = {"StatusSNS":{"Time":"2018-07-31T20:25:28","ENERGY":{"Total":0.011,"Yesterday":0.006,"Today":0.005,"Power":4,"Factor":0.44,"Voltage":231,"Current":0.042}}}
20:25:28 RSL: home/pow001/stat/STATUS11 = {"StatusSTS":{"Time":"2018-07-31T20:25:28","Uptime":"0T00:33:20","Vcc":3.540,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","RSSI":90,"APMac":"78:8A:20:xx:xx:xx"}}}
```

Hi there,

i flashed a gosund SP1 yesterday and everything is working fine except one thing.
Telemetry (STATE/SENSOR) gets displayed fine in the console but not transmitted via MQTT.
Every Console-Message containing "MQT:" in front gets delivered fine.
Even if i publish STATUS 10 via MQTT i get the Output only on the console, beginning with RSL instead of MQT.
I think it's not a telemetry-problem in general, because i receive the telemetry-topics LWT, INFO1, INFO2, INFO3 on every reboot of the SP1.

Here is a short snippet showing the STATE and SENSOR working in general but not to mqtt

```
...
20:42:45 RSL: home/pow001/tele/STATE = {"Time":"2018-07-31T20:42:45","Uptime":"0T00:10:15","Vcc":3.538,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","RSSI":86,"APMac":"78:8A:20:xx:xx:xx"}}
20:42:45 RSL: home/pow001/tele/SENSOR = {"Time":"2018-07-31T20:42:45","ENERGY":{"Total":0.007,"Yesterday":0.006,"Today":0.001,"Period":0,"Power":3,"Factor":0.40,"Voltage":232,"Current":0.028}}
20:43:45 RSL: home/pow001/tele/STATE = {"Time":"2018-07-31T20:43:45","Uptime":"0T00:11:15","Vcc":3.538,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","RSSI":86,"APMac":"78:8A:20:xx:xx:xx"}}
20:43:45 RSL: home/pow001/tele/SENSOR = {"Time":"2018-07-31T20:43:45","ENERGY":{"Total":0.007,"Yesterday":0.006,"Today":0.001,"Period":0,"Power":3,"Factor":0.40,"Voltage":232,"Current":0.027}}
20:44:45 RSL: home/pow001/tele/STATE = {"Time":"2018-07-31T20:44:45","Uptime":"0T00:12:15","Vcc":3.538,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","RSSI":84,"APMac":"78:8A:20:xx:xx:xx"}}
20:44:45 RSL: home/pow001/tele/SENSOR = {"Time":"2018-07-31T20:44:45","ENERGY":{"Total":0.007,"Yesterday":0.006,"Today":0.001,"Period":0,"Power":2,"Factor":0.38,"Voltage":232,"Current":0.026}}
...
```

Any ideas?
Thanks in advance and best regards
Hofi

Jason2866, 2018-07-31:
Have you checked mqtt connection?
RSL means the device has no connection to the mqtt broker
I use the gosund too:
22:06:42 MQT: tele/sonoff-16CE26/STATE = {"Time":"2018-07-31T22:06:42","Uptime":"3T03:04:47","Vcc":3.218,"POWER":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":66,"APMac":"00:A0:57:1E:69:50"}}
22:06:42 MQT: tele/sonoff-16CE26/SENSOR = {"Time":"2018-07-31T22:06:42","ENERGY":{"Total":15.718,"Yesterday":0.664,"Today":0.497,"Period":2,"Power":24,"Factor":0.46,"Voltage":234,"Current":0.223}}

hofismp, 2018-07-31:
MQTT is working absolutely fine, i can send commands via mqtt to power on/off and also receive the results via mqtt if i press the button on the SP1.
Everything (also mqtt) is working fine except State and Sensor -> MQTT.

Just pressed the button to show the console-results:
```
21:26:46 RSL: home/pow001/tele/STATE = {"Time":"2018-07-31T21:26:46","Uptime":"0T00:54:16","Vcc":3.538,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","RSSI":84,"APMac":"78:8A:20:xx:xx:xx"}}
21:26:46 RSL: home/pow001/tele/SENSOR = {"Time":"2018-07-31T21:26:46","ENERGY":{"Total":0.007,"Yesterday":0.006,"Today":0.001,"Period":0,"Power":0,"Factor":0.00,"Voltage":231,"Current":0.000}}
21:27:06 MQT: home/pow001/stat/RESULT = {"POWER":"OFF"}
21:27:06 MQT: home/pow001/stat/POWER = OFF
21:27:08 MQT: home/pow001/stat/RESULT = {"POWER":"ON"}
21:27:08 MQT: home/pow001/stat/POWER = ON
21:27:46 RSL: home/pow001/tele/STATE = {"Time":"2018-07-31T21:27:46","Uptime":"0T00:55:16","Vcc":3.538,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","RSSI":84,"APMac":"78:8A:20:xx:xx:xx"}}
21:27:46 RSL: home/pow001/tele/SENSOR = {"Time":"2018-07-31T21:27:46","ENERGY":{"Total":0.007,"Yesterday":0.006,"Today":0.001,"Period":0,"Power":0,"Factor":0.00,"Voltage":232,"Current":0.000}}
```

Jason2866, 2018-07-31:
I think you have somewhere a mistake in your mqtt topic definition.
Try a clean setup and change nearly nothing in this part.
You find a good explanation in the wiki in the setup for OpenHab

hofismp, 2018-07-31:
I had the SP1 on tasmota stock-settings yesterday, changing the topics to my standard was just another try to get it running somehow and i also saw the order xxx/%topic%/%prefix%/ in some official tasmota documents, so it should be valid :(

wongnam, 2018-08-01:
@hofismp Your full topic look strange. let try to use the TASMOTA default full topic first, if any thing is ok then you can change them to your favorite.
it should be like this: %prefix%/%topic%/

`10:22:54 MQT: **tele/smartplug/STATE** = {"Time":"2018-08-01T10:22:54","Uptime":"0T00:31:22","Vcc":3.162,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
10:22:54 MQT: **tele/smartplug/SENSOR** = {"Time":"2018-08-01T10:22:54","AM2301":{"Temperature":31.2,"Humidity":73.0},"TempUnit":"C"}
10:23:56 MQT: tele/smartplug/STATE = {"Time":"2018-08-01T10:23:56","Uptime":"0T00:32:24","Vcc":3.160,"POWER":"ON","Wifi":{"AP":1,"SSId":"Multimedia Wi-Fi Network","RSSI":100,"APMac":"B0:C7:45:36:85:E0"}}
10:23:56 MQT: tele/smartplug/SENSOR = {"Time":"2018-08-01T10:23:56","AM2301":{"Temperature":31.3,"Humidity":72.8},"TempUnit":"C"}`

hofismp, 2018-08-01:
Thanks so far, i'm still confused that almost all topics always work and two others dont, where is the crucial difference...
However, i'll start from scratch tomorrow and setup a dedicated tasmota-environment with VSC/PlatformIO and core 2.3.0.
I'll give feedback then.
BR
Hofi

hofismp, 2018-08-01:
Maybe i found at least the difference beween the working and non working topics.
It seems that every topic with a payload containing double-curly brackets gets the RSL ...
Will see if it works with the new flash tomorrow.
Take care

hofismp, 2018-08-02:
I flashed again with PlatformIO and core 2.3.0 and it's working! It even kept all my setttings.

```
18:48:18 MQT: home/pow001/tele/STATE = {"Time":"2018-08-02T18:48:18","Uptime":"0T00:04:18","Vcc":3.252,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","RSSI":80,"APMac":"78:8A:20:xx:xx:xx"}}
18:48:19 MQT: home/pow001/tele/SENSOR = {"Time":"2018-08-02T18:48:18","ENERGY":{"Total":0.015,"Yesterday":0.007,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":230,"Current":0.000}}
```

Thanks for all your suggestions :)

closed 2018-08-02 by hofismp

Far-Ken, 2018-08-12:
Hi hofismp,
Encountering the exact same issue. Could you please point me in the right direction on how to switch to core 2.3.0 ?

Arduino 1.8.5
Hardware:
NodeMCU v1.0 (ESP-12E Module)
Sonoff-Tasmota 6.1.1c by Theo Arends
SI7021 Temperature and Humidity sensor on pins
SCL d6 gpio12
SDA d5 gpio14

Frogmore42, 2018-08-12:
The information is in the wiki and a few other issues, search for core 2.3.0 and I am sure you will find it.

Frogmore42, 2018-08-12:
See https://github.com/arendst/Sonoff-Tasmota/issues/3427 and the other previous issue that it links to.

kbhuinfo, 2018-09-12:
VSCode+PlatformIO made my day too!

kxn25, 2019-02-04:
Iam also having problems with RSL instead of mqt.

[image]

I have read the thread but for me it does not work.
i dont have issues with my Topic nor with the connection to ioBroker...
iam having this with all my Tasmota from time to time,
On the Sonoff Toch and the Wemos D1 mini.
i have tryed the Version 6.3.0, 6.4.0 and 5.2.0 and always come up with the same Promlem...

i quit hant understood the core thing?
how can i check which core is running in what version?

ggoelles, 2019-02-17:
I had to change the MQTT client in the config to Arduion MQTT client.

Search for MQTT_LIBRARY_TYPE in "my_user_config.h":
#define MQTT_LIBRARY_TYPE      MQTT_ARDUINOMQTT

ggoelles, 2019-02-17:
> Iam also having problems with RSL instead of mqt.
> 
> its all uploading fine and its running for days until all of a sudden this happens:
> [image]
> When i resart the device its all back to normal... until...
> 
> I have read the thread but for me it does not work.
> i dont have issues with my Topic nor with the connection to ioBroker...
> iam having this with all my Tasmota from time to time,
> On the Sonoff Toch and the Wemos D1 mini.
> i have tryed the Version 6.3.0, 6.4.0 and 5.2.0 and always come up with the same Promlem...
> 
> i quit hant understood the core thing?
> how can i check which core is running in what version?

I had to change the MQTT client in the config to Arduion MQTT client.

Search for MQTT_LIBRARY_TYPE in "my_user_config.h":
#define MQTT_LIBRARY_TYPE MQTT_ARDUINOMQTT

kxn25, 2019-02-17:
@ggoelles thank you for your replay i try it out tomorrow.

:-)

kxn25, 2019-02-19:
@ggoelles 
I have flashed Thema noch with your settings. All went well   
I will See the next few days if it worked. 

Thank you  for your help.

suprimex, 2019-03-06:
With MQTT_LIBRARY_TYPE=MQTT_PUBSUBCLIENT  I have similar troubles after update to the latest version.  Some messages go to MQT,  but the state - only to RSL:

`12:17:42 CMD: cmnd/state
12:17:42 RSL: tele/esp12s-small/STATE = {"Time":"2019-03-06T12:17:42","Uptime":"0T00:27:16","SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"Wifi":{"AP":1,"SSId":"WLAN","BSSId":"F8:1A:67:BD:C1:80","Channel":1,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:08"}}
12:17:42 RSL: stat/esp12s-small/RESULT = {"Time":"2019-03-06T12:17:42","Uptime":"0T00:27:16","SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"Wifi":{"AP":1,"SSId":"WLAN","BSSId":"F8:1A:67:BD:C1:80","Channel":1,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:08"}}
12:20:44 RSL: tele/esp12s-small/STATE = {"Time":"2019-03-06T12:20:44","Uptime":"0T00:30:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"WLAN","BSSId":"F8:1A:67:BD:C1:80","Channel":1,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:08"}}
12:20:44 MQT: tele/esp12s-small/SENSOR = {"Time":"2019-03-06T12:20:44","ANALOG":{"A0":93}}
`

with MQTT_LIBRARY_TYPE=MQTT_ARDUINOMQTT - it seems problem gone:
`12:59:01 MQT: tele/rollerLive1/STATE = {"Time":"2019-03-06T12:59:01","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"Wifi":{"AP":1,"SSId":"WLAN","BSSId":"F8:1A:67:BD:C1:80","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:59:01 MQT: tele/rollerLive1/SENSOR = {"Time":"2019-03-06T12:59:01","ANALOG":{"A0":127}}
13:02:00 MQT: tele/rollerLive1/UPTIME = {"Time":"2019-03-06T13:02:00","Uptime":"0T00:03:13"}
13:02:18 CMD: cmnd/state
13:02:18 MQT: stat/rollerLive1/RESULT = {"Time":"2019-03-06T13:02:18","Uptime":"0T00:03:31","SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"Wifi":{"AP":1,"SSId":"WLAN","BSSId":"F8:1A:67:BD:C1:80","Channel":1,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:04"}}
13:04:01 MQT: tele/rollerLive1/STATE = {"Time":"2019-03-06T13:04:01","Uptime":"0T00:05:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"WLAN","BSSId":"F8:1A:67:BD:C1:80","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
13:04:01 MQT: tele/rollerLive1/SENSOR = {"Time":"2019-03-06T13:04:01","ANALOG":{"A0":131}}`

samezrp, 2019-03-30:
> Search for MQTT_LIBRARY_TYPE in "my_user_config.h":
> #define MQTT_LIBRARY_TYPE MQTT_ARDUINOMQTT

Guys, do you have directive MQTT_LIBRARY_TYPE in "my_user_config.h" at all?
I checked in clean Tasmota 6.5 and this is present only in two files:
_changelog.ino
support_features.ino

I do not think it is even used. In ver 6.2.1 it was present in user_config.h
In my_user_config.h whole below part disapeared:

/*-------------------------------------------------------------------------------------------*\
 * Select ONE of possible three MQTT library types below
\*-------------------------------------------------------------------------------------------*/
  // Default MQTT driver for both non-TLS and TLS connections. Blocks network if MQTT server is unavailable.
#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT   // Use PubSubClient library
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT    // Use TasmotaMqtt library (+4k4 code, +4k mem) - non-TLS only
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_ESPMQTTARDUINO // Use (patched) esp-mqtt-arduino library (+4k8 code, +4k mem) - non-TLS only

arendst, 2019-03-30:
See https://github.com/arendst/Sonoff-Tasmota/releases

samezrp, 2019-03-30:
OK, so leaving code in:
support_features.ino
is misleading.
BTW it looks like switching OFF, restart and ON again MQTT in "configuration/configure other" helped.

kxn25, 2019-03-30:
Yen IT helped me too. But After a while IT will Come again.
The mqtt Chance to arduino did the Job

g146m026, 2019-05-25:
I have the same issue but in tasmota 6.5.
When i plug only one sensor, i got MQT message
But when I plug more than one sensor, i got only RSL message.

Anyone can help me ?

cruisinger, 2019-06-08:
same problem here:
I'm using several tasmota devices. After updateing from v5.12.0.j  to 6.5.0.15 all sonoff and esp12 with more then one configured sensors doesn't send mqtt sensor, but info messages. Instead mqt sensor, I only can see RSL messages in console.
restarting and resetting doesn't help.

00:00:00 CFG: Loaded from flash at F4, Count 8
00:00:00 Project speisekammer Sonoff Version 6.5.0.15(sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 AP in mode 11N as speisekammer-4501...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on speisekammer-4501 with IP address 192.168.1.98
17:48:32 MQT: Attempting connection...
17:48:32 MQT: Connected
17:48:32 MQT: tele/speisekammer/LWT = Online (retained)
17:48:32 MQT: cmnd/speisekammer/POWER = 
17:48:32 RSL: tele/speisekammer/INFO1 = {"Module":"Sonoff Basic","Version":"6.5.0.15(sonoff)","FallbackTopic":"cmnd/DVES_957195_fb/","GroupTopic":"sonoffs"}
17:48:32 MQT: tele/speisekammer/INFO2 = {"WebServerMode":"Admin","Hostname":"speisekammer-4501","IPAddress":"192.168.1.98"}
17:48:32 MQT: tele/speisekammer/INFO3 = {"RestartReason":"Software/System restart"}
17:48:32 MQT: stat/speisekammer/RESULT = {"POWER":"OFF"}
17:48:32 MQT: stat/speisekammer/POWER = OFF
17:48:41 RSL: tele/speisekammer/STATE = {"Time":"2019-06-08T17:48:41","Uptime":"0T00:00:17","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AP","BSSId":"E0:28:6D:68:37:5B","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:48:41 RSL: tele/speisekammer/SENSOR = {"Time":"2019-06-08T17:48:41","Switch1":"OFF","AM2301":{"Temperature":24.4,"Humidity":38.0},"TempUnit":"C"}
17:48:51 RSL: tele/speisekammer/STATE = {"Time":"2019-06-08T17:48:51","Uptime":"0T00:00:27","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AP","BSSId":"E0:28:6D:68:37:5B","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:48:51 RSL: tele/speisekammer/SENSOR = {"Time":"2019-06-08T17:48:51","Switch1":"OFF","AM2301":{"Temperature":24.4,"Humidity":38.1},"TempUnit":"C"}
17:49:01 RSL: tele/speisekammer/STATE = {"Time":"2019-06-08T17:49:01","Uptime":"0T00:00:37","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AP","BSSId":"E0:28:6D:68:37:5B","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:49:01 RSL: tele/speisekammer/SENSOR = {"Time":"2019-06-08T17:49:01","Switch1":"OFF","AM2301":{"Temperature":24.4,"Humidity":38.0},"TempUnit":"C"}

dmitriytim, 2019-12-05:
:) 

`/*
  tasmota.ino - Tasmota firmware for iTead Sonoff, Wemos and NodeMCU hardware

  Copyright (C) 2019  Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
/*====================================================
  Prerequisites:
    - Change libraries/PubSubClient/src/PubSubClient.h
        #define MQTT_MAX_PACKET_SIZE 1000

    - Select IDE Tools - Flash Mode: "DOUT"
    - Select IDE Tools - Flash Size: "1M (no SPIFFS)"
  ====================================================*/`

ascillato, 2019-12-05:
Great that you found your issue. 

If you use platformio, this change is already there in the lib folder.

If you use arduino ide, you must copy all the libraries provided in the lib folder, because those are adapted or optimized for Tasmota. These steps of compiler usage are explained in the docs.